ms-iot/iot-adk-addonkit #36: Execution of BuildPkg All fails

masterkawaster, 2016-08-11:
What I have:
windows 10 
adk ver. 1511
wdk
add on kit from branch for windows in version 1511
iot core dashboard
installed msi from this: https://msdn.microsoft.com/subscriptions/downloads/#FileId=67415

I'm using virtual machine to build my image.

I'm following this instruction: https://msdn.microsoft.com/en-us/windows/hardware/commercialize/manufacture/iot/create-a-basic-image

Below some logs from console output:

IoTCore x86 10.0.0.0
C:\iot-adk-addonkit\Tools>BuildPkg All
Building all packages under C:\iot-adk-addonkit\Common\Packages
 Processing Custom.Cmd.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\Custom.Cmd.pkg.log
 Processing Device.SystemInformation.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\Device.SystemInformation.pkg.log
 Processing DeviceLayout.GPT2GB.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\DeviceLayout.GPT2GB.pkg.log
 Processing DeviceLayout.GPT4GB.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\DeviceLayout.GPT4GB.pkg.log
 Processing DeviceLayout.MBR2GB.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\DeviceLayout.MBR2GB.pkg.log
 Processing DeviceLayout.MBR4GB.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\DeviceLayout.MBR4GB.pkg.log
 Processing ImageSettings.CrashControl.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\ImageSettings.CrashControl.pkg.log
 Processing ImageSettings.VideoMode.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\ImageSettings.VideoMode.pkg.log
 Processing OemTools.InstallTools.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\OemTools.InstallTools.pkg.log
 Processing Provisioning.Auto.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\Provisioning.Auto.pkg.log
 Processing Provisioning.Manual.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\Provisioning.Manual.pkg.log
 Processing Registry.Version.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\Registry.Version.pkg.log
Building all packages under C:\iot-adk-addonkit\Source-x86\Packages
 Processing Appx.Main.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\Appx.Main.pkg.log
 Processing Appx.TextGlowClock.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\Appx.TextGlowClock.pkg.log
Building all packages under C:\iot-adk-addonkit\Source-x86\BSP
 Processing CustomMBM.GPIO.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
 Error : Failed to create package. See C:\iot-adk-addonkit\Build\x86\pkgs\logs\CustomMBM.GPIO.pkg.log

masterkawaster, 2016-08-11:
I checked one more time with win 10 enterprise x86 in version 1511 and it shows me error like this:
[image]

Previously I was working on win 10 enterprise x64 in version 1607.

All are downloaded from msdn subscriber downloads.

Could you point me what am I missing?

parameshbabu, 2016-08-11:
The issue is due to the WDK version (14393). You have the latest WDK (1607 version) installed with older tools and they are not compatible. 
I noticed that the link to download WDK 1511 is not listed anymore and have given feedback to restore this link. Meanwhile, if you have the WDK 1511 version already available, please install that and try.

masterkawaster, 2016-08-16:
yes that was it :) thanks!

closed 2016-08-16 by masterkawaster

hubert-associates, 2016-10-05:
Hi, I'm looking at the same errors but I don't see where my version mismatch is. Here is my post with a screenshot of all my current installed versions and the output. Thanks for any tips!

[https://social.msdn.microsoft.com/Forums/windowshardware/en-US/e9d23187-70d8-409c-9655-fde0d727a9f5/errors-creating-basic-image-for-iotcore-arm-using-iotadkaddonkit?forum=HardwareBoards](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/e9d23187-70d8-409c-9655-fde0d727a9f5/errors-creating-basic-image-for-iotcore-arm-using-iotadkaddonkit?forum=HardwareBoards)