ms-iot/iot-adk-addonkit #37: I would like to add a service and task into image...

masterkawaster, 2016-08-16:
I haven't found the solution in examples nor in here:
https://msdn.microsoft.com/en-us/windows/hardware/commercialize/manufacture/iot/add-a-registry-setting-to-an-image

The question is: how to add a service and task into image?

One dirty solution would be to add "sc.exe create" and "schtasks.exe /Create" to oemcustomization.cmd but this script will be invoked after system starts so they won't be included in image for real...

By the way - what mechanism fire oemcustomization.cmd at the system start? There's no run and runonce in registry, I couldn't find anything that is responsible for triggering this...

masterkawaster, 2016-08-17:
Ok, I found that the oemcustomization.cmd is triggered by system task that is scheduled on system startup. Script name is hardcoded... But this seems not to be the proper place for putting my service and task installation as it is possible that this file can be overrided by someone else...

parameshbabu, 2016-08-17:
OEMCustomization.cmd is indeed the option but this needs to be defined by OEM for his image and should not be used by apps.  For background task, there is a way to wrap into appx - see IoTOnboardingtask sample in  ms-iot/samples.
Can you elaborate on what kind of service are you trying to develop?

masterkawaster, 2016-08-18:
Thanks for suggestions!
ok, so there's probably an option to wrap this system task into background task and then into appx. I wanted it to listen to the wmi events and it looks that when it would be a background task it should work similar... But still it would have to be installed somehow on this windows image wouldn't it? I currently do not see any other option than installing this through the OEMCustomization.cmd...

Service is simple windows system service similar as you can see in services.msc on regular windows machines with simple api and responsible for communication with other system components. 

I want my application to port from regular windows to win iot core - and it is working now but I have a problem of how to deliver and install all components :) I tried with appx and inf files and now I want to be a part of image which would still make sense to me.

parameshbabu, 2016-09-28:
We are looking into other options here and will post an update when available.

hubert-associates, 2016-11-16:
If you are creating an custom OEM-Image/FFU (i.e. using this github adk-addonkit to build it) with your app package pre-deployed, then look in / use the AppInstall.cmd in your Source-arm/Packages/Appx.MyApp/AppInstall folder. This is called upon install for your app with admin privs.

parameshbabu, 2016-11-16:
That's a neat way if you are installing the appx via packaging way. thanks @hubert-associates for the tip.  We are still looking at more formal way of declaring a service in the system and I'll update this thread when we have a solution.

parameshbabu, 2017-10-17:
Adding services is now possible with the wm.xml file format. See Source-arm\Packages\AzureDM.Services\AzureDM.Services.wm.xml  for an example.

closed 2017-10-17 by parameshbabu