ms-iot/iot-adk-addonkit #38: SampleAProv.ppkg for x64 doesn't exist

jgoyens, 2016-08-21:
For x64, "BuildPkg All" fails because
C:\iot-adk-addonkit\Source-x64\Products\SampleA\prov\SampleAProv.ppkg doesn't exist.

How to correct?

parameshbabu, 2016-08-21:
you can create the SampleAProv.ppkg with the command below
createprovpkg C:\iot-adk-addonkit\Source-x64\Products\SampleA\prov\cuztomizations.xml C:\iot-adk-addonkit\Source-x64\Products\SampleA\prov\SampleAProv.ppkg

Normally, if you try to build the SampleA product using `buildimage SampleA Test`, this provisioning package is created if not present.

parameshbabu, 2016-08-22:
Added `buildprovpkg.cmd` and this is now invoked in the `buildpkg all`.
So this issue should be fixed with this.

closed 2016-08-22 by parameshbabu