ms-iot/iot-adk-addonkit #47: Execution of BuildPkg All fails (like issue #36) but what is missing here??

hubert-associates, 2016-10-05:
Hi, I'm looking at the same errors as issue #36 but I don't see where my version mismatch is. Here is my post with a screenshot of all my current installed versions and the output. Thanks for any tips!

[https://social.msdn.microsoft.com/Forums/windowshardware/en-US/e9d23187-70d8-409c-9655-fde0d727a9f5/errors-creating-basic-image-for-iotcore-arm-using-iotadkaddonkit?forum=HardwareBoards](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/e9d23187-70d8-409c-9655-fde0d727a9f5/errors-creating-basic-image-for-iotcore-arm-using-iotadkaddonkit?forum=HardwareBoards)

parameshbabu, 2016-10-05:
can you please provide the Windows 10 Desktop version as well?

hubert-associates, 2016-10-06:
Thanks. Sure. From "About my computer":  10.0.14393, HW abstraction layer 14393.206

parameshbabu, 2016-10-06:
Ok thanks. From the configuration point of view, all seems correct and this should work normally. It seems the tools folder may be corrupt and the required tools are not present. Can you try reinstalling the ADK and see if this fixes the issue?

hubert-associates, 2016-10-07:
Thanks. I had already tried that. I did it again for good measure, and I assured that I had the right version and I installed all recomended ADK packages plus the client and the Application Compatability Tools. The installation went fine. However, below is the output, same problem :-< I am wondering where these PkgBldr.Tools really come from and where I should find them.  I also installed your newest master drop and made zero changes. Is there a sequence in which I have to install things perhaps? 

IOTADK_ROOT : C:\rht\_HxAOffCloud\Oe\IoTCore-Oem-Product-Stuff\iot-adk-addonkit-master\iot-adk-addonkit-master
WDK_VERSION : 10.0.14393.0
COREKIT_VER : 14393.0
OEM_NAME    : Contoso

Set Environment for Architecture
Choose 1 for ARM, 2 for x86 and 3 for x64 : 1

Configuring for arm architecture
BSP_ARCH    : arm
BSP_VERSION : 10.0.0.0

IoTCore arm 10.0.0.0
C:\rht\_HxAOffCloud\Oe\IoTCore-Oem-Product-Stuff\iot-adk-addonkit-master\iot-adk-addonkit-master\Tools>
IoTCore arm 10.0.0.0
C:\rht\_HxAOffCloud\Oe\IoTCore-Oem-Product-Stuff\iot-adk-addonkit-master\iot-adk-addonkit-master\Tools>InstallOEMCerts
MY "Personal"
CertUtil: -delstore command completed successfully.
Root "Trusted Root Certification Authorities"
Signature matches Public Key
Related Certificates:

Exact match:
Element 7:
Serial Number: 6f4206c6fda56090458a7602f83db375
Issuer: CN=Windows Phone OEM Root 2013 (TEST ONLY), O=Microsoft Partner, OU=Windows Phone, L=Redmond, S=Washington, C=US
 NotBefore: 4/9/2013 11:17 PM
 NotAfter: 4/9/2018 11:17 PM
Subject: CN=Windows Phone OEM Root 2013 (TEST ONLY), O=Microsoft Partner, OU=Windows Phone, L=Redmond, S=Washington, C=US
Signature matches Public Key
Root Certificate: Subject matches Issuer
Cert Hash(sha1): 53 67 1a 6d e6 e8 d9 fa 0d 61 0e c5 63 f1 0f 2c 85 39 eb ee

Certificate "Windows Phone OEM Root 2013 (TEST ONLY)" already in store.
CertUtil: -addstore command completed successfully.
Root "Trusted Root Certification Authorities"
Signature matches Public Key
Related Certificates:

Exact match:
Element 1:
Serial Number: be1324dde4dd39bd49d85f56fc365bdb
Issuer: CN=Windows Phone OEM Root 2013 (TEST ONLY), O=Microsoft Partner, OU=Windows Phone, L=Redmond, S=Washington, C=US
 NotBefore: 4/9/2013 11:14 PM
 NotAfter: 4/9/2018 11:14 PM
Subject: CN=Windows Phone OEM Root 2013 (TEST ONLY), O=Microsoft Partner, OU=Windows Phone, L=Redmond, S=Washington, C=US
Signature matches Public Key
Root Certificate: Subject matches Issuer
Cert Hash(sha1): c0 4a 40 00 e1 c8 66 48 57 7a 03 fe 46 09 fc fb 1b d4 72 1f

Certificate "Windows Phone OEM Root 2013 (TEST ONLY)" already in store.
CertUtil: -addstore command completed successfully.
The object or property already exists. 0x80092005 (-2146885627 CRYPT_E_EXISTS)

Certificate "Windows Phone OEM Root 2013 (TEST ONLY)" already in store.

CertUtil: -importPFX command completed successfully.
The object or property already exists. 0x80092005 (-2146885627 CRYPT_E_EXISTS)

Certificate "Windows Phone Intermediate 2013 (TEST ONLY)" already in store.

CertUtil: -importPFX command completed successfully.
The object or property already exists. 0x80092005 (-2146885627 CRYPT_E_EXISTS)

Certificate "Windows Phone Intermediate FFU Cert 2013 (TEST ONLY)" already in store.

CertUtil: -importPFX command completed successfully.
The object or property already exists. 0x80092005 (-2146885627 CRYPT_E_EXISTS)

Certificate "Windows Phone OEM Test Platform Key Cert 2013 (TEST ONLY)" already in store.

CertUtil: -importPFX command completed successfully.
The object or property already exists. 0x80092005 (-2146885627 CRYPT_E_EXISTS)

Certificate "Windows Phone OEM Test Cert 2013 (TEST ONLY)" already in store.

CertUtil: -importPFX command completed successfully.
Certificate "Windows Phone OEM HAL Extension Test Cert 2013 (TEST ONLY)" added to store.

The object or property already exists. 0x80092005 (-2146885627 CRYPT_E_EXISTS)

Certificate "Windows Phone OEM HAL Extension Test Cert 2013 (TEST ONLY)" already in store.

CertUtil: -importPFX command completed successfully.
The object or property already exists. 0x80092005 (-2146885627 CRYPT_E_EXISTS)

Certificate "Windows Phone OEM App Test Cert 2013 (TEST ONLY)" already in store.

CertUtil: -importPFX command completed successfully.
The object or property already exists. 0x80092005 (-2146885627 CRYPT_E_EXISTS)

Certificate "Windows Phone OEM PP Test Cert 2013 (TEST ONLY)" already in store.

CertUtil: -importPFX command completed successfully.
The object or property already exists. 0x80092005 (-2146885627 CRYPT_E_EXISTS)

Certificate "Windows Phone OEM PPL Test Cert 2013 (TEST ONLY)" already in store.

CertUtil: -importPFX command completed successfully.
MY "Personal"
Deleting Certificate 5: CN=Windows Phone OEM HAL Extension Test Cert 2013 (TEST ONLY), O=Microsoft Partner, OU=Windows Phone, L=Redmond, S=Washington, C=US:ae55c35208026c0ed24b41e2c36db95c5e6635c3
CertUtil: -delstore command completed successfully.

IoTCore arm 10.0.0.0
C:\rht\_HxAOffCloud\Oe\IoTCore-Oem-Product-Stuff\iot-adk-addonkit-master\iot-adk-addonkit-master\Tools>BuildPkg All
Building all provisioning packages
Processing all provisioning packages
 Processing SampleA
 Processing SampleB
 Processing SampleC
Building all packages under C:\rht\_HxAOffCloud\Oe\IoTCore-Oem-Product-Stuff\iot-adk-addonkit-master\iot-adk-addonkit-master\Common\Packages
 Processing Custom.Cmd.pkg.xml

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
Could Not Find C:\rht\_HxAOffCloud\Oe\IoTCore-Oem-Product-Stuff\iot-adk-addonkit-master\iot-adk-addonkit-master\Build\arm\pkgs\*.spkg
 Processing Device.SystemInformation.pkg.xml

parameshbabu, 2016-10-07:
---`
However, there is nothing in the output folder: /Build/arm/pkgs except for the log directory???
Shouldn't I find something there?
---
No this is fine. The ppkg is created in the same folder as the customizations.xml and only the cab files are created in the above folder.
Seems that in your machine PkgBldr.Tools.dll is not found. Can you go to the directory containing pkggen.exe and just run pkggen.exe without parameters?  You should find it printing the usage instructions. If not, then the PkgBldr.Tool.dll is not present or not found.

hubert-associates, 2016-10-08:
Yes, pkggen.exe is there and throws pretty much the same exception (below). So why is PkgBldr.Tool.dll missing and where does it come from? I see Microsoft.CompPlat.PkgBldr. What/where is CompPlat? Searching...
Thanks for any tips..

C:\Program Files (x86)\Windows Kits\10\Tools\bin\i386>pkggen.exe

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'PkgBldr.Tools, Version=10.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)

C:\Program Files (x86)\Windows Kits\10\Tools\bin\i386>

hubert-associates, 2016-10-09:
A fresh install of everything on another machine solved the problem: pkgblder.tools.dll is back in \Windows Kits\10\Tools\bin\i386. After re-installing anything I could think of on my main machine, this (and others?) are still missing.

closed 2016-10-29 by hubert-associates