ms-iot/iot-adk-addonkit #49: Getting a crash in combase.dll Azure StorageClient ONLY on our oem-image/FFU build.

hubert-associates, 2016-10-29:
Hi Paramesh, do you have a tip regarding this crash which happens only on our oem custom image/FFU build? I used your example "to the T" as far as I can tell. Do I need to add capabilities or do some other configuration to make the Microsoft Azure StorageClient (header creation) work, as it does with identical code on the retail build?

Here are details: [http://stackoverflow.com/questions/40267303/iot-core-custom-oem-image-ffu-crashing-in-windowsruntime-blob-cloudblob-cs-only](details)

Any tips or suspicions welcome.
Thanks in advance.

parameshbabu, 2016-11-11:
Checking with the Azure team on what can cause this issue. Will keep posted

hubert-associates, 2016-11-12:
Thanks, still working on this since I can't get the debugger to work with the OEM-deployment on the board :-( I've tried all the tricks I know. Any tip here on how to connect to my OEM-FFU-App from a VisualStudio2015 in LAN?

hubert-associates, 2016-11-16:
OK, I found the difference, now is the question why it is different. Since iot-core/UWP provices no way to get the board-UUID, I use the MAC of the primary network interface. To get this, I use this [http://embedded101.com/BruceEitman/entryid/676/Windows-10-IoT-Core-Getting-the-MAC-Address-from-Raspberry-Pi](http://embedded101.com/BruceEitman/entryid/676/Windows-10-IoT-Core-Getting-the-MAC-Address-from-Raspberry-Pi ) which requires that a webserver be running, which it normally is, otherwise the console webapp would not work. However, on the OEM-Custom-Build-Version I get a crash in this routine. I don't know where since I can't debug, but it crashes, and I get a null back, which causes my azure storage connect to crash. I do not block processing since I have a retry loop... Anyway, what is the difference or what must we do to enable this code to also work in the OEM build?

hubert-associates, 2016-11-17:
The grass routes issue is: all  I really need is a friggin unique ID for the RPi board from somewhere... which does not seem possible via C#!?

hubert-associates, 2016-11-24:
Any ideas here how to get that MAC address or a board ID?

parameshbabu, 2016-11-28:
If you are using RPi, then relying on mac address is also not a good idea as we have noticed that these also have not been unique in some cases. 
See [Building Secure Apps for Windows 10 IoT Core](https://blogs.windows.com/buildingapps/2016/07/20/building-secure-apps-for-windows-iot-core/#BFGu4ry0LtV7Tijo.97) for a possible solution

hubert-associates, 2016-11-28:
Thanks, yes, MACs are not what they used to be :-) I will certainly take a look at your link! Greatly appreciated.