ms-iot/iot-adk-addonkit #50: Buildpkg no longer works after 1607 install

dvescovi1, 2016-11-11:
After installing update IoT 1607 packages BuildPkg All no longer works.
Went through the tool install steps, set the OEM name and installed certs etc.
BuildPkg All fails and I can not find the issue.
I think this worked before with the original release. 

Fails with:
C:\iot-adk-addonkit\Tools>BuildPkg All
Building all provisioning packages
Processing all provisioning packages
 Processing Provisioning.Enroll
 Processing Provisioning.ProSKU
 Processing ProductA
 Processing SampleA
 Processing SampleB
 Processing SampleC
Building all packages under C:\iot-adk-addonkit\Common\Packages
 Processing Custom.Cmd.pkg.xml
Microsoft (C) spkggen 8.0.11111.0

fatal error : Invalid command line arguments:Command-line has too many parameters
info: >> Command-line has too many parameters
info: Usage: SPkgGen.exe [project] [/config:x] [/xsd:x] [/output:x] [/version:x] [/build:x] [/cpu:x] [/languages:x] [/resolutions:x] [/variables:x] [/spkgsout:x] [/toolPaths:x] [[+|-]toc] [[+|-]compress] [[+|-]diagnostic] [[+|-]nohives] [[+|-]isRazzleEnv] [/?]
info:   [project]············· Full path to the package project file. Values: free
info:                          text; default=
info:   [/config]············· File with globally defined variables. Values: free
info:                          text; default=
info:   [/xsd]················ Path to write the PkgGen auto-generated schema to.
info:                          Values: free text; default=
info:   [/output]············· Output directory. Values: free text; default= .
info:   [/version]············ Version string in the form of <major>.<minor>.<qfe>.<b
info:                          uild>. Values: free text; default= 0.0.0.0
info:   [/build]·············· Build type string. Values: {fre|chk}; default= fre
info:   [/cpu]················ CPU type. Values: {x86|arm|arm64|amd64}; default= ARM
info:   [/languages]·········· Supported language identifier list  separated by ';'.
info:                          Values: free text; default=
info:   [/resolutions]········ Supported resolution identifier list  separated by
info:                          ';'. Values: free text; default=
info:   [/variables]·········· Additional variables used in the project file
info:                          syntax:<name>=<value>;<name>=<value>;.... Values: free
info:                          text; default=
info:   [/spkgsout]··········· Create an output file containing a list of generated
info:                          SPKG's. Values: free text; default=
info:   [/toolPaths]·········· Directories containing tools needed by spkggen.exe.
info:                          Values: free text; default=
info:   [toc]················· Building TOC files instead of the actual package.
info:                          Values: precede by [+] or [-]; default= False
info:   [compress]············ Compressing the generated package. Values: precede by
info:                          [+] or [-]; default= False
info:   [diagnostic]·········· Enable debug output. Values: precede by [+] or [-];
info:                          default= False
info:   [nohives]············· Indicates whether or not this package has no hive
info:                          dependency. Values: precede by [+] or [-]; default=
info:                          False
info:   [isRazzleEnv]········· Indicates whether or not spkggen is running in a
info:                          razzle environment. Values: precede by [+] or [-];
info:                          default= False
info:   [/?]·················· Displays this usage string

 Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\Custom.Cmd.pkg.log
 Processing Device.SystemInformation.pkg.xml
Microsoft (C) spkggen 8.0.11111.0

fatal error : Invalid command line arguments:Command-line has too many parameters
info: >> Command-line has too many parameters
info: Usage: SPkgGen.exe [project] [/config:x] [/xsd:x] [/output:x] [/version:x] [/build:x] [/cpu:x] [/languages:x] [/resolutions:x] [/variables:x] [/spkgsout:x] [/toolPaths:x] [[+|-]toc] [[+|-]compress] [[+|-]diagnostic] [[+|-]nohives] [[+|-]isRazzleEnv] [/?]
info:   [project]············· Full path to the package project file. Values: free
info:                          text; default=
info:   [/config]············· File with globally defined variables. Values: free
info:                          text; default=
info:   [/xsd]················ Path to write the PkgGen auto-generated schema to.
info:                          Values: free text; default=
info:   [/output]············· Output directory. Values: free text; default= .
info:   [/version]············ Version string in the form of <major>.<minor>.<qfe>.<b
info:                          uild>. Values: free text; default= 0.0.0.0
info:   [/build]·············· Build type string. Values: {fre|chk}; default= fre
info:   [/cpu]················ CPU type. Values: {x86|arm|arm64|amd64}; default= ARM
info:   [/languages]·········· Supported language identifier list  separated by ';'.
info:                          Values: free text; default=
info:   [/resolutions]········ Supported resolution identifier list  separated by
info:                          ';'. Values: free text; default=
info:   [/variables]·········· Additional variables used in the project file
info:                          syntax:<name>=<value>;<name>=<value>;.... Values: free
info:                          text; default=
info:   [/spkgsout]··········· Create an output file containing a list of generated
info:                          SPKG's. Values: free text; default=
info:   [/toolPaths]·········· Directories containing tools needed by spkggen.exe.
info:                          Values: free text; default=
info:   [toc]················· Building TOC files instead of the actual package.
info:                          Values: precede by [+] or [-]; default= False
info:   [compress]············ Compressing the generated package. Values: precede by
info:                          [+] or [-]; default= False
info:   [diagnostic]·········· Enable debug output. Values: precede by [+] or [-];
info:                          default= False
info:   [nohives]············· Indicates whether or not this package has no hive
info:                          dependency. Values: precede by [+] or [-]; default=
info:                          False
info:   [isRazzleEnv]········· Indicates whether or not spkggen is running in a
info:                          razzle environment. Values: precede by [+] or [-];
info:                          default= False
info:   [/?]·················· Displays this usage string

 Error : Failed to create package. See C:\iot-adk-addonkit\Build\arm\pkgs\logs\Device.SystemInformation.pkg.log
 Processing DeviceLayout.GPT2GB.pkg.xml
Microsoft (C) spkggen 8.0.11111.0

fatal error : Invalid command line arguments:Command-line has too many parameters

What can be causing these issues?

dvescovi1, 2016-11-11:
All associated tools (ADK, WDK etc.) installed were the 1607 version.

parameshbabu, 2016-11-11:
looks like the cmd file corruption or something. The syntax for the spkggen is erroneous here. 
Can you check if the createpkg.cmd file is intact? The below is the line in createpkg.cmd that's failing.

call pkggen.exe "%INPUT_FILE%" /config:"%PKG_CONFIG_XML%" /output:"%PKGBLD_DIR%" /version:%PKG_VER% /build:fre /cpu:%BSP_ARCH% /variables:"_RELEASEDIR=%BLD_DIR%\;PROD=%PRODUCT%;PRJDIR=%SRC_DIR%;COMDIR=%COMMON_DIR%;BSPVER=%PKG_VER%;BSPARCH=%BSP_ARCH%;OEMNAME=%OEM_NAME%" /nohives

dvescovi1, 2016-11-11:
Adk add on directly from GitHub baseline. Tried reinstalling several times ...same result.

dvescovi1, 2016-11-14:
Can anyone verify that with the 1607 version of the tools with the master branch of the ADK ADD ON really does work?
By the way the 1607 Imaging and Configuration Designer build in the 1607 ADK works just fine for IoT core builds.

dvescovi1, 2016-11-19:
So I re-verified the 1511 build is good. Installed 1511 ADK, IoTcore and ADKAddOn.
Did a Buildpkg All and it works just fine. 
Uninstalled all three and installed 1607 ADK, IoTcore and master branch of ADKAddOn.
Buildpkg All FAILS as described above.
...waiting workaround or fix.

dvescovi1, 2016-11-19:
so a closer look at the logs shows:
Running SPkgGen.exe Custom.Cmd.pkg.xml /config:"C:\Program Files (x86)\Windows Kits\10\Tools\bin\i386\pkggen.cfg.xml" /output:C:\iot-adk-addonkit-master\Build\x86\pkgs /build:fre /cpu:X86 /variables:_RELEASEDIR=C:\iot-adk-addonkit-master\Build\x86\;PROD=SampleA;PRJDIR=C:\iot-adk-addonkit-master\Source-x86;COMDIR=C:\iot-adk-addonkit-master\Common;BSPVER=10.1.14393.0;BSPARCH=x86;OEMNAME=Contoso /toolPaths:%RazzleToolPath%\x86 /nohives /version:10.1.14393.0 /spkgsout:C:\Users\David Vescovi\AppData\Local\Temp\tmp62E.tmp 

is the problem. a look at the last parameter /spkgsout:C:\Users\David Vescovi\AppData\Local\Temp\tmp62E.tmp  seems to be what is causing the issue. I think it should be something like /spkgsout:%TEMP%\tmp62E.tmp or at least have quotes around path as the space in my name in the path is what is causing the issue ..the parser is seeing this as a parameter separator and choking.

dvescovi1, 2016-11-19:
so even though my environment variables from within ADKAddOn shell shows:
....
SystemRoot=C:\WINDOWS
TEMP=C:\Users\DAVIDV~1\AppData\Local\Temp
TMP=C:\Users\DAVIDV~1\AppData\Local\Temp
TOOLS_DIR=C:\ms-iot\iot-adk-addonkit\Tools

It appears when executing the Buildpkg All command the TMP variable got expanded to the long file name path with spaces. 
I worked around the issue by adding the line SET TMP=C:\TMP to the setenv.cmd in the ADKAddOn tools directory.
My guess is something changed in the ADK package gen parser between versions.

Dave

parameshbabu, 2017-02-13:
Sorry for the late ack on this issue. The issue is reported as a bug internally and will update this thread when we have a fix for this. Till then, recommend to work with this workaround of setting own TMP directory.

parameshbabu, 2017-10-17:
Please try with the latest release and confirm.

closed 2017-10-17 by parameshbabu