ms-iot/iot-adk-addonkit #56: when update ffu image, always get the error :  0x80070002

txiangyang, 2017-03-06:
Hi, 
I am using the updateinputxml way to update flash.ffu image, but it always gives such error: 
know how to fix it?
In IoTShell env, run cmd: 
imageapp.exe D:\Image\Update\flash.ffu /updateinputxml:"D:\iot-adk-addonkit-master\iot-adk-addonkit-master\Templates\UpdateInput.xml"
the updateinput.xml is: 
<?xml version="1.0" encoding="utf-8"?>
<UpdateOSInput 
      xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" 
      xmlns:xsd="http://www.w3.org/2001/XMLSchema" 
      xmlns="http://schemas.microsoft.com/embedded/2004/10/ImageUpdate">
      <Description>ISH</Description>
      <PackageFiles>
            <PackageFile>D:\iot-adk-addonkit-master\iot-adk-addonkit-master\Build\amd64\pkgs\%OEM_NAME%.ISH.Main.cab</PackageFile>
            <PackageFile>D:\iot-adk-addonkit-master\iot-adk-addonkit-master\Build\amd64\pkgs\%OEM_NAME%.ISH.Bus.cab</PackageFile>
            <PackageFile>D:\iot-adk-addonkit-master\iot-adk-addonkit-master\Build\amd64\pkgs\%OEM_NAME%.ISH.HID.cab</PackageFile>
            <PackageFile>D:\iot-adk-addonkit-master\iot-adk-addonkit-master\Build\amd64\pkgs\%OEM_NAME%.ISH.UMDF.cab</PackageFile>
      </PackageFiles>
</UpdateOSInput>

The error in the log is: 

{3/6/2017 3:20:47 PM} Error: [00:00:13] onecore\base\cbs\mobile\iucore\updateinput.cpp, UpdateInput::LoadXML, line 102, Error  , Failed call to FileStream::OpenFile on file '\\?\C:\Users\admin\AppData\Local\Temp\ebrtf15r.mrz.mnt\Data\SharedData\DuShared\UpdateInput.xml', error is 0x80070002
{3/6/2017 3:20:47 PM} Error: [00:00:13] onecore\base\cbs\mobile\iucore\updateinput.cpp, UpdateIsUsingSD, line 770, Error  , Failed to initialize input file, error is 0x80070002

Any one know how to fix it?

parameshbabu, 2017-03-06:
updating FFU is only for experimental purposes and is not supported for formal image creation. 
I would recommend to create the FFU directly instead of updating the FFU.

IoTGirl, 2017-05-02:
Closing based on answer given.

closed 2017-05-02 by IoTGirl