ms-iot/iot-adk-addonkit #90: 1703 supported, do I generate 1703 error

Yuexs, 2017-05-02:
1703 supported, do I generate 1703 error

IoTGirl, 2017-05-02:
Hi @Yuexs,
Can you please explain what error you get and how you get it?
Sincerely,
IoTGirl

Yuexs, 2017-05-03:
ADK 1703,iot-adk-addonkit 1703 Error when creating image, but reinstall ADK 14393,iot-adk-addonkit 14393 is normal

Yuexs, 2017-05-03:
I'm creating a raspberry pie 3B image

IoTGirl, 2017-05-04:
This is really not enough information @Yuexs.  Can you please provide more details as to your setup?

**ENVIRONMENT:**
What Computers, OS, tools and devices are involved with this repro? How are they connected? What versions?

**REPRO STEPS:**
What steps would someone else need to follow to see the issue?

**RESULTS:**
What did you Expect to happen?
What did Actually happen?

**NOTES:**
Any other details/workarounds potentially related info

IoTGirl, 2017-05-16:
Closing due to lack of response: Please see [https://social.msdn.microsoft.com/Forums/en-US/home?forum=WindowsIoT](https://social.msdn.microsoft.com/Forums/en-US/home?forum=WindowsIoT) for other build issues and their solutions

closed 2017-05-16 by IoTGirl