ms-iot/iot-adk-addonkit #104: Never install my application

Monigass, 2017-06-05:
Hi, I have a problem, when I build an image and flash it to the SD memory, the Raspberry pi doesn't even install my app, it only restarts itself.

I have this isue on 15063 version only, the past versions worked perfectly.

What I am doing wrong? 

I've also tried the "Lab 1a: Create a basic image" but I can not see any application.

IoTGirl, 2017-06-06:
Hi @Monigass , 
With the very little information you have provided it is impossible to know but you state that previous versions are working and the latest is not so my first suspicion is that you do not have the tools for 15063 installed, only the tools for prior builds.

If you want further assistance you will need to provide far more detail as to your setup, the steps you take and the various error logs you have reviewed.

Sincerely,
IoTGirl

parameshbabu, 2017-06-13:
@Monigass, do you install your app using the provisioning package ( using updated newappxpkg.cmd )?

Monigass, 2017-06-14:
@parameshbabu, Yes, I follow this steps:

https://docs.microsoft.com/es-es/windows-hardware/manufacture/iot/deploy-your-app-with-a-standard-board

I've done this with a Test package, but I can't with the PreinstallKit version (from my Windows Dev Center account): 

https://developer.microsoft.com/en-us/windows/iot/docs/store

Thaks.

parameshbabu, 2017-06-17:
@Monigass, I found an issue with the license installation for the store apps and this is now fixed. 
Use the latest version of the develop branch ( main change is in appx2pkg.cmd that gets called from newappxpkg.cmd). Let me know if this works for you and I can close this issue.

parameshbabu, 2017-06-17:
Fix part of PR #106 . The master branch will be also updated shortly.

parameshbabu, 2017-06-17:
Use PR #107 for the master branch for this fix.

IoTGirl, 2017-06-19:
@Monigass Can you please confirm this fix addresses your issue?

closed 2017-08-06 by parameshbabu