ms-iot/iot-adk-addonkit #113: Building 15063 Image for Pi 3, Bluetooth Not Working

brazierd, 2017-06-24:
Hi

I had successfully built images for the Pi 3 for 14393 with the ADK Add-On Kit but with 15063 I found that the built-in Bluetooth didn't work - the settings page shows "No usable Bluetooth adapter can be found on this computer: The device is not ready for use. (Excep_FromHResult 0x800710DF)".

After a lot of trial and error I found I could fix this by explicitly including the RPI_BTWSERIALH5BUS_DRIVER feature in TestOEMInput.xml:

    <OEM>
      
      <Feature>RPI2_DRIVERS</Feature>
      <Feature>RPI3_DRIVERS</Feature>
      **<Feature>RPI_BTWSERIALH5BUS_DRIVER</Feature>**

Though I think it would work equally well to add the RPI3_DRIVERS feature id to the relevant package in RPi2FM.xml:

      <PackageFile Path="%BSPPKG_DIR%" Name="%OEM_NAME%.RPi.piminiuart.cab">
        <FeatureIDs>
          <FeatureID>RPI_MINIUART</FeatureID>
          **<FeatureID>RPI3_DRIVERS</FeatureID>**
        </FeatureIDs>
      </PackageFile>

However, I don't understand why either should be needed, as I expected an image built with the default scripts from the Add-On Kit to be the same as the standard image from the IoT Dashboard etc., which was the case with 14393.

I could propose either change in the repository but worndered if I was misunderstanding something?

Regards

David

parameshbabu, 2017-06-27:
@brazierd this is an acknowledgement for the request to have reference configuration in line with our shipping FFU. The existing samples were more of a concept demonstrator and was not intentional to be consistent with our shipping FFU.

parameshbabu, 2017-06-28:
Fixed this by adding it to RPi3_Drivers. Note that the SampleA/SampleB products do not include this at present. However any new product you create will have this included as default.

closed 2017-06-28 by parameshbabu