ms-iot/iot-adk-addonkit #114: <OEMFeatureGroups /> bad closing label

juasmal, 2017-06-27:
Hi, 

I think that this label should be closed as </OEMFeatureGroups>, not <OEMFeatureGroups />

I 'm doing a image for an app in the store (OEM based).

When i executed the step "Run buildfm oem to generate updated files in the MergedFMs folder. This has to be done every time any time an FM file is modified."  in lab 1c

The console show me a msg: "error: Featuremerger failed for OEMFMfilesList.xml see c:\C:\IoT-ADK-AddonKit\iot-adk-addonkit-master\Build\arm\buildfm_oem.log"

that has...

_> info: Trying to load file 'C:\IoT-ADK-AddonKit\iot-adk-addonkit-master\Build\arm\InputFMs\OEMFMFileList.xml' as a FM file list ...
fatal error : Error: FeatureAPI!ValidateInput: Unable to parse Feature Manifest XML file 'C:\IoT-ADK-AddonKit\iot-adk-addonkit-master\Build\arm\InputFMs\OEMCommonFM.xml'.System.InvalidOperationException: Error en el documento XML (81, 5). ---> System.Xml.XmlException: La etiqueta de apertura 'OEMFeatureGroups' en la l¡nea 71 posici¢n 6 no coincide con la etiqueta de cierre de 'Features'. l¡nea 81, posici¢n 5.
fatal error :    en System.Xml.XmlTextReaderImpl.Throw(Exception e)
fatal error :    en System.Xml.XmlTextReaderImpl.Throw(String res, String[] args)
fatal error :    en System.Xml.XmlTextReaderImpl.ThrowTagMismatch(NodeData startTag)
fatal error :    en System.Xml.XmlTextReaderImpl.ParseEndElement()
fatal error :    en System.Xml.XmlTextReaderImpl.ParseElementContent()
fatal error :    en System.Xml.XmlTextReaderImpl.Read()
fatal error :    en System.Xml.XmlTextReader.Read()
fatal error :    en System.Xml.XmlTextReaderImpl.Skip()
fatal error :    en System.Xml.XmlTextReader.Skip()
fatal error :    en System.Xml.Serialization.XmlSerializationReader.UnknownNode(Object o, String qnames)
fatal error :    en Microsoft.Xml.Serialization.GeneratedAssembly.XmlSerializationReaderFeatureManifest.Read22_FMFeatures(Boolean isNullable, Boolean checkType)
fatal error :    en Microsoft.Xml.Serialization.GeneratedAssembly.XmlSerializationReaderFeatureManifest.Read32_FeatureManifest(Boolean isNullable, Boolean checkType)
fatal error :    en Microsoft.Xml.Serialization.GeneratedAssembly.XmlSerializationReaderFeatureManifest.Read33_FeatureManifest()
fatal error :    --- Fin del seguimiento de la pila de la excepci¢n interna ---
fatal error :    en System.Xml.Serialization.XmlSerializer.Deserialize(XmlReader xmlReader, String encodingStyle, XmlDeserializationEvents events)
fatal error :    en System.Xml.Serialization.XmlSerializer.Deserialize(TextReader textReader)
fatal error :    en Microsoft.WindowsPhone.FeatureAPI.FeatureManifest.ValidateAndLoad(FeatureManifest& fm, String xmlFile, IULogger logger)
fatal error : Error: failed to load Feature Manifest: $(FMDirectory)\OEMCommonFM.xml_

The <OEMFeatureGroups /> it's defined at BSPFMTemplate, and others like OEMCOmmonFM.xml ...this was that caused my issue when y was doing "buildfm oem".

I changed the label ending at:  my  "C:\IoT-ADK-AddonKit\iot-adk-addonkit-master\Common\Packages\OEMCommonFM.xml",  and all work fine.

But i modified that label there, not at BSPFMTemplate. I Don't khow if when i created the common packages it was generated using the BSPFMTemplate. 

Thanks

parameshbabu, 2017-06-28:
Hi,
do you see this error only with OEMFeatureGroups and not with other tags such as <Microsoft />  <MSFeatureGroups />?

Paramesh

parameshbabu, 2017-08-06:
Lacking more info, closing now. Reopen if more details available.

closed 2017-08-06 by parameshbabu