ms-iot/iot-adk-addonkit #130: newupdate.cmd not found

IotCore-Developer, 2017-09-11:
When I try to create an update to test the command cannot be found.

See tutorial :

[https://docs.microsoft.com/en-us/windows-hardware/service/iot/updating-iot-core-apps#Test_and_track_the_update](url)

Step 1.

How must this be done without this tool ?

parameshbabu, 2017-09-13:
@IotCore-Developer this command had been taken off as the mechanism was changed. We will add an equivalent of this soon and update the doc page as well. Stay tuned.

IotCore-Developer, 2017-09-15:
Hi,

When Will this Be realized?
Is IT a Matter of days, weeks?

Best regards

> @IotCore-Developer this command had been taken off as the mechanism was
changed. We will add an equivalent of this soon and update the doc page as
well. Stay tuned.
https://ci6.googleusercontent.com/proxy/ax8PZbh8Klb19zV8zWAjKLBIlb9d8hMFAOwzViKqLZZWT7Bxd9ponUj8gLLd7QKNPQ9HVtUukXy5690t-nZtn_Byn0QqVlsRg28IpujuAzH48IyAWD6_NnDwDOyAee8N5Omn6_6IX1ufLi6zlIgm2nrf97zt5Q=s0-d-e1-ft#https://github.com/notifications/beacon/AeYvV2ayWdesnFvYvwxJND7IDn3cTYH8ks5siBhagaJpZM4PTkxi.gif>

parameshbabu, 2017-09-16:
@IotCore-Developer this should be available in a months time. early oct.

IotCore-Developer, 2017-09-17:
Ok thx for the update

From: Paramesh Babu Rengarao Thyada
Sent: Saturday, September 16, 2017 3:13 AM
To: ms-iot/iot-adk-addonkit
Cc: IotCore-Developer; Mention
Subject: Re: [ms-iot/iot-adk-addonkit] newupdate.cmd not found (#130)

@IotCore-Developer this should be available in a months time. early oct.
—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

IotCore-Developer, 2017-10-13:
Hi,

What  is the status of this update ?

Best Regards

From: Paramesh Babu Rengarao Thyada
Sent: Saturday, September 16, 2017 3:13 AM
To: ms-iot/iot-adk-addonkit
Cc: IotCore-Developer; Mention
Subject: Re: [ms-iot/iot-adk-addonkit] newupdate.cmd not found (#130)

@IotCore-Developer this should be available in a months time. early oct.
—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

parameshbabu, 2017-10-17:
newupdate.cmd / buildupdates.cmd are available now. The instructions to use will be posted soon, listed here in a short summary
1. newupdate updatename version
2. set USEUPDATE=updatename
3. newappxpkg to create a new version of the appx package - use same name as the original package. The new pkg will be created in the updatename directory.
4. bp all - build base packages, 
5. buildimage - this will build the  update packages (set with USE_UPDATE flag) and copy them in the right folder to create the image with this update package.

closed 2017-10-17 by parameshbabu

IotCore-Developer, 2017-10-18:
Ok thank you for your infomation.

> newupdate.cmd / buildupdates.cmd are available now. The instructions to
use will be posted soon, listed here in a short summary
>
> newupdate updatename version
> set USEUPDATE=updatename
> newappxpkg to create a new version of the appx package - use same name as
the original package. The new pkg will be created in the updatename
directory.
> bp all - build base packages,
> buildimage - this will build the update packages (set with USE_UPDATE
flag) and copy them in the right folder to create the image with this
update package.
https://ci6.googleusercontent.com/proxy/KqMfDj6fdUUScGMPrB4kk2Vi0uzknjtgK0-T0IYNRial-vPr2aC8BzVYX57SEnYd6QuCOS2gXP10ImVac-JWgNyrYEwycfTnHptTmS7MwBE2A2lDQCC9e3Cj5dNO1O3gHd_fdV9L0ExOFcRf-RXlUoGsKAGuOQ=s0-d-e1-ft#https://github.com/notifications/beacon/AeYvV9gx0qkfKi2NFQznzx0EG0lqIgDxks5stRcegaJpZM4PTkxi.gif>