ms-iot/iot-adk-addonkit #131: Failed to mount UpdateOS WIM

BjornVT, 2017-09-22:
Hi,

I'm trying to build an windows IoT image for custom hw. But i keep running into the following error: 
fatal error : [00:00:00] onecore\base\cbs\mobile\iucore\iustore.cpp, IUStore::MountPendingUpdateOS, line 724, Error  , Failed to mount UpdateOS WIM at ... , error is 0x80070241.
I'v searched the internet but I can't seem to find people with similar errors. Can you give any help/tips to where this error is comming from and how I can fix this?

Kind regards,
Bjorn

f-israel, 2017-09-26:
Hi

I got the error while having a removeable drive (thumbdrive, sd card, etc.) attached. Maybe that's also your error?

parameshbabu, 2017-10-17:
Hi, can you provide more information on this issue? 
As mentioned above, we recommend not to have any detachable drives connected when you do image creation.

mjmeans, 2017-10-31:
I had a similar message a week ago after I tried to stop a build. DISM had left a mounted volume. I had to use "dism /get-MountedWiminfo" to see if you have any left over mounted images and then use "dism /unmount-Wim  /MountDir:<mountDir> /discard" and "dism /cleanup-wim". After that I was able to build again.

parameshbabu, 2017-10-31:
this could be a reason. I'll include a script to clean up potentially stray mount volumes. This mainly occurs when the build process is interrupted mid way.

parameshbabu, 2018-04-03:
Closing this issue. Please open issue if you see this still.

closed 2018-04-03 by parameshbabu