ms-iot/iot-adk-addonkit #139: buildpkg ALL fails on missing dependencies

EngineerLLC, 2017-10-07:
When trying to create an image for IoT Core targeting Raspberry Pi 3, I have followed all of the instructions as described in the document "IoT Core manufacturing guide"  and at "Lab 1a: Create a basic image" the buildpkg ALL fails on missing Qualcomm BSP dependencies.

https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/create-a-basic-image

[buildfm_bsp_QCDB410C.log](https://github.com/ms-iot/iot-adk-addonkit/files/1364157/buildfm_bsp_QCDB410C.log)

Additionally, the links to IoT Core BSP download page(s) appear to be bad...  https://docs.microsoft.com/windows/iot-core/commercialize-your-device/CreateBSP

parameshbabu, 2017-10-07:
If you are building for RPi, you can safely ignore the Qualcomm bsp errors. I will amend the documentation to mention that.
Thanks for pointing out the broken link, i'll fix that as well. Here is the new link
https://docs.microsoft.com/windows/iot-core/build-your-image/createbsps

mjmeans, 2017-10-11:
I too have an error at lab 1a buildpkg all. Here are the details:

 ```
Running FeatureMerger for CustomRpi2
  Exporting CustomRpi2 FM files
  Processing CustomRpi2FMFileList.xml
  Error: Featuremerger failed for CustomRpi2FMFileList.xml. See C:\IoT-ADK-AddonKit\Build\arm\buildfm_bsp_CustomRpi2.log
 Running FeatureMerger for QCDB410C
  Exporting QCDB410C FM files
  Processing QCDB410CFMFileList.xml
  Error: Featuremerger failed for QCDB410CFMFileList.xml. See C:\IoT-ADK-AddonKit\Build\arm\buildfm_bsp_QCDB410C.log
 Running FeatureMerger for Rpi2
  Exporting Rpi2 FM files
  Processing Rpi2FMFileList.xml
  Error: Featuremerger failed for Rpi2FMFileList.xml. See C:\IoT-ADK-AddonKit\Build\arm\buildfm_bsp_Rpi2.log
```

C:\IoT-ADK-AddonKit\Build\arm\buildfm_bsp_Rpi2.log
```
info: Trying to load file 'C:\IoT-ADK-AddonKit\Build\arm\InputFMs\CustomRpi2FMFileList.xml' as a FM file list ...
fatal error : Error: Missing package: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre\RASPBERRYPI.RPi2.BootFirmware.cab
fatal error : Error: Missing package: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre\RASPBERRYPI.RPi2.dwcUsbOtg.cab
fatal error : Error: Missing package: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre\RASPBERRYPI.RPi.LAN9500N650F.cab
fatal error : Error: Missing package: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre\RASPBERRYPI.RPi2.USBWIFI.cab
fatal error : Error: Missing package: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre\RASPBERRYPI.RPi2.RT.WLAN.cab
fatal error : Error: Missing package: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre\RASPBERRYPI.RPi.FT5406.Touch.cab
fatal error : Error: Missing package: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre\RASPBERRYPI.RPi.Wifi.bcmdhd63.cab
fatal error : Error: Missing package: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre\RASPBERRYPI.RPi.Wifi.bcmdhd63.cab
fatal error : Error: Missing package: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre\RASPBERRYPI.RPi.BtwSerialH5Bus.cab
fatal error : Error: Missing package: C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\arm\fre\RASPBERRYPI.RPi.BtwSerialH5Bus.cab

```

parameshbabu, 2017-10-11:
You will have to define BSPPKG_DIR to point to the place where you have stored the BSP cab files. By default it is pointing to the Build\arm\pkgs folder and if the files are not present it flags this error.

mjmeans, 2017-10-11:
I followed the steps in the lab 1a. What you say isn't mentioned at all. Where is this BSPPKD_DIR setting supposed done to be in this lab?

parameshbabu, 2017-10-11:
Took a closer look at the error. Looks like your PC is missing IoTCore Kits.  Is the item 3 in https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/set-up-your-pc-to-customize-iot-core installed?

3. IoT Core Kit package. The .iso package adds the IoT Core packages and feature manifests used to create IoT Core images. By default, these packages are installed to C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail.

mjmeans, 2017-10-11:
I've had the IoT Code ISO installed and using Rpi with Windows IoT core (Dashboard and Remote) for over a year now. The version currently installed is 10.1.14931.1000. The step 3 you mention doesn't state which version to use, or even if that's important. But it is already installed.

EngineerLLC, 2017-10-11:
I suspect an inconsistencies in the folder paths for the IoT ADK between the manufacturing guide and the actual installations may be at least one of the causal factors...
 
Example:  From the manufacturing guide, Lab 1.a: step 1...  
The path to the shell "IoTCoreShell.cmd" is described as "C:\IoT-ADK-AddonKit\".
The path to the shell "IoTCoreShell.cmd" that was created when I unzipped the resource is "C:\IoT-ADK-AddonKit\iot-adk-addonkit-master\".
 
Another factor is missing files...

My original report cited one of the generic error logs "buildfm_bsp_QCDB410C.log" which I understand is not relevant for Raspberry Pi, but that was only one of 4 error logs.  The similar error log "buildfm_bsp_CustomRpi2.log" contains multiple errors such as "C:\IoT-ADK-AddonKit\iot-adk-addonkit-master\Build\arm\pkgs\CustomOrg.RPi.bcm2836sdhc.cab".
 
I installed the IoT Core Kit package from the .iso for Raspberry Pi as per the instructions.  The resultant path is "C:\Program Files (x86)\Windows Kits\10\MSPackages\retail\arm\fre".  There are no other branches in this path after MSPackages, the only content in this path is .cab archives in the fre folder and a search of the entire C:\ drive fails to locate the missing file "bcm2836sdhc.cab" from the error log example in the previous paragraph.

mjmeans, 2017-10-11:
Okay, I see. That IoT Core package is not the same package as the one that has the FFU in it. After downloading that package, all I have left is the Qualcomm log errors. From another forum I see that there is a different download that has those files, but I didn't search for the location of that download.

mjmeans, 2017-10-11:
Well it bugged me so I searched for it. The answer is here: https://social.msdn.microsoft.com/Forums/en-US/1bfe3bea-e1c4-48d1-ba74-8336efb5f8f1/problem-ordering-windows-10-iot-core-pro-from-arrow?forum=WindowsIoT

Basically you have to sign an NDA with Arrow to get the Qualcomm BSP. That thread has a phone number for a year ago and a contact email address at Arrow from about 6 months ago. Good luck.

parameshbabu, 2017-10-11:
I recommend to take a look at this page https://docs.microsoft.com/en-us/windows/iot-core/build-your-image/createbsps and see the link for Qualcomm to get the Qualcomm BSP. The Windows IoTCore Kits are available for download for public since march this year. IoTCore Pro is discontinued.

closed 2017-10-12 by EngineerLLC

SibeeshVenu, 2018-11-21:
I am also seeing the same with RPi2.  Below are the log files. 

**FIPPackage_oem**

`info: Trying to load file 'C:\OEEWorkspace\Build\arm\InputFMs\OEMFMFileList.xml' as a FM file list ...
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.OEM.Sample.cab
`
**FIPPackage_RPi2**

`info: Trying to load file 'C:\OEEWorkspace\Build\arm\InputFMs\RPi2FMFileList.xml' as a FM file list ...
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.Device.SystemInformation.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\SV.PlatExtensions.UpdateOS.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.bcm2836sdhc.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.rpisdhc.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.rpiq.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.Customization.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.bcm2836pwm.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.rpiwav.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.SerPL011.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.bcmgpio.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.bcmgpio.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.bcmi2c.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.bcmspi.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.bcmauxspi.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.vchiq.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.piminiuart.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.RPi.piminiuart.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.DeviceLayout.MBR4GB.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.DeviceLayout.MBR8GB-R.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.DeviceLayout.MBR4GB.cab
fatal error : Error: Missing package: C:\OEEWorkspace\Build\arm\pkgs\medialesson.DeviceLayout.MBR8GB-R.cab
`
I have followed almost all the documents available and couldn't fix this error. Can anyone please let me know what am I missing here.

SibeeshVenu, 2018-11-21:
> I am also seeing the same with RPi2. Below are the log files.
>
> **FIPPackage_oem**
>
> `info: Trying to load file 'C:\OEEWorkspace\Build\arm\InputFMs\OEMFMFileList.xml' as a FM file list ... fatal error [...]

Huh!. It was a silly mistake. I didn't run the `New-IoTCabPackage All ` (or) `buildpkg all ` after building the BSP (`Import-IoTBSP RPi2 C:\Downloads\RPi_BSP.zip` (or) `importbsp RPi2 C:\Downloads\RPi_BSP.zip`
`buildpkg RPi2`). 

After running the `buildpkg all ` I am able to see all the required cab files in the pkgs folder, thus no error.