ms-iot/iot-adk-addonkit #153: Test Image Crashes when not including IOT_TOOLKIT

eriksimonic, 2017-10-25:
Hi. 
I know this problem is not 100% for this project but. I guess that you have some knowledge about it.
I have narrowed a specific problem that I can't find documented.
When I build an image without IOT_TOOLKIT the image crashes With an error that a critical service did't start.

Here is my manifest.
```
 <Microsoft>
      <Feature>IOT_EFIESP</Feature>
      <Feature>IOT_EFIESP_BCD_MBR</Feature>
      <Feature>IOT_DMAP_DRIVER</Feature>
      <Feature>IOT_CP210x_MAKERDRIVER</Feature>
      <Feature>IOT_FTSER2K_MAKERDRIVER</Feature>
      <Feature>IOT_GENERIC_POP</Feature>
      <Feature>IOT_POWER_SETTINGS</Feature>
      <Feature>IOT_TOOLKIT</Feature>
      <Feature>IOT_UAP_OOBE</Feature>
      <Feature>IOT_APP_TOOLKIT</Feature>
      <Feature>IOT_POWERSHELL</Feature>
      <Feature>IOT_NETCMD</Feature>
      <Feature>IOT_SSH</Feature>
      <Feature>IOT_DISABLE_UMCI</Feature>
      <Feature>IOT_ENABLE_TESTSIGNING</Feature>
      <Feature>IOT_UMDFDBG_SETTINGS</Feature>
      <Feature>IOT_WDTF</Feature>
      <Feature>IOT_CRT140</Feature>
      <Feature>IOT_DIRECTX_TOOLS</Feature> -->
      <Feature>PRODUCTION_CORE</Feature>
    </Microsoft>
```
What I am missing here?
It is the IOT_TOOLKIT package required, or any other package is depending on it.
Thank you.

parameshbabu, 2017-11-01:
Which platform are you building this for? When you include the debug tools, it is always better to include the toolkit.

anphel31, 2017-11-07:
Hi,
Please let us know which device you are building an image for. We'll see if we can track down the issue.

eriksimonic, 2017-11-07:
Hi.
Thanks for the replay.
I am building for Raspberry pi 3, arm. 
The problem it seam somewhat wired. If I do restart the device after the crash, all installs as it should. and the crash in not always there.. 
So yes I am a bit confused here.
My latest manifest is like this.

 ```
 <Features>
    <Microsoft>
      <Feature>IOT_EFIESP</Feature>
      <Feature>IOT_EFIESP_BCD_MBR</Feature>
      <Feature>IOT_DMAP_DRIVER</Feature>
      <Feature>IOT_CP210x_MAKERDRIVER</Feature>
      <Feature>IOT_FTSER2K_MAKERDRIVER</Feature>
      <Feature>IOT_GENERIC_POP</Feature>
      <Feature>IOT_POWER_SETTINGS</Feature>
      
      <Feature>IOT_UAP_OOBE</Feature>
      <Feature>IOT_APP_TOOLKIT</Feature>
      
      <Feature>IOT_POWERSHELL</Feature>
      <Feature>IOT_NETCMD</Feature>
      <Feature>IOT_SSH</Feature>
      
      <Feature>IOT_DISABLE_UMCI</Feature>
      <Feature>IOT_ENABLE_TESTSIGNING</Feature>
      
      <Feature>PRODUCTION_CORE</Feature>
    </Microsoft>
    <OEM>
      
      <Feature>RPI2_DRIVERS</Feature>
      <Feature>RPI3_DRIVERS</Feature>
      
      
      <Feature>OEM_CustomCmd</Feature>
      <Feature>OEM_ProvAuto</Feature>
      <Feature>OEM_MyPackage</Feature>
    </OEM>
  </Features>
```

parameshbabu, 2018-03-04:
Toolkit package is not required as per your configuration. This looks good. I'm not sure what causes the crash on boot for the first time. Can you please try with the basic apps (IOT_BERTHA) and see if the issue is seen? Most likely it is the custom packages that you have added.

parameshbabu, 2018-05-14:
Closing this issue, please reopen a new issue if you are still facing this problem with the latest releases/tools.

closed 2018-05-14 by parameshbabu