ms-iot/iot-adk-addonkit #155: Missing Recovery.MBR-BcdEdit, etc.

mjmeans, 2017-10-30:
The repo includes MBR layouts for -R but no Recovery package(s) for MBR.

parameshbabu, 2017-11-01:
Recovery packages are same for both MBR/GPT. However there is so sample yet show-casing the recovery on the MBR layout.. it is in the backlog for now.

mjmeans, 2017-11-01:
I have copied the GPT recovery packages to MBR and changed GPT to MBR wherever it was found. I added it to the product and was able to buildimage and buildrecovery successfully. However, when I powershell into the device and run the recovery script, the RPi boots into an error message. I posted the error on the Windows IoT Forums with no response yet. I'm assuming that the cmd scripts from the GPT package won't work for MPR.

parameshbabu, 2017-11-01:
ok looks like there is more changes required in the scripts to handle the MBR partition guids. We will take a look a that and update.

mjmeans, 2017-11-12:
Are there any updates on this issue?

parameshbabu, 2018-04-25:
With #225 you should be able to perform recovery on rpi platform. Please use Recovery-MBR-BCD package instead of the GPT-BCD.

closed 2018-04-25 by parameshbabu