ms-iot/iot-adk-addonkit #157: Error for Common Custom package with product specific file

mjmeans, 2017-11-01:
So I need to include a oemcustomization.ps1 along with oemcustomization.cmd for one or more products, but not necessarily for all products. So I create a new Common package called Custom.Ps1 using Custom.Cmd as a model. I then add a section into the product specific area of OEMCommonFM.xml for it just like the section for Custom.Cmd.cab. I then add oemcustomization.ps1 file to ProductA and add the featureid OEM_CustomPs1 to ProductA\TestOEMInput.xml.

When I "buildpkg custom.ps1" I get an error about the missing oemcustomization.ps1 file in C:\IoT-ADK-AddonKit\Source-arm\Products\DigitalSign. ?? I never added the "OEM_CustomPs1" feature ID to that product. So package build fails.

I nee to build ProductA, so to get past the error message, I add an oemcusomization.ps1 file to DigitalSign (without adding the feature id to it's OEMInput files or many any changes to any XML file for DigitalSign). Then the "buildpkg custom.ps1" succeeds!

This doesn't make any sense. Is there a problem with the script, or my understanding of what is supposed to happen?

parameshbabu, 2017-11-01:
this is because the PRODUCT variable is not defined and so it defaults to the first available product, in this case DigitalSign. You can suppress this by explicitly defining the default product in setoem.cmd ( set PRODUCT=SampleA)

mjmeans, 2017-11-01:
That is very counterintuitive. Pkgs, especially common pkgs, should never be dependent on any product or products.

There is also a related problem. I add an empty oemcustomization.ps1 into DigitalSign to satisfy the error condition. I then add the oemcustomization.ps1 file that I want into ProductA. Then I "buildpkg all" and "buildimage ProductA Test" The resulting flash.ffu file has the wrong oemcustomization.ps1 file in it. i.e. After building ProductA, it has DigitalSign's version of oemcustomization.ps1.file. This is very bad! The package need to use the product local oemcustomization.ps1 file. So if I "buildimage ProductA Test" or "buildimage SampleA Test" the resulting flash.ffu file ABSOLUTLY MUST be the one defined in that product, exactly the same as oemcustomization.cmd file already is.

I want to include product specific file "oemcustomization.ps1" in exactly the same way that "oemcustomization.cmd" file is specified.

mjmeans, 2017-11-01:
If buildpkg needs the file as a stub in the resulting cab, then perhaps the process can be changed so that a zero length stub file can be included in the Common\* package folder. Then it should be replaced with the actual file when building the image.

parameshbabu, 2017-11-01:
Currently the product specific packages are rebuilt during the image creation, see createimage.cmd. The mechanism was not designed for easy extension as I thought this will be one off use. For your requirement now, you can add the buildpkg custom.ps1 similar to the custom.cmd there. I will look at a structural means to enable product specific packages as I see the need from users.

parameshbabu, 2018-03-04:
This is now fixed in the develop branch. Product specific packages are isolated in a ProdPackages directory under Common and is built only when building the image.

closed 2018-03-04 by parameshbabu