ms-iot/iot-adk-addonkit #159: Support for LattePanda and Up Boards

Untitled86, 2017-11-02:
I'm trying to use the LattePanda (similar to UP Boards "Core") which runs an Intel Atom 8350 CPU.

The vendor has provided a Win10 LTSB 2016 image, and that installs just fine.  

Shouldn't I be able to just create a provisioning package for IoT Core using ICD and then apply that to my device? 

If not, has anyone else had success with a compatible image?  This IoT Core image building process seems pretty horrible.

VladimirAkopyan, 2017-11-04:
Hey, I am currently in the weed with the same issue - I was trying to get the IoT core working on a tablet. 
Long story short, Intel has released drivers for BayTrail and Apollo lake CPUs. It;s possible to get other cpus working, but most likely some peripherals won't work. 
[I had a discussion with this stranger on Reddit who explained the process](https://www.reddit.com/r/windows10iot/comments/78bxbv/anyone_tried_intel_compute_stick/) a bit. 
So I think you can get Latte working with WIfi , but GPIO and GPU probably won't work. Same for Up board, but up Squared is basedon Apollo lake and they claim to be working on an image for it.

I is possible to get Windows IoT to work on many off-the-shelf devices if they are based on bay trail or apollo lake. I literally just tested a Linx1010 tablet, it works very well. [I wrote an article about it](https://blog.quickbird.uk/run-windows-iot-everywhere-1fc6943bcc23) and hope to pool together a community effort to test some devices and document which ones work. 

[image]
[Here is the github](https://github.com/VladimirAkopyan/WindowsIoTEverywhere), I will be release FFU images there.

albi90, 2017-11-24:
I have been playing with the lattepanda, the intel braswell bsp should give cherry trail support but the GFX drivers never load

albi90, 2017-11-25:
Ah looks like only x5-e8000 processors are supported not the Z series so looks like we are out of luck for Lattepanda support :(

Untitled86, 2017-11-27:
I'm confused.  I thought we could customize the drivers using the IoT ADK?  So why wouldn't the GPU be supported?  LattePanda publishes their drivers here...

https://github.com/LattePandaTeam/LattePanda-Win10-Software/tree/master/Drivers/2G32GB_Drivers/All-in-one%20Pack/CHT%20T3%20MRD%20PR2%20Driver%20WW51.2/Installer/PlatformInstaller/Drivers

What am I missing?

Untitled86, 2017-11-29:
:) I just built and deployed my first IoT Core FFU to the LattePanda.   Everything seems to be working.  Video, WiFi, etc.   I didn't have a touchscreen to try, but I am installing all of the Panda's drivers.  So I'll test that tomorrow.

My experience with this ADK was frustrating.  It seems like these kitchen type things are never quite perfect.  I ran into a lot of little problems along the way.  I'll upload my FFU, the documentation of my struggle and the BSP soon.

Any way we can add this BSP to the repo so some other guy doesn't have to go through this hell?

VladimirAkopyan, 2017-11-29:
please do share your work, and I agreee- this isn't the point every developer should have to go through again and again.

parameshbabu, 2017-11-29:
@Untitled86 Please do share your experience and the problems you encountered - we will review that and see how to address them.

Untitled86, 2017-12-01:
Ok, I have a 64bit and 32bit build working-ish now.  Seems like everything BUT the touchscreen :\  I'm working on resolving that now.  

The first thing is that the newdrvpkg command that turns an INF based driver package into a new wm.xml based driver package does NOT always work.  Often it includes things that cause collisions (I suggest we have a flag in the XML to ignore collisions like the old INFs had).

Second thing is that, for me, not all the drivers worked.  Including the Wifi driver from LattePanda caused a fatal error *every* time.  Wifi is working of course, just not with the LattePanda provided driver (seemingly one baked into IoTCore).

Last, aside from the touch-screen ignoring my fists, the display is scaling things at what seems like 300%.  Not stretching or drawing off-screen, it looks like the DPI/Font setting in Windows is wrong.  If I build the CustomMBMx64 package and deploy that (most things don't work because I don't have drivers) but the display appears perfectly at 100% scaling as expected.  So...I also need to resolve that issue.

I'll be working on this 8hrs a day until it's right.

Untitled86, 2017-12-01:
:) Just solved the DPI/Scaling issue.  ...I had to modify the graphics driver wm.xml AND inf file.  There were a bunch of OpenCL files that were throwing errors during BuildImage.  ...Hopefully none of those Intel OpenCL files were essential.  Everything seems to be working properly.

Now it's just that freakin' touch-screen drama...

Untitled86, 2017-12-01:
:) My life story.  It's 7:30pm.  The office is empty, I'm the only one here.  ...But it's ALL *finally* working!

1) Integrated Arduino (LattePanda has one of those that presents itself on a COM port)
2) USB
3) SD
4) Bluetooth
5) Graphics - Display Serial Interface (7" display attaches to Panda via ribbon)
6) Wifi
7) TOUCH SCREEN!!! :)  

...At the moment you have to manually copy a file into the Windows INF folder.  I'm going to fix that and upload my docs.

Is there any chance the FFU can be linked to somewhere on this repo?  I'll post it to the Panda form as well.  Hopefully host on OneDrive.  ...Any legal issues with distributing my FFU?

VladimirAkopyan, 2017-12-01:
Someone give this man a medal and publish the FFU as github release pelase!

albi90, 2017-12-02:
@Untitled86 would be great to get a gist howto on this I have been playing around for a few days trying to get the gpu drivers to load (just get stuck with the basic rendering driver)

Untitled86, 2017-12-02:
Ok, so I went ahead and build a BSP (board support package) for the LattePanda, built the IoT Core myself and I've posted the FFU here...

https://1drv.ms/u/s!AsDrOapsdMSUk_pNUL8PyKQ71FV5Jw

I presume it's OK to share these FFUs?  Let me know anyone if you know differently.

It supports everything, touch/wifi/etc.  All the drivers required were in the LattePanda github repo.

The nice thing about IoT core is that you can run it off USB.  The easiest way to try this out is to get the Windows IOT Core Image Helper.  Use that tool to flash these FFU onto a USB stick, plug it in and boot the device.  

If you want to deploy IoT Core to your device and run it off internal memory...

Note: These steps presume E: is your USB stick and C: is your hard drive.  You'll also need to download some tools from Microsoft (WDKs, etc) if you don't have them.  Use the Googles to find those.

1) Run "copype amd64 C:\WinPE_amd64"
This makes a folder and drops the Windows PE environment files in there.

2) Run "MakeWinPEMedia /UFD C:\WinPE_amd64 E:"
This puts the Windows PE environment on a USB stick.

3) Run "Copy C:\Program Files (x86)\Windows Kits\10\Assessment and Deployment Kit\Deployment Tools\arm64\DISM\* E:"
This copies the image deployment tool to the USB stick.

4) Run "ECHO dism -apply-image /imagefile:d:\flash.ffu /aaplydrive:\\.\PhysicalDrive0 > E:\ApplyFFU.bat"
This makes a batch file on the USB stick to apply the FFU.  DO NOT RUN THIS FILE ON YOUR PC!  Only on the Panda.

5) Place the USB stick in the LattePanda and boot it up.

6) When the Windows PE command line appears, run "ApplyFFU.bat"

Enjoy!  Making this was a real pain, hopefully nobody ever has to go through that again.

Added because Microsoft will see this...
It wasn't easy using those to build the new image though, a lot of things required manual changes.  I doubt most devs would do this on their own.  I did this because eventually I'm using the commercial ($39?) license of IoT core for a commercial product.  The LattePanda was the best board for my project so I went with that.  I underestimated this the challenge of creating a BSP.  Microsoft's documentation makes it sound like the IoTADK is perfect.  Their instructions walk you through fail-safe scenarios.

Untitled86, 2017-12-02:
@albi90 I will just add a public repo to my own account once I have this done.  ...I presume that's all fair and legal.  I've got a deadline I'm up against.  Having to make this BSP set me back (like I wrote above, I presumed it was as easy as Microsoft's instructions).  Give me a bit to upload the documentation.  I'm an imperfect perfectionist and I'd like more time polishing them up.

albi90, 2017-12-03:
@Untitled86 thanks for the FFU and the effort, I totally agree with you about the documentation making it sound like it just works I have found lots of missing things and inconsistencies but I also enjoy the challenge of learning something new :) will try your FFU tonight, and will keep trying to isolate why my gpu drivers won’t load, I must be missing something.

albi90, 2017-12-03:
@Untitled86 working perfectly :)

VladimirAkopyan, 2017-12-04:
If you write a tutorial that would be incredibly helpful - I was under impression that there is an issue with z8350 drivers and I'd lvoe to see if I can get Up boards and most tablets working!

Untitled86, 2017-12-04:
@VladimirAkopyan The tutorial is coming!  There were issues with the drivers.  I had to blow away some of Intel's references to WoW files in their INF, etc.

I'm still making improvements though.  I setup my partitions wrong.  Trying to setup "FastBoot" I'm running Powercfg and getting... "The system could not create the hibernation file.  The specific error code is 0xc000007f"  ...So I've changed it so Windows has nearly unlimited space and my app has 1.5GB (overkill).

FastBoot seems like it should be a prerequisite for IoT Core.  Especially since that's such a bas-ass Windows feature.  Any help on getting that working is appreciated.

Untitled86, 2017-12-04:
Oh, and in case anyone noticed, I didn't use the official Arduino serial port drivers.  I've fixed that.  My app needs to communicate to the on-board Arduino sometimes.  Next build has the fix.  ...Maybe it worked as was, didn't test properly.  The official driver is still a better bet.

albi90, 2017-12-04:
@VladimirAkopyan it’s actually not to bad once you know the process I have solved my issues (messed up the project file so the drives weren’t copying) and have a minix Z83-4 and a LattePanda working with two different BSPs I built where are you getting stuck?

Untitled86, 2017-12-04:
@albi90 The BSP is OK now (adding the official Arduino drivers worked fine) but I want to add FastBoot.  For that I need enough disk space to create a hibernate file and... Do you know?  I modified my DeviceLayout XML to create enough space and rebuilt my image.  Been busy working on my actual App so I haven't tried it out yet.

I love how fast my surfacebook boots.  Do you know if anyone has got FastBoot working on IoT core?

albi90, 2017-12-05:
@Untitled86 haven't had a chance to try fastboot yet but its on my list to :) 
(working on a digital signage solution from my work)

VladimirAkopyan, 2017-12-05:
@albi90 I seem to have trouble adding CPU drivers to the BSP. 
As for hosting FFUs - github releases are good for that. @Untitled86 I have tried uploading [your FFU here](https://github.com/VladimirAkopyan/WindowsIoTEverywhere/releases/tag/0.10) and github is perfectly happy with it. I'd love to collect FFUs and BSPs in one place from those that are happy to share them.

Untitled86, 2017-12-06:
@albi90 I will upload there.  I've also been accepted into TeamLattePanda's repo, so the source will go there.

albi90, 2017-12-07:
@Untitled86 have you noticed any issues with the GPU driver, I have an WinJS app that loads a video (mp4) and this crashes almost instantly when trying to play using the default GPU drivers, I managed to get the video playing smoothly by modifying the Braswell BSP GPU drivers to load for the cherry trail (there's just a few sections missing in the inf) but its not releasing the memory consumed then the video changes seems like there's a memory leak in the drivers when using the VPU

FireSale93, 2018-02-06:
Same problem here, I think the default output is set to the ribbon connector and not to HDMI, does anyone know how to change that in IoT Core? Thank you!

johnkattenhorn, 2018-05-28:
@Untitled86 - thanks for your work on this - I'm going to check this out later today

dlarbale74, 2018-09-09:
I have managed to get your FFU working but how do you code against the Arduino (specifically GPIO)?

Thanks in advance.

parameshbabu, 2018-11-16:
Closing as there is no outstanding action.

closed 2018-11-16 by parameshbabu

edtguy, 2019-01-08:
> Okay I managed the display issue by playing around with the BIOS settings and everything seems to work fine now over HDMI.
> @Untitled86: To go further into this could you please provide us your BSP files? That would be great!

@FireSale93: I am having the same black screen issue. Can you please share what BIOS settings you modified to get HDMI working? Thanks!

edtguy, 2019-01-08:
> @FireSale93: I am having the same black screen issue. Can you please share what BIOS settings you modified to get HDMI working? Thanks!

Nevermind. found it: BIOS->"Chipset"->"North Bridge"->"IGD - LCD Control"->"Force Lid Status"->"Off" .

aretecam, 2019-04-14:
Dear Sir,
I need the FFU 32bit please if you can upload it, Thanks in advance

Aisha1590, 2022-01-05:
> Ok, so I went ahead and build a BSP (board support package) for the LattePanda, built the IoT Core myself and I've posted the FFU here...
> 
> https://1drv.ms/u/s!AsDrOapsdMSUk_pNUL8PyKQ71FV5Jw
> 
> This is for the newer boards v1.2? that support 64bit EUFI. I have a 32bit build for the older boards. I presume that's not important. If anyone wants it I need to go back and add touch/graphics to that. I can, but I'd rather not. I made the mistake of starting to build my BSP for that board, had to start all over!

Hey from where did you get the schematic to prepare the bsp code for Lattepanda board. Also could you please share the code files.
Thanks in advance!

CarteKiwi, 2022-01-11:
HI everyone, 
This is closed but... is Windows 10 IoT Core supported by LP ?
I mean, @Untitled86 did a great job creating FFU but, It was in 2018... we are in 2022.
Since Windows 11 Iot Core is not out yet, I need to use W10 Iot Core instead of W10.
Thanks in advance, I feel like it's a bottle is the sea...