ms-iot/iot-adk-addonkit #163: Fails on Intel Braswell BSP

JeffMarqMetrix, 2017-11-06:
Trying to use Intel's latest BSP release for Braswell processors is giving me issues.

Here's where I obtained the BSP:
https://www.intel.com/content/www/us/en/embedded/products/braswell/software-and-drivers.html

BSP Download URL:
https://edc.intel.com/Link.aspx?id=14935

Using the 1709 Addon Kit downloaded from the master branch in GitHub results in the following errors:

[JetwaySBC1_Test.log]:
```
Creating JetwaySBC1 Test Image
Build Start Time : 13:47:45.22
Building Packages with product specific contents with version 10.0.0.0
 Processing Registry.Version.wm.xml
 Processing Custom.Cmd.wm.xml
 Processing JetwaySBC1
 Processing Provisioning.Auto.wm.xml
Running FeatureMerger for OEM packages
  Exporting OEMFM files
  Processing OEMFMFileList.xml
Running FeatureMerger for BSWx64
  Exporting BSWx64 FM files
  Processing BSWx64FMFileList.xml
Creating Image...
Microsoft (C) imageapp 10.0.10011.16384

ImageApp!ConfigureServicingStackLogging: Failed initializing WdsCore.dll.
ImageApp: Setting CPU Type to 'amd64'.
info: Imaging: Reading the OEM Input XML file...
info: FeatureAPI: Successfully validated the OEM Input XML: C:\IoT-ADK-AddonKit\Source-x64\Products\JetwaySBC1\TestOEMInput.xml
info: Imaging: Reading the FM XML file 'C:\IoT-ADK-AddonKit\Source-x64\BSP\BSWx64\Packages\BSWx64FM.xml'...
info: FeatureAPI: Successfully validated the Feature Manifest XML: C:\IoT-ADK-AddonKit\Source-x64\BSP\BSWx64\Packages\BSWx64FM.xml
fatal error : Imaging!ProcessCompDBPackages: FIP information in FM 'BSWx64FM.xml' packages for the following Features are incorrect: 
fatal error : 	Error : BASE
fatal error : 	Error : OEM_BSW64_DRIVERS
fatal error : 	Error : OEMDEVICEPLATFORM_BSWX64_DEVICE
fatal error : 	Error : DEVICELAYOUT_BSWX64_SOC
info: Imaging: See Flash.cbs.log and Flash.csi.log for details.
info: Imaging: Performance Results:
info: 	Total Run Time:	00:00:01.1726105
info: 	Storage Stack Time:	00:00:00
info: 	CompDB Total Time:	00:00:00.8819263
ImageApp: Error log can be found at 'C:\IoT-ADK-AddonKit\Build\amd64\JetwaySBC1\Test\Flash.ImageApp.log'.
Failed to add CBS file logger.
"CreateImage JetwaySBC1 Test" failed with error 1
```

I was then told that the version 3.2 of the AddonKit is what the current Intel BSP works against. And that there are changes checked into the develop branch that should address the issue.

For the version 3.2, buildpkg all doesn't even complete without errors. Here are the errors for buildpkg:

[Appx.IoTCoreDefaultApp.pkg.log]:
```
Creating Provisioning Package using C:\iot-adk-addonkit-3.2\Source-x64\Packages\Appx.IoTCoreDefaultApp\customizations.xml

Microsoft (C) Windows Imaging and Configuration Designer

ICD.exe : ERROR: The following errors were detected in customizations:
ICD.exe : ERROR: Settings were not found in the store file.
ICD.exe : ERROR: Parse error (0x80070490): 'ApplicationManagement' is not a valid child node for /
ICD.exe : ERROR: C:\Users\jscherrer\Documents\Windows Imaging and Configuration Designer (WICD)\Common\ICDCommon_20171106-131905-279_6636.log
"provpkg C:\iot-adk-addonkit-3.2\Source-x64\Packages\Appx.IoTCoreDefaultApp\customizations.xml" failed with error 100
```

[Appx.IoTOnboardingTask.prov.log]:
```
Creating Provisioning Package using C:\iot-adk-addonkit-3.2\Source-x64\Packages\Appx.IoTOnboardingTask\customizations.xml

[... 172 lines cut ...]
  Exporting BSWx64 FM files
  Processing BSWx64FMFileList.xml
Creating Image...
Microsoft (C) imageapp 10.0.10011.16384

ImageApp!ConfigureServicingStackLogging: Failed initializing WdsCore.dll.
ImageApp: Setting CPU Type to 'amd64'.
info: Imaging: Reading the OEM Input XML file...
info: FeatureAPI: Successfully validated the OEM Input XML: C:\Users\jscherrer\source\repos\iot-adk-addonkit\Source-x64\Products\JetwaySBC1\TestOEMInput.xml
info: Imaging: Reading the FM XML file 'C:\Users\jscherrer\source\repos\iot-adk-addonkit\Source-x64\BSP\BSWx64\Packages\BSWx64FM.xml'...
info: FeatureAPI: Successfully validated the Feature Manifest XML: C:\Users\jscherrer\source\repos\iot-adk-addonkit\Source-x64\BSP\BSWx64\Packages\BSWx64FM.xml
fatal error : Imaging!ProcessCompDBPackages: FIP information in FM 'BSWx64FM.xml' packages for the following Features are incorrect: 
fatal error : 	Error : BASE
fatal error : 	Error : OEM_BSW64_DRIVERS
fatal error : 	Error : OEMDEVICEPLATFORM_BSWX64_DEVICE
fatal error : 	Error : DEVICELAYOUT_BSWX64_SOC
info: Imaging: See Flash.cbs.log and Flash.csi.log for details.
info: Imaging: Performance Results:
info: 	Total Run Time:	00:00:01.1442189
info: 	Storage Stack Time:	00:00:00
info: 	CompDB Total Time:	00:00:00.8467266
ImageApp: Error log can be found at 'C:\Users\jscherrer\source\repos\iot-adk-addonkit\Build\amd64\JetwaySBC1\Test\Flash.ImageApp.log'.
Failed to add CBS file logger.
"CreateImage JetwaySBC1 Test" failed with error 1
```

parameshbabu, 2017-11-07:
Hi @JeffMarqMetrix 
For using the 15063_v3 you also need to use the corresponding ADK tools. Looks like you are using the latest ADK tools. I suggest you look at the outstanding PR changes and give it a try, if you want to stay with the latest tools and master branch. This is pending validation currently.

JeffMarqMetrix, 2017-11-07:
Hi @parameshbabu 
I did try the latest PR changes (I think). The very last log output was the result I got from that attempt.

I'm pretty sure I tested the right bits. I fetched everything from the "develop" branch. I will try again today, and let you know.

parameshbabu, 2017-11-07:
My bad, I should have mentioned that you need to run the script that is part for the PR , tools\bsptools\BSWx64\convert.cmd, which does the patching of the files to make it work with latest tools.

JeffMarqMetrix, 2017-11-07:
Hey @parameshbabu 
That could be the issue. But I'm not seeing that file in the develop branch. The only folder I see in bsptools is QCDB410C (I'm guessing for the DragonBoard). And there is no convert.cmd in there.

parameshbabu, 2017-11-07:
https://github.com/ms-iot/iot-adk-addonkit/pull/161/commits/62466ffd21bb6457a13ffff977413060ee1f63ed 
this is not integrated yet.

JeffMarqMetrix, 2017-11-07:
It builds the image successfully.

But I'm now having an issue when the OS starts. The screen goes blank. I think it's just because I didn't specify a default app to run, since I've been moving these XML files around quite a bit.

Will try to see if that's the issue and report back.

Thanks for your help!

JeffMarqMetrix, 2017-11-07:
Everything works!!! Thanks again!

closed 2017-11-07 by parameshbabu