ms-iot/iot-adk-addonkit #165: buildimage to specific version of IoT core

mjmeans, 2017-11-10:
I need to be able to specify which version of IoT core to use when building an image. This is necessary for archival purposes for product management so that I can check in the iot-adk-addonkit folder tree into the company source control without having to keep the build folder tree. At any later time, another worker should be able to check out the source tree and rebuild the exact same flash.ffu from source if I ever need to go back to a previous build, How can I do this?

Also, I would like to be able to build a flash.ffu based on the original creators update and I need to know how to do that. The purposes is so that I can test user and app interaction with an OS update process and user initiated installation time.

parameshbabu, 2017-11-10:
You should archive the ADK / OS Kits along with your iot-adk-addonkit folder tree to reproduce the same image again, as the ADK also evolves over time and you will need the matching ADK for the files you have in the iot-adk-addonkit.
By installing the required OS kits version, the iot-adk-addonkit will not need any further changes. 
Side note : You can set the MSPACKAGE flag to point to the directory containing the OS packages for a specific version.

mjmeans, 2017-11-12:
Thanks. I think this is a candidate for addition to the documentation for project and build management.

closed 2017-11-15 by parameshbabu