ms-iot/iot-adk-addonkit #166: Production build to RPi3 fails

mjmeans, 2017-11-12:
```
newproduct.cmd Prod
buildimage.cmd Prod
```
This is on a RPi3 with RPi touch screen. No customizations. This builds both test and retail with no reported errors. Test boots and runs fins. Retail doesn't. Upon first boot of retail I see the windows logo very briefly followed by a completely blue screen. After a few moments the blue screen turns black and the unit shuts down.

parameshbabu, 2017-11-12:
Retail image needs to have the bsp signed properly. Please see instructions at https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/build-retail-image. I assume you have them still test signed, the device doesn't boot.

mjmeans, 2017-11-12:
Does it require Standard of EV?

parameshbabu, 2017-11-12:
Standard will do, if you are not considering modifying the UEFI ( See https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-code-signing-certificate ).

mjmeans, 2017-11-12:
Thanks. EV is far too expensive a cost for one-off a prototype. With the Store cost for distribution, Azure costs for management, and a cert, I think all the costs together are a severe restriction for an independent developer/consultant developing and testing a prototype intended to be sold by another company.

mjmeans, 2017-11-12:
Under what conditions is a retail image (and cert) required? What are the limitations of a test image? Is it feasible to distribute a test image for a prototype run of a dozen units or so? It is feasible to distribute a test image for a limited production run (of 50-100 units) to a select group of customers for reliability testing?

EngineerLLC, 2017-11-15:
I'd really like to see a documented, more definitive answer to this question as well.

parameshbabu, 2018-11-16:
Please refer to the EULA included in the image.

closed 2018-11-16 by parameshbabu