ms-iot/iot-adk-addonkit #170: How to update AddonKit for existing projects

mjmeans, 2017-11-15:
Currently, updating the AddokKit for existing projects is tedious and error prone. This is because developer provided packages, bsps and products all exist within the same folder tree as AddonKit packages, bsps and products. So when there is a new ADK or new AddonKit version the developer is left with extracting a new copy of AddonKit and porting over all their specific items and changes.

If the AddonKit can be restructures so that AddonKit provided packages, bsps, and proucts are all in the same folder tree and developer created packages, bsps and products are in a separate folder tree then updating AddonKit becomes significantly simpler. And this would also enable the developer to manage their projects using GitHub and to pull updates from AddonKit directly into their projects while minimizing the possibility of merge conflicts.

parameshbabu, 2018-10-02:
@mjmeans new powershell version of the scripts support the workspace concept so that you can keep your workspace independent of the tools as you requested. Please try and give us feedback. thanks. Closing this issue.

closed 2018-10-02 by parameshbabu

mjmeans, 2018-10-02:
I will try it in the next round of builds, but my prototype stage is complete and I have to wait to see if it is accepted by my customer to go into production. When it gets to that point, I will be spending more time setting up the workspaces for a supportable production. Thanks.