ms-iot/iot-adk-addonkit #175: Issue compiling braswell BSP

albi90, 2017-11-20:
Hi, I'm using th elatest ADK and addon Kit with the intel braswell BSP x64 and after I run ".\bsptools\BSWx64\convert.cmd" and "buildbsp BSWx64" I get lots of errors about a universalbsp argument, any ideas what I'm doing wrong?

`Building all bsp packages
 Processing BSW64.GFX.wm.xml
Specified argument was out of the range of valid values.
Parameter name: Unknown argument "universalbsp" was provided.
Error : Failed to create package. See C:\iot-adk-addonkit-develop\Build\amd64\pkgs\logs\BSW64.GFX.wm.log`

parameshbabu, 2017-11-20:
this seems to be an installation issue. the pkggen.exe that is part of the ADK is proper, we reverified that. Can you please provide the following details?
- Which version of ADK/WDK was installed previously on the machine
- Install order of the new ADK/WDK ( was ADK installed first and WDK later or vice versa?)

We suspect an older version of the pkggen or one of the dependent dlls is present in your system

albi90, 2017-11-21:
Thanks parameshbabu, turns out I had two copies of the WDK installed removed he old one and repaired the new one and we are back in business.

closed 2017-11-21 by albi90