ms-iot/iot-adk-addonkit #179: REG_MULTI_SZ

brazierd, 2017-11-21:
Having trouble adding a multi value registry entry.  The documentation https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/package-schema has this example with a comma separator:

<regValue buildFilter="buildFilter2" name="name2" value="value1,value2" type="REG_MULTI_SZ" />

But that results in one value with a comma in it.  Same with semicolon, as used to work in the old pkg.xml format.  I will try \0 like reg.exe uses but one way or another the documentation doesn't seem to match the implementation.

brazierd, 2017-11-22:
\0 doesn't work either.  TO clarify, if I put this in the .wm.xml:

`<regKey
        keyName="$(hklm.software)\$(OEMNAME)\Test">
      <regValue
          name="MultiValue0"
          type="REG_MULTI_SZ"
          value="12AB\0CDEF" />
      <regValue
          name="MultiValueSemi"
          type="REG_MULTI_SZ"
          value="12AB;CDEF" />
      <regValue
          name="MultiValueComma"
          type="REG_MULTI_SZ"
          value="12AB,CDEF" />
    </regKey>`

Then the resulting values are all single strings, not multi values, using reg export shows:

`"MultiValue0"=hex(7):31,00,32,00,41,00,42,00,**5c,00,30,00**,43,00,44,00,45,00,46,\
  00,00,00,00,00
"MultiValueComma"=hex(7):31,00,32,00,41,00,42,00,**2c,00**,43,00,44,00,45,00,46,00,\
  00,00,00,00
"MultiValueSemi"=hex(7):31,00,32,00,41,00,42,00,**3b,00**,43,00,44,00,45,00,46,00,\
  00,00,00,00`

In each case, the separator is just appearing in the string, it is not replaces by a 00,00 marking the end of a string in a multi-value.

parameshbabu, 2017-11-22:
Can you try with this below? This should work..
`<regKey
        keyName="$(hklm.software)\$(OEMNAME)\Test">
      <regValue
      <regValue
          name="MultiValueComma"
          type="REG_MULTI_SZ"
          value="&quot;12AB&quot;,&quot;CDEF&quot;" />
    </regKey>`

closed 2018-03-04 by parameshbabu