ms-iot/iot-adk-addonkit #183: SMBIOS.CFG and commercialization

mjmeans, 2017-12-02:
Looking at the example projects that include an SMBIOS.CFG file, it appears that building an image hard codes SMBIOS data such as serial number into the flash.ffu. So when mass producing SD cards for production, all devices would have the same SMBIOS serial number unless buildimage.cmd is run once for each SD card being produced.

This is not appropriate for commercialization. When ordering SD cards from a mass production vendor that creates a big problem. The verbatim image copies would all have the same serial number unless they had the correct ADK version, correct IoT Core image, and correct ADK AddonKit with customizations to be able to run a modified buildimage.cmd that sets unique SMBIOS serial numbers for each and every SD card produced. No SD card production house would ever be expected to run all these tools, nor would any OEM want to provide a SD card mass production vendor with appx packages and image signing keys.

How can ADK AddonKit be used to create production images that support mass duplication and still conform to the commercialization license requirements?

imingc, 2018-11-16:
During mass production for a device, there needs to be a point where unique serial number is assigned to a device.  When this happens, the next step should to overwrite the serial number that's already burnt into the image with a new one.   Alternatively, don't include the serial number in the image, and provision the serial number on to the device at a later point in time.

mjmeans, 2018-11-17:
The mass production process should not involve editing an SMBIOS.CFG file in adk-addonkit and rebuilding the image for each SD card produced. It would not be a solution for mass production (1,000,000 units). The image needs to be created exactly ONCE and that image is sent to the mass production house.

From that point there are two options. Write the image to SD card with a script that modifies the already existing SMBIOS info, or have an image that can modify itself upon first boot. The former would be fine for a few dozen to a hundred or so SD cards, but not a good solution for several thousands of them.

There needs to be an automated way to set the SMBIOS data in an already existing image file, or image on SD card, or there needs to be a way to make the first boot of the image self modify the SMBIOS data to a unique serial number retrieved from MAC address or other silicon serial number on the device.

parameshbabu, 2018-11-26:
@mjmeans, SMBIOS.CFG is the mechanism for Qualcomm devices and is not the same for other platforms.  The SMBIOS configuration mechanism for RPi2 / Intel are pretty different.

mjmeans, 2018-11-26:
Thanks for the clarification. It's been so long since I reported this. But there will should be a way to be able to send a single image to a mass production house that will either be able to be quickly modified with unique serial numbers by the modification house (seconds of processing time for a duplicating script rather than minutes to recompile the entire image), or the image needs to self-differentiate upon first boot. Further, it would be a bad practice to send the entire adk-addonkit with all the separate custom packages to a foreign duplicating house that could accidentally modify your configuration, or even steal it and produce a competing product with trivial effort.

closed 2019-04-17 by saraclay