ms-iot/iot-adk-addonkit #184: Recovery doesn't work for Raspbery Pi

mjmeans, 2017-12-02:
I prepare a sample project using RPi2-R SOC and Device. Then buildimage and buildrecovey. write the recovery flash to the SD card. Boots fine. App runs. Debug works, etc. I then powershell into the device and cd MMOS. Then run startrecovery.cmd. Windows reboots and then Windows Boot Manager appears with a "Windows failed to start." message. Status code 0xc000000e. Info: A required device isn't connected or can't be accessed.

VitorX, 2018-03-28:
Hi mjmeans,
Have you fixed this issue? I also get this issue by following [this document](https://docs.microsoft.com/en-us/windows/iot-core/build-your-image/addrecovery)

mjmeans, 2018-03-28:
No I never did find a resolution. After spending several days trying I gave up. I finished the prototype without this feature and it was presented to my customer. My prototype was not approved for production development. Other boards were not an option because the cost per unit on those boards was too high in the quantities we envisioned during the first year of production. Since this feature for RPi3 was not available during the momentum of prototyping I doubt the RPi3 based project will ever go to production. It's too bad, because RPi3, Rpi 7" Touchscreen, SmartiPi case and cover, and WittyPi power board was a nice low cost combination and worked well. But the SDCard in this design is not accessible without complete disassembly; so it makes system restore functionality an imperative from a customer service standpoint. What I need is something as robust a set of features as an Android 7" tablet with a stand and USB, plus is low cost and runs Windows IoT. As of December, there were no options.

parameshbabu, 2018-04-25:
With #225 you should be able to perform recovery on rpi platform. Please use Recovery-MBR-BCD package instead of the GPT-BCD.

closed 2018-04-25 by parameshbabu