ms-iot/iot-adk-addonkit #187: oemcustomization and restart

mawett, 2017-12-20:
We currently have two scenarios that would require a restart after the first boot has finished.
In the oemcustimzation.cmd we call a powershell script that adds some registry entries for our touch driver (has to be done during runtime) and we read the hostname for the device from a file on the SD card. Both of these actions require a restart.
However, if I just trigger the restart after completion of the oemcustomization.cmd, I sometime see some issues with applications not being installed (e.g. screen keeps black after OOBE has finished, because default app has not been installed).
What is the proper way to trigger a restart, once all packages have been installed?

mawett, 2018-02-22:
Is there an update on this? How should we approach this issue?
Would it make sense to write a custom OOBE application that does this changes and then initializes a restart? Would that solve the problem with a restart during application installation?

anphel31, 2018-02-23:
One thing you can do is to add a provisioning package to your image that requires a reboot. This will cause the device to reboot a few minutes after provisioning completes. For example, setting the computer name with a provisioning package will trigger reboot. Does that work for your scenario?

Another option is to check that your apps are installed in the script before triggering the reboot. You can check using iotstartup.exe

mawett, 2018-02-23:
Unfortunately setting the computer name with the provisioning package won't work for our development process. We want to have the same unique name after every reflash of the OS image, so that a developer doesn't have to change the deployment target in VS every time they reflash the OS.

But thank you for the tip with the iotstartup.exe. I think that's currently the best option that I see for us, I'll give that a try.

closed 2018-02-23 by mawett