ms-iot/iot-adk-addonkit #189: Image Build Issue for DragonBoard 410C

LeonardBae, 2018-01-10:
I downloaded the Dragon BSP from Qualcomm site and tested it as shown in the link below.
(All versions, such as WDK, ADK, etc., are identical to 16299.)
https://docs.microsoft.com/en-us/windows/iot-core/build-your-image/createbsps

but, The image build error log is shown below.

{1/8/2018 7:58:23 PM}  Imaging: Reading the OEM Input XML file...
{1/8/2018 7:58:23 PM}  FeatureAPI: Successfully validated the OEM Input XML: C:\iot-adk-addonkit-master\Source-arm\Products\DragonBSP\RetailOEMInput.xml
{1/8/2018 7:58:23 PM}  Imaging: Reading the FM XML file 'C:\iot-adk-addonkit-master\Build\arm\MergedFMs\QCDB410CFM.xml'...
{1/8/2018 7:58:23 PM}  FeatureAPI: Successfully validated the Feature Manifest XML: C:\iot-adk-addonkit-master\Build\arm\MergedFMs\QCDB410CFM.xml
{1/8/2018 7:58:24 PM} Error: Imaging!ProcessCompDBPackages: FIP information in FM 'QCDB410CFM.xml' packages for the following Features are incorrect: 
	Error : BASE
	Error : OEM_QC_UEFI_PRODUCTION
	Error : OEM_SBC
	Error : OEM_CUSTOM_SMBIOS
	Error : OEMDEVICEPLATFORM_8016SBC
	Error : DEVICELAYOUT_QC8016
{1/8/2018 7:58:24 PM}  Imaging: See Flash.cbs.log and Flash.csi.log for details.
{1/8/2018 7:58:24 PM}  Imaging: Performance Results:
{1/8/2018 7:58:24 PM}  	Total Run Time:	00:00:01.8594682
{1/8/2018 7:58:24 PM}  	Storage Stack Time:	00:00:00
{1/8/2018 7:58:24 PM}  	CompDB Total Time:	00:00:01.4276662

parameshbabu, 2018-03-16:
This can be caused if you used the bsp cabs with earlier version of the kits. I recommend to use fresh copy of the bsp cabs and retry this process.

bibble235, 2018-03-23:
Do be more explicit, under each manufacturer the is a convert which needs to be run. Look at the one for I did at https://github.com/bibble235/iot-adk-addonkit. Happy to help more if required

parameshbabu, 2018-06-20:
Rebuild with new packages should solve this issue. Please reopen if you still have this issue. Closing.

closed 2018-06-20 by parameshbabu