ms-iot/iot-adk-addonkit #190: How to fix issue like "Could not load file or assembly"

EastGaint, 2018-01-10:
IOTADK_ROOT : D:\Tools\iot-adk-addonkit-3.2\iot-adk-addonkit-3.2
ADK_VERSION : 15063
WDK_VERSION : 15063
COREKIT_VER : 15063.0

When I try to use buildpkg.cmd to buid a cab file, an error like my topic is happened.
part of text like these:
info: Import Log: : >>>  [Unload Offline Registry Hive - SYSTEM]
info: Import Log: : >>>  Section start 2018/01/10 17:43:16.688
info: Import Log: :        os: Version = 10.0.16299  Service Pack = 0.0  Suite = 0x0100  ProductType = 1  Architecture = x86
info: Import Log: :      sto: Closed hive key '{bf1a281b-ad7b-4476-ac95-f47682990ce7}C:/Users/root/AppData/Local/Temp/y5pmaa0q.oit/windows/System32/config/SYSTEM'.
info: Import Log: :      sto: Unloaded hive key '{bf1a281b-ad7b-4476-ac95-f47682990ce7}C:/Users/root/AppData/Local/Temp/y5pmaa0q.oit/windows/System32/config/SYSTEM'. Time = 0 ms
info: Import Log: : <<<  Section end 2018/01/10 17:43:16.704
info: Import Log: : <<<  [Exit status: SUCCESS]
info: Import Log: :
info: Import Log: :
info: Import Log: : >>>  [Unload Offline Registry Hive - SOFTWARE]
info: Import Log: : >>>  Section start 2018/01/10 17:43:16.704
info: Import Log: :        os: Version = 10.0.16299  Service Pack = 0.0  Suite = 0x0100  ProductType = 1  Architecture = x86
info: Import Log: :      sto: Closed hive key '{bf1a281b-ad7b-4476-ac95-f47682990ce7}C:/Users/root/AppData/Local/Temp/y5pmaa0q.oit/windows/System32/config/SOFTWARE'.
info: Import Log: :      sto: Unloaded hive key '{bf1a281b-ad7b-4476-ac95-f47682990ce7}C:/Users/root/AppData/Local/Temp/y5pmaa0q.oit/windows/System32/config/SOFTWARE'. Time = 0 ms
info: Import Log: : <<<  Section end 2018/01/10 17:43:17.033
info: Import Log: : <<<  [Exit status: SUCCESS]
info: Import Log: :
info: Import Log: :
info: Import Log: : >>>  [Unload Offline Registry Hive - DRIVERS]
info: Import Log: : >>>  Section start 2018/01/10 17:43:17.033
info: Import Log: :        os: Version = 10.0.16299  Service Pack = 0.0  Suite = 0x0100  ProductType = 1  Architecture = x86
info: Import Log: :      sto: Closed hive key '{bf1a281b-ad7b-4476-ac95-f47682990ce7}C:/Users/root/AppData/Local/Temp/y5pmaa0q.oit/windows/System32/config/DRIVERS'.
info: Import Log: :      sto: Unloaded hive key '{bf1a281b-ad7b-4476-ac95-f47682990ce7}C:/Users/root/AppData/Local/Temp/y5pmaa0q.oit/windows/System32/config/DRIVERS'. Time = 0 ms
info: Import Log: : <<<  Section end 2018/01/10 17:43:17.080
info: Import Log: : <<<  [Exit status: SUCCESS]
info: Import Log: :
fatal error : Uncaught exception occured: System.IO.FileLoadException: Could not load file or assembly 'Microsoft.Phone.Test.TestMetadata  Version=8.1.1702.2001  Culture=neutral  PublicKeyToken=b3f029d4c9c2ec30' or one of its dependencies. The located assembly's manifest definition does not match the assembly reference. (Exception from HRESULT: 0x80131040)
fatal error : File name: 'Microsoft.Phone.Test.TestMetadata  Version=8.1.1702.2001  Culture=neutral  PublicKeyToken=b3f029d4c9c2ec30'
fatal error :    at Microsoft.WindowsPhone.ImageUpdate.PkgGenCommon.BaseComponents.Internal.PackageSetBuilder.BuildPackage(SatelliteId satelliteId  IEnumerable`1 files  String outputDir)
fatal error :    at Microsoft.WindowsPhone.ImageUpdate.PkgGenCommon.BaseComponents.Internal.PackageSetBuilder.Save(String outputDir)
fatal error :    at Microsoft.WindowsPhone.ImageUpdate.PkgGenCommon.BaseComponents.Internal.PackageGenerator.Build(String projPath  String outputDir  Boolean compress)
fatal error :    at Microsoft.WindowsPhone.ImageUpdate.PackageGenerator.Program.Main(String[] args)
fatal error : WRN: Assembly binding logging is turned OFF.
fatal error : To enable assembly bind failure logging  set the registry value [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD) to 1.
fatal error : Note: There is some performance penalty associated with assembly bind failure logging.
fatal error : To turn this feature off  remove the registry value [HKLM\Software\Microsoft\Fusion!EnableLog].

Error : Failed to create package. See D:\Tools\iot-adk-addonkit-3.2\iot-adk-addonkit-3.2\Build\arm\pkgs\logs\Drivers.oem11.pkg.log

IoTCoreShellv3.2 arm 10.0.0.0
Could you give me some help?

csabar, 2018-02-01:
I had the same issue, and I solved it by uninstalling the older version of WDK and repairing the current version.

parameshbabu, 2018-03-04:
Indeed ensuring that the correct versions of the ADK and WDK fixes these sporadic issues. Recommend to uninstall older versions always when moving to newer ADK versions.

closed 2018-03-04 by parameshbabu