ms-iot/iot-adk-addonkit #198: Can only build ffu once without rebooting

matelich, 2018-03-07:
When working on our build server, on the second build, I get the following error:

```
fatal error : StorageService::CreateVolumeList: VolumeTrackerList::PopulateList failed with error code 0x80070003.
fatal error : Imaging!CreateImage: Failed to create FFU: CreateImageStorageService failed: 80070003.
fatal error :   CreateImageStorageService failed: 80070003.
```

If I reboot, I can build again. Needless to say this is not ideal. I can't find anything about the names listed or the error code. Doesn't matter which one of my products I'm building.

matelich, 2018-03-07:
I guess this is an issue for ImageApp. Where do I find support for it?

parameshbabu, 2018-03-07:
it seems that the mountpoints after the imaging process is not cleanup properly. Is the DeviceNodeCleanup.x86.exe or DeviceNodeCleanup.x64.exe run after the Imageapp is run?

matelich, 2018-03-07:
I see it in the script. Of course, I'm now building my second ffu on this reboot without issue (after 3 prior recurrences of the problem).

I'll try to do the DeviceNodeCleanup if I run into this again. Thanks.

matelich, 2018-03-07:
Well, DeviceNodeCleanup didn't help. And I was wrong one the second ffu working. Just took longer to fail than I thought.

[miz21c_sf_test.log](https://github.com/ms-iot/iot-adk-addonkit/files/1790617/miz21c_sf_test.log)

matelich, 2018-03-07:
Oh, one other thing - I'm doing buildrecovery after buildimage.

matelich, 2018-03-08:
I do not see them in Disk Management, I am running as admin. It looks like the dism command worked. Got this output:

> Stale mount entry pointing to C:\Users\\...\AppData\Local\Temp\IMGE294.tmp\Data\USERS\System\AppData\Local\UpdateStagingRoot\UPDATEOS_Mount removed

Will see if the next one succeeds now, thanks!

matelich, 2018-03-08:
Sadly no. After the previous output, I still got this on my next build:

> info: ETW Log Path: C:\Users\\...\AppData\Local\Temp\storage_session_3154.etl
> info: OS Version: Microsoft Windows NT 10.0.15063.0
> info: CleanupAllMountedDisks: Cleaning up all mounted disks.
> info: InitializeVirtualHardDisk: Creating the VHD on volume C:\.
> info: InitializeVirtualHardDisk: C:\ ==> \\?\Volume{1c562498-8f65-11df-9034-806e6f6e6963}\.
> info: InitializeVirtualHardDisk: \\?\Volume{1c562498-8f65-11df-9034-806e6f6e6963}\ mount point: C:\.
> info: OpenDiskInternal: Creating empty virtual disk.
> info: OpenDiskInternal: Mounted virtual disk at phys \\?\PHYSICALDRIVE3
> info: InitializeVirtualHardDisk: Partitioning VHD.
> fatal error : PartitionVirtualHardDisk: VolumeTrackerList::WaitForVolume(EFIESP) failed with error code 0x80070003.
> fatal error : InitializeVirtualHardDisk: PartitionVirtualHardDisk failed with error code 0x80070003.
> fatal error : Imaging!CreateImage: Failed to create FFU: InitializeVirtualHardDisk failed with error code: 80070003
> fatal error :   InitializeVirtualHardDisk failed with error code: 80070003
> info: DismountVirtualHardDiskByFileName: dismounting image C:\Users\\...\AppData\Local\Temp\5502496494a84953958cf2038c91da9d.vhd

parameshbabu, 2018-03-08:
Can you give the details of the host OS you are running on.. version info as well. Thanks.

matelich, 2018-03-08:
OS Version	10.0.15063 Build 15063
Adksetup.exe that I used says it is version 10.1.16299.15
Addon kit currently at c30bbb622dc84695e5182986ccaa5c4a9f90fdba

IOTADK_ROOT : C:\systems\Miz21C\BSP
ADK_VERSION : 16299
WDK_VERSION : 16299
COREKIT_VER : 16299.15
OEM_NAME    : Zetec

Setting Environment for Architecture arm
Configuring for arm architecture
BSP_ARCH    : arm
BSP_VERSION : 10.0.0.0
BSPPKG_DIR  : C:\systems\Miz21C\BSP\bspcabs

matelich, 2018-03-14:
Do you think this would go away if I upgraded my build server to 16299?

parameshbabu, 2018-03-14:
Yes, sorry didnt notice your earlier message. It is always recommended to keep the host OS version and the ADK version in sync.

matelich, 2018-03-14:
Closing, assuming 16299 will fix the issue. Not sure when I'll get a chance to upgrade that server.

closed 2018-03-14 by matelich