ms-iot/iot-adk-addonkit #199: App was not updated but the package is updated

VitorX, 2018-03-08:
Here is my steps to update the apps:

1. build a basic image and add my first version app to the image( refer [this link](https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/deploy-your-app-with-a-standard-board#create-an-appx-package))

2. Create 2.0 version app and package via Visual Studio and then create the new package for the updated apps via the commands below
newappxpkg Hello_2.0.0.0.appx fga
buildpkg Appx.Hello 10.0.0.1
copy the the Product.Appx.Hello.cab under build/arm/pkgs to the C:\ of win 10 iot core(Rasp3)

3. Update the apps via command below through PowerShell connect to rasp3
cd C:\
applyupdate -stage Product.Appx.DMHelloWorld.cab
applyupdate -commit

Then the device is rebooted and I can see the gear appeared.  Start the app through device portal, it was no updated. Check the apps whether is installed successfully via the command **applyupdate -getinstalledpackages**. I can see the package is update to the version  10.0.0.1.

Did I misunderstood or made any mistake? Any help is appreciated.

parameshbabu, 2018-03-16:
Can you confirm if the Appx version was incremented when you did this process? If the appx version is not incremented then the appx will not be processed.

VitorX, 2018-03-19:
Hi parameshbabu,
The first version app should be 1.0.0.0. And the version of update app is 2.0.0.0. And the first package version is 10.0.0.

parameshbabu, 2018-06-20:
@VitorX do you still have this issue?

parameshbabu, 2018-11-16:
Closing this. IF the issue is still seen please report with further details. Thanks.

closed 2018-11-16 by parameshbabu