ms-iot/iot-adk-addonkit #202: App not being added to image

jkanounji, 2018-03-16:
created FFUs for BYT and APL families, following this tutorial : 
https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/deploy-your-app-with-a-standard-board

However, the app is not included in the FFUs.
In the case of the Minnowboard, nothing comes up on the screen, but the device is accessible through the portal. However, the app doesn't show up under the Apps tab. 
In the case of Upsquared , the boot up never gets past the splash screen "Please wait while we get things ready" and the device does not show up on IoT Dashboard.

I have tried both a custom app and the default Windows IOT app as in the tutorial.

parameshbabu, 2018-03-16:
This looks like related to your next issue - featureid not being properly processed. Please provide the version details of the tools and host OS to help further. Thanks

jkanounji, 2018-03-16:
as mentioned in the other Issue thread, all versions are 16299

parameshbabu, 2018-03-16:
Closing this as its tracked in the next issue 203

closed 2018-03-16 by parameshbabu