ms-iot/iot-adk-addonkit #203: Parsing is inconsistant and buggy (and confusing)

jkanounji, 2018-03-16:
there seems to be an inconsistency in the way FM files are handled. 
For example, in OEMFM.xml:
```
<PackageFile Path="%PKGBLD_DIR%" Name="%OEM_NAME%.Appx.IoTCoreDefaultApp.cab">
                <FeatureIDs>
                  <FeatureID>APP_IOTCOREDEFAULTAPP</FeatureID>
                  <FeatureID>OEM_IoTCoreDefaultApp</FeatureID> <!-- keeping old id for compatibility, dropping OEM prefix -->
                </FeatureIDs>
            </PackageFile>
```
in OEMCommonFM.xml:
      ```
<PackageFile Path="%PKGBLD_DIR%" Name="%OEM_NAME%.Custom.Cmd.cab">
        <FeatureIDs>
          <FeatureID>OEM_CustomCmd</FeatureID>
          <FeatureID>CUSTOM_CMD</FeatureID>
        </FeatureIDs>
      </PackageFile>
```

however, in the product's TestOEMInput.xml , if I use the correct ID , I get an error when trying to build the image. so if I have: 
```
	  <Feature>OEM_IoTCoreDefaultApp</Feature>
	  <Feature>OEM_CustomCmd</Feature>
```
I will get an error saying that the FM files do not contain information on OEM_OEM_IoTCoreDefaultApp and OEM_OEM_CustomCmd , which means, while building the image the tool is adding "OEM_" to the feature IDs it reads from its TestOEMInput.xml . to get around this, I included the packages' featureIDs as follows:
```
	  <Feature>IoTCoreDefaultApp</Feature>
	  <Feature>CustomCmd</Feature>
```
and this makes it get through building the image. 

Another problem I've run into is that I cannot build an image when including the OEMCommonFM.xml unless I comment or remove the base package part : 
```
  <BasePackages>
    <PackageFile Path="%PKGBLD_DIR%" Name="%OEM_NAME%.Registry.Version.cab"/>
  </BasePackages> 
```
leaving this in the Common FM file will cause building the image to exit with an error:
FIP information in FM 'OEMCommonFM.xml' packages for the following Features are incorrect:
Error : BASE

What do you suggest as a workaround for this other than removing the base package part?

parameshbabu, 2018-03-16:
Hi, can you provide the version info of the tools you use and also the host os version? Thanks.

jkanounji, 2018-03-16:
hello , 
host OS version: 10.0.16299 Build 16299
as for the tools, they're all 16299 as well, as is shown when the IoTCoreShell is launched:
IOTADK_ROOT : C:\iot-adk-addonkit-master
ADK_VERSION : 16299
WDK_VERSION : 16299
COREKIT_VER : 16299.15

jkanounji, 2018-03-16:
since it seems both my issues are related, you may go ahead and close the other one (about the app not being included in the final image)
worthy of noting is that this behavior does not happen when working with ARM architecture. I just build a RPi image, and I didn't omit the OEM_ from the feature IDs in TestOEMInput.xml , and the image built successfully:

```
<!-- Include OEM features -->
<Feature>OEM_CustomCmd</Feature>
<Feature>OEM_ProvAuto</Feature>
<Feature>OEM_IoTCoreDefaultApp</Feature>
<Feature>OEM_IoTOnboardingTask</Feature>
```

however, just like the upsquared, the boot hangs on "Please wait while we get things ready" , as if the IoTCoreDefaultApp is not there, and the device doesn't show up on the IoT Core Dashboard, probably meaning that the network service isn't running (and thus unable to open the device portal to check the apps). 

any suggestions?

jkanounji, 2018-03-16:
I just realized that in the case of x64 arch , the log files doesn't show that IoTCoreDefaultApp cab file was processed/added.
while in the RPi logs, there is mention of vnext.Appx.IoTCoreDefaultApp and vnext.OEM_OEM_IoTCoreDefaultApp.OEMFM.MainOS.
I have included both log files here (renamed with the arch in the file name)
[AMD64_Afif_test.log](https://github.com/ms-iot/iot-adk-addonkit/files/1820134/AMD64_Afif_test.log)
[ARM_3effo_test.log](https://github.com/ms-iot/iot-adk-addonkit/files/1820136/ARM_3effo_test.log)

parameshbabu, 2018-03-16:
Can you confirm that in the OEMInput.xml files, is the AdditionalFMs section includes MergedFMs\OEMFM.xml and MergedFMs\OEMCommon.xml ; and in the OEM section the feature IDs IoTCoreDefaultApp is included similar to the example for RPi?

jkanounji, 2018-03-16:
indeed. 
I will include the file here (extension renamed to txt to be able to attach)
[OEMInput.txt](https://github.com/ms-iot/iot-adk-addonkit/files/1820185/OEMInput.txt)

parameshbabu, 2018-03-16:
The Error is in the paths of the Additional FMs.. With the below corrected paths and also with proper feature id ( ie using OEM_IoTCoreDefaultApp), the build should be fine. The reason is that the FM files are processed in the build step and you have to include the processed FM in the OEMInput.
    <AdditionalFM>C:\iot-adk-addonkit-master\Build\amd64\MergedFMs\BYTx64FM.xml</AdditionalFM>
    <AdditionalFM>C:\iot-adk-addonkit-master\Build\amd64\MergedFMs\OEMCommonFM.xml</AdditionalFM>
    <AdditionalFM>C:\iot-adk-addonkit-master\Build\amd64\MergedFMs\OEMFM.xml</AdditionalFM>

jkanounji, 2018-03-16:
Thank you. 
Indeed, this made me able to build an image without errors, with the right featureid , and with the base package section there. 

I've noticed though that in the generated TestOEMInput.xml , the one for the Rpi2 had the right paths , I only needed to uncomment them. when working with x64 , the generated TestOEMInput.xml file contained the wrong paths which were commented, thus my scenario. 

That being said, even though the image builds fine, the startup still hangs at "Please wait while we get things ready"

parameshbabu, 2018-03-16:
You may want to try IOT_BERTHA feature added to the Microsoft section and remove OEM_IoTCoreDefaultApp - to isolate if this is an app issue. IOT_BERTHA will add the MS shipped version of the default app.

jkanounji, 2018-03-20:
thank you, this is working fine now .

parameshbabu, 2018-03-20:
Good to know. Closing this.

closed 2018-03-20 by parameshbabu