ms-iot/iot-adk-addonkit #204: (PkgToolBox) : error : SecurityToolBox::SignFile sign.cmd failed to resign

robp2175, 2018-03-16:
Can anyone help me figure out what might be going on. This is my first attempt at trying to create an arm driver. Thank you
```
inf2cab.cmd  e:\arm32\lvPRO5v.inf
 Authoring Drivers.lvPRO5v.wm.xml
 Processing Drivers.lvPRO5v.wm.xml
(PkgToolBox) : error : SecurityToolBox::SignFile sign.cmd failed to resign C:\Users\username\AppData\Local\Temp\3mkx3fv1.e1r\update.cat.
Err: 1
Output: signtool.exe sign /v /a /s my /i "Windows Phone Intermediate 2013" /n "Windows Phone OEM Test Cert 2013 (TEST ONLY)" /fd SHA256     "C:\Users\username\AppData\Local\Temp\3mkx3fv1.e1r\update.cat"
signtool.exe : fatal error : Signing failed with 1 on "C:\Users\username\AppData\Local\Temp\3mkx3fv1.e1r\update.cat"
Sign.Cmd RC=1

 Error : Failed to create package. See e:\arm32\Drivers.lvPRO5v.wm.log

```

```
Creating Drivers.lvPRO5v.wm.xml Package with version 10.0.0.0 for DigitalSign
Microsoft (C) PkgGen 10.0.10011.16384
Found Driver = C:\Users\username\AppData\Local\Temp\bquibzit.nyp\drivers\lvpro5v.inf
--Stack Trace--
   at Microsoft.Composition.ToolBox.Security.SecurityToolBox.SignFile(String path, Boolean testOnly)
   at Microsoft.Composition.Packaging.PkgManifest.GenerateAndSignCatalog(Boolean signCatalog)
   at Microsoft.Composition.Packaging.PkgManifest.SerializeMum(String outputFolder)
   at Microsoft.Composition.Packaging.PkgManifest.SaveManifest(String outputFolder)
   at Microsoft.Composition.Packaging.CbsPackage.SavePackage(String outputFolder)
   at Microsoft.Composition.Packaging.CbsPackage.SaveCab(String cabPath, CompressionType compressionType)
   at Microsoft.CompPlat.PkgBldr.CabGen.Assembly.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--
Package creation failed with error -1
```

parameshbabu, 2018-03-16:
Is the certs installed on the machine ( you can do that by calling "installoemcerts" at the prompt)

robp2175, 2018-03-16:
Well that was easy. Thanks @parameshbabu. I greatly appreciate it.

closed 2018-03-16 by robp2175

moyanming, 2018-04-20:
Hi @parameshbabu , I have the same problem but can't resolve after call again the "installoemcerts". I have finished the Lab 1a~1d successfully. But it seems that after I delete all the files in the %temp% folder I got this error. How to fix this problem?

parameshbabu, 2018-04-20:
Deleting files in %temp% will not influence this behavior. On the admin prompt, please install the certs and check again. If you get this error, most likely the cert is not installed properly.