ms-iot/iot-adk-addonkit #205: Add recovery for raspberry Pi 3

VitorX, 2018-03-28:
**Environment**
Raspberry Pi 3 B with Win 10 Iot Core 

**Target**
Add recovery mechanism 

**Steps**

1. Follow [Add a recovery mechanism to your image](https://docs.microsoft.com/en-us/windows/iot-core/build-your-image/addrecovery) to build the recovery FFU successfully(Flash_Recovery.ffu).

2. Flash this ffu to the card, the win 10 iot core also boot successfully. 

3. Start the recovery the OS by running the command **f:\startrecovery.cmd**  from device portal
[image]

**Result**
Not able to boot. It stayed on  **Windows Boot Manager**  like figure blew:
[image]

So here is my questions, any help info is appreciate:

1. How can I make the recovery mechanism works for Raspberry Pi 3 with Win 10 Iot Core 
2. Can I trigger it before boot to the normal OS by press the key?
3. Is there any document about how the recovery mechanism will be triggered automatly?

parameshbabu, 2018-03-30:
we will take a look at this and update

parameshbabu, 2018-04-25:
With #225 you should be able to perform recovery on rpi platform. Please use Recovery-MBR-BCD package instead of the GPT-BCD.

closed 2018-04-25 by parameshbabu

VitorX, 2018-04-25:
Thanks all for the help. I will test this issue and update whether it works:)

VitorX, 2018-04-30:
@parameshbabu , @anphel31 
Thanks for your help, it woks well now:)

alexiordan, 2018-10-03:
Why this functionality is not included in the standard image for Raspberry PI?