ms-iot/iot-adk-addonkit #210: Retail vs Test and

bibble235, 2018-04-09:
Hi,

I would like to make an image which supports 
1/ Secure Boot, 
2/ BitLocker, and 
3/ Device Guard

As per [https://docs.microsoft.com/en-us/windows/iot-core/secure-your-device/securebootandbitlocker](url)

I also want to use 

4/ the TPM 2.0 with the Device Provisioning Service
5/ containers

Currently 16299 it says the containers must only be enabled in the test images and not the retail ones. 

Questions
1/ What is the deal with retail vs test? Licensing? and 
2/ When will it be possible to do this?
3/ Preview 17127 says there are on custom builds, is this because it is in preview

Thanks,

saraclay, 2018-11-16:
#in-progress 

Hi @bibble235, with the Long-Term Servicing Channel (LTSC) released that came out, regarding Windows 10 IoT Containers, Windows 10 Containers are still for development and testing only. However, according to our license...

"You may only use Windows 10 Containers for commercial purposes and activities within Microsoft Azure IoT Edge. You may use any number of virtual OSEs instantiated as Windows 10 Containers by the Microsoft Azure IoT Edge runtime on the device."

Regarding retail vs test, you can learn more about that here:
https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/iot-core-manufacturing-guide 

Essentially, test images are great for quickly prototyping, accessing, and modifying devices.
Retail images, however, can be made more secure for public or corporate networks while receiving updates.

saraclay, 2018-11-26:
#please-close

closed 2018-11-26 by saraclay