ms-iot/iot-adk-addonkit #212: Build processes fail because of expired Windows Phone OEM Root 2013 (TEST ONLY) certificates

bderuijsscher, 2018-04-10:
The build processes fail to execute properly as of yet since apparently several of the certificates installed as per the instructions have expired since 4/9/2018.

What would be a proper workaround for this issue?

parameshbabu, 2018-04-10:
1709 ADK has the updated certs. Please install the ADK and run installoemcerts again.

Regards,
Paramesh

parameshbabu, 2018-04-10:
If you have installed ADK/WDK both, please make sure that you uninstall both ADK and WDK and then install the 1709 ADK. We have seen issues with older WDK and newer ADK as they share some common files and end up not updated at all. 
The file you need to check if all is ok is C:\Program Files (x86)\Windows Kits\10\tools\bin\i386\sign.cmd which should have references to 2017 certs and not 2013.

bderuijsscher, 2018-04-12:
Thank you, that solves the problem!