ms-iot/iot-adk-addonkit #218: Windows Phone OEM Test Cert Expired

mawett, 2018-04-12:
We currently can't build any Test OS Images, because the Windows Phone OEM Test Cert 2013 (TEST ONLY) has expired at the beginning of this week.
I tried to update to an insider preview version of the WDK and installed the new certificates. This solved the certificate issue, but now I got build errors during the image creation process (all packages were built successfully, but the ffu creation fails). It could be because the versions I had installed were incompatible.
I don't want to upgrade everything to a new versison yet.
What can we do to solve this issue? It's currently blocking many paths for us, since they all depend on new OS images.
See also the MSDN forum thread here: https://social.msdn.microsoft.com/Forums/en-US/5edf869e-4a9f-4c96-a315-e38114a25536/signing-error-while-creating-cab-files-for-arm-iot-core-using-inf2cabcmd?forum=WindowsIoT

parameshbabu, 2018-04-12:
Hi, you can leave the sign.cmd in the addon kit tools folder itself. The addon kit sign.cmd will be used always due to path precedence. This helps avoiding tampering with ADK folder. The upcoming release will have this fixed in ADK.

mawett, 2018-04-12:
Yeah that's what I thought, but I still got the error. I'll give it another try. Maybe I needed to close the cmd window first and then open a new instance, so the sign.cmd file gets loaded from the Tools directory.
Thanks for the hint.