ms-iot/iot-adk-addonkit #219: Retail signing of BSP keeps using Test Certificates

bderuijsscher, 2018-04-12:
Hi,

as mentioned on [this page](https://docs.microsoft.com/en-us/windows/iot-core/build-your-image/createbsps) the BSP files need to be retail signed when building a retail image. Despite setting this retail signing certificate correctly in the setsignature.cmd file (and which turns out to function properly in the buildpkg all command), the retail signing certificates are not picked up when running the build.cmd file in the BSP folder (for raspberry pi).

What would be the advice to achieve retail signed BSP files?

Thanks in advance.

parameshbabu, 2018-04-12:
You will have to call retailsign.cmd On to enable using the signature set in setsignature.cmd

Regards,
Paramesh

bderuijsscher, 2018-04-12:
Hi,

That is exactly what I did but that didn’t make any difference

parameshbabu, 2018-04-12:
Can you please confirm that the SIGNTOOL_OEM_SIGN is set as you expect in the env variables? This is what used for signing the files in the sign.cmd.

bderuijsscher, 2018-04-12:
Well, when running

buildpkg All 

it does seem to be signing the other packages using the correct certificate

bderuijsscher, 2018-04-13:
Here is an excerpt from the buildfm_oem.log file generated after running buildfm oem command.

Apparently this process involves first signing the cabs with a test certificate before signing them with a retail certificate? 

`info: Trying to load file 'C:\iot-adk-addonkit\Build\arm\InputFMs\OEMFMFileList.xml' as a FM file list ...
FeatureAPI: Successfully validated the Feature Manifest XML: C:\iot-adk-addonkit\Build\arm\InputFMs\OEMCommonFM.xml
info: Merging packages for feature 'BASE.OEMCOMMON'
info: Merging packages for feature 'OEM_Sec_LockDown.OEMCOMMON'
info: Merging packages for feature 'OEM_CUSTOM_CMD.OEMCOMMON'
info: Merging packages for feature 'OEM_OEMCommon_HotKey.OEMCOMMON'
info: Merging packages for feature 'OEM_CUSTOM_BCD.OEMCOMMON'
info: PkgCommonManaged: CreateCatalog: Signing catalog path C:\Users\MyUserName\AppData\Local\Temp\eehy5ti4.ko2\glamwqj1.z0a_content.cat with -testonly for package 'MyProductName.OEM_CUSTOM_BCD.OEMCOMMON.MainOS'.
info: Calling sign.cmd with options '-testonly' and file 'C:\Users\MyUserName\AppData\Local\Temp\eehy5ti4.ko2\glamwqj1.z0a_content.cat'.
info: PkgCommonManaged: CreateCatalog: Signing catalog path C:\Users\MyUserName\AppData\Local\Temp\pndufyll.sov\gvecd2pz.mf0_content.cat with -testonly for package 'MyProductName.OEM_Sec_LockDown.OEMCOMMON.MainOS'.
info: Calling sign.cmd with options '-testonly' and file 'C:\Users\MyUserName\AppData\Local\Temp\pndufyll.sov\gvecd2pz.mf0_content.cat'.
info: PkgCommonManaged: CreateCatalog: Signing catalog path C:\Users\MyUserName\AppData\Local\Temp\q3pazaf1.ni3\t1qbxosv.oay_content.cat with -testonly for package 'MyProductName.BASE.OEMCOMMON.MainOS'.
info: Calling sign.cmd with options '-testonly' and file 'C:\Users\MyUserName\AppData\Local\Temp\q3pazaf1.ni3\t1qbxosv.oay_content.cat'.
info: PkgCommonManaged: CreateCatalog: Signing catalog path C:\Users\MyUserName\AppData\Local\Temp\2oqdblzr.hpy\wf2wefkn.jcm_content.cat with -testonly for package 'MyProductName.OEM_CUSTOM_CMD.OEMCOMMON.MainOS'.
info: Calling sign.cmd with options '-testonly' and file 'C:\Users\MyUserName\AppData\Local\Temp\2oqdblzr.hpy\wf2wefkn.jcm_content.cat'.
info: PkgCommonManaged: CreateCatalog: Signing catalog path C:\Users\MyUserName\AppData\Local\Temp\qd1jn1cn.ehx\jfqdzait.qbn_content.cat with -testonly for package 'MyProductName.OEM_OEMCommon_HotKey.OEMCOMMON.MainOS'.
info: Calling sign.cmd with options '-testonly' and file 'C:\Users\MyUserName\AppData\Local\Temp\qd1jn1cn.ehx\jfqdzait.qbn_content.cat'.
info: Merging packages for feature 'OEM_Sec_SecureBoot.OEMCOMMON'
info: PkgCommonManaged: CreateCatalog: Signing catalog path C:\Users\MyUserName\AppData\Local\Temp\pqkmtfix.f1g\1a1hv3l0.u20_content.cat with -testonly for package 'MyProductName.OEM_Sec_SecureBoot.OEMCOMMON.MainOS'.
info: Calling sign.cmd with options '-testonly' and file 'C:\Users\MyUserName\AppData\Local\Temp\pqkmtfix.f1g\1a1hv3l0.u20_content.cat'.
signtool.exe sign /v /s my /i "DigiCert EV Code Signing CA (SHA2)" /n "MyCompanyName B.V." /ac "C:\iot-adk-addonkit\DigiCert High Assurance EV Root CA.crt" /fd SHA256  /t http://timestamp.verisign.com/scripts/timestamp.dll   "C:\Users\MyUserName\AppData\Local\Temp\q3pazaf1.ni3\t1qbxosv.oay_content.cat"
The following certificate was selected:
    Issued to: MyCompanyName B.V.

    Issued by: DigiCert EV Code Signing CA (SHA2)

    Expires:   Tue Mar 23 14:00:00 2021

    SHA1 hash: omitted

Cross certificate chain (using machine store):
    Issued to: Microsoft Code Verification Root

    Issued by: Microsoft Code Verification Root

    Expires:   Sat Nov 01 15:54:03 2025

    SHA1 hash: omitted

        Issued to: DigiCert High Assurance EV Root CA

        Issued by: Microsoft Code Verification Root

        Expires:   Thu Apr 15 21:55:33 2021

        SHA1 hash: omitted

            Issued to: DigiCert EV Code Signing CA (SHA2)

            Issued by: DigiCert High Assurance EV Root CA

            Expires:   Sun Apr 18 14:00:00 2027

            SHA1 hash: omitted

                Issued to: MyCompanyName B.V.

                Issued by: DigiCert EV Code Signing CA (SHA2)

                Expires:   Tue Mar 23 14:00:00 2021

                SHA1 hash: omitted

Done Adding Additional Store
Successfully signed: C:\Users\MyUserName\AppData\Local\Temp\q3pazaf1.ni3\t1qbxosv.oay_content.cat

[... 51 lines cut ...]

                Issued by: DigiCert EV Code Signing CA (SHA2)

                Expires:   Tue Mar 23 14:00:00 2021

                SHA1 hash: omitted

Done Adding Additional Store
Successfully signed: C:\Users\MyUserName\AppData\Local\Temp\eehy5ti4.ko2\glamwqj1.z0a_content.cat

Number of files successfully Signed: 1

Number of warnings: 0

Number of errors: 0

signed:  "C:\Users\MyUserName\AppData\Local\Temp\eehy5ti4.ko2\glamwqj1.z0a_content.cat"
Sign.Cmd RC=0
`

Anyway, it appears that this process also reverts to the expired 2013 test certificates as mentioned in [this issue](https://github.com/ms-iot/iot-adk-addonkit/issues/212)

How can we proceed to prevent this error? 

Many thanks in advance.

bderuijsscher, 2018-04-13:
I managed to build the BSP, packages and final retail image by manually replacing the references to the obsolete certificates in the sign.cmd from the ADK, however, this probably is not the most elegant solution.

parameshbabu, 2018-04-13:
You can use the sign.cmd added to the addonkit recently in the tools folder. This will be used due to the path precedence and you can leave the sign.cmd in the ADK folder intact.

The updated sign.cmd will also avoid signing the intermediate files with test cert. though that is not an issue as the cab files used are properly signed.

Regards,
Paramesh

parameshbabu, 2018-04-24:
Let me know if the above solution works for you so I can close this issue. Thanks

bderuijsscher, 2018-04-26:
Hi, 

after some testing this appears to be functioning properly. However, only after disabling integration of a Write Filter (which was initialized at first boot through oemcustomization.cmd) because apparently there is some initialization done or saved by the SystemConfigurator or any related service.

Regards

closed 2018-05-07 by parameshbabu