ms-iot/iot-adk-addonkit #224: Test certificates expired

IotCore-Developer, 2018-04-23:
buildfm oem fails because of expired test certificates.
The test certificates are expired so I can't sign my app anymore.

on the 9th of april 2018 they expired.

[image]

Errorlog:

signtool.exe sign /v /s my /i "Windows Phone Intermediate 2013" /n "Windows Phone OEM Test Cert 2013 (TEST ONLY)" /fd SHA256     "C:\iot-adk-addonkit-master\Build\arm\pkgs\TESTNAME.OEM_Sec_LockDown.OEMCOMMON.MainOS.spkg"
signtool.exe : fatal error : Signing failed with 1 on -pkg -testonly "C:\iot-adk-addonkit-master\Build\arm\pkgs\TESTNAME.OEM_Sec_LockDown.OEMCOMMON.MainOS.spkg"
Sign.Cmd RC=1

One or more errors occurred.
	Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PackageException: Failed to sign generated package: C:\iot-adk-addonkit-master\Build\arm\pkgs\TESTNAME.OEM_OEMCommon_HotKey.OEMCOMMON.MainOS.spkg ---> Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PackageException: Failed to sign file C:\iot-adk-addonkit-master\Build\arm\pkgs\TESTNAME.OEM_OEMCommon_HotKey.OEMCOMMON.MainOS.spkg with options -pkg -testonly, exit code 1
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PackageTools.SignFileWithOptions(String file, String options)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PkgFileSignerDefault.SignFileWithOptions(String fileToSign, String options)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PkgBuilder.SaveCab(String cabPath, CompressionType compressionType, PackageStyle outputStyle, SIGNING_HINT signingHint)
   --- End of inner exception stack trace ---
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PkgBuilder.SaveCab(String cabPath, CompressionType compressionType, PackageStyle outputStyle, SIGNING_HINT signingHint)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PkgBuilder.SaveCab(String cabPath, Boolean compress)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.MergeWorker.Merge(IPkgBuilder pkgBuilder, List`1 pkgs, String outputPkg, Boolean compress, Boolean incremental)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.MergeGroup.Merge(String outputDir, Boolean compress, Boolean incremental)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.FBMerger.Merge(IEnumerable`1 inputPkgs, String outputDir, Boolean compress, Boolean incremental)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.FBMerger.Merge(IEnumerable`1 inputPkgs, String featureKey, VersionInfo outputVersion, String ownerOverride, OwnerType ownerTypeOverride, ReleaseType expectedReleaseType, CpuId expectedCpuType, BuildType expectedBuildType, String outputDir, Boolean compress, Boolean incremental)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.Package.MergePackage(String[] inputPkgs, String outputDir, String featureKey, VersionInfo version, String ownerOverride, OwnerType ownerTypeOverride, ReleaseType releaseType, CpuId cpuType, BuildType buildType, Boolean compress, Boolean incremental)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.<>c__DisplayClass45_1.<MergeOneFM>b__13(String feature)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass31_0`2.<ForEachWorker>b__0(Int32 i)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass17_0`1.<ForWorker>b__1()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.InnerInvokeWithArg(Task childTask)
   at System.Threading.Tasks.Task.<>c__DisplayClass176_0.<ExecuteSelfReplicating>b__0(Object )
An unhandled exception was thrown: System.AggregateException: One or more errors occurred. ---> Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PackageException: Failed to sign generated package: C:\iot-adk-addonkit-master\Build\arm\pkgs\TESTNAME.OEM_OEMCommon_HotKey.OEMCOMMON.MainOS.spkg ---> Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PackageException: Failed to sign file C:\iot-adk-addonkit-master\Build\arm\pkgs\TESTNAME.OEM_OEMCommon_HotKey.OEMCOMMON.MainOS.spkg with options -pkg -testonly, exit code 1
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PackageTools.SignFileWithOptions(String file, String options)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PkgFileSignerDefault.SignFileWithOptions(String fileToSign, String options)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PkgBuilder.SaveCab(String cabPath, CompressionType compressionType, PackageStyle outputStyle, SIGNING_HINT signingHint)
   --- End of inner exception stack trace ---
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PkgBuilder.SaveCab(String cabPath, CompressionType compressionType, PackageStyle outputStyle, SIGNING_HINT signingHint)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PkgBuilder.SaveCab(String cabPath, Boolean compress)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.MergeWorker.Merge(IPkgBuilder pkgBuilder, List`1 pkgs, String outputPkg, Boolean compress, Boolean incremental)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.MergeGroup.Merge(String outputDir, Boolean compress, Boolean incremental)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.FBMerger.Merge(IEnumerable`1 inputPkgs, String outputDir, Boolean compress, Boolean incremental)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.FBMerger.Merge(IEnumerable`1 inputPkgs, String featureKey, VersionInfo outputVersion, String ownerOverride, OwnerType ownerTypeOverride, ReleaseType expectedReleaseType, CpuId expectedCpuType, BuildType expectedBuildType, String outputDir, Boolean compress, Boolean incremental)
   at Microsoft.WindowsPhone.ImageUpdate.PkgCommon.Package.MergePackage(String[] inputPkgs, String outputDir, String featureKey, VersionInfo version, String ownerOverride, OwnerType ownerTypeOverride, ReleaseType releaseType, CpuId cpuType, BuildType buildType, Boolean compress, Boolean incremental)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.<>c__DisplayClass45_1.<MergeOneFM>b__13(String feature)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass31_0`2.<ForEachWorker>b__0(Int32 i)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass17_0`1.<ForWorker>b__1()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.InnerInvokeWithArg(Task childTask)
   at System.Threading.Tasks.Task.<>c__DisplayClass176_0.<ExecuteSelfReplicating>b__0(Object )
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Task.Wait()
   at System.Threading.Tasks.Parallel.ForWorker[TLocal](Int32 fromInclusive, Int32 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal](IList`1 list, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEach[TSource](IEnumerable`1 source, Action`1 body)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.MergeOneFM(FMCollectionItem fmItem)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.MergeFeatures(String inputFile, String outputPackageDir, String outputFMDir, String versionStr, String inputFMDir, String critical, String packagePathReplacement, String variablesStr, String fmID, String ownerTypeStr, String ownerName, String languages, String resolutions, Boolean compress, Boolean convertToCBS, Boolean incremental, Boolean doPlatformManifest)
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.Main(String[] args)
---> (Inner Exception #0) Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PackageException: Failed to sign generated package: C:\iot-adk-addonkit-master\Build\arm\pkgs\TESTNAME.OEM_OEMCommon_HotKey.OEMCOMMON.MainOS.spkg ---> Microsoft.WindowsPhone.ImageUpdate.PkgCommon.PackageException: Failed to sign file C:\iot-adk-addonkit-master\Build\arm\pkgs\TESTNAME.OEM_OEMCommon_HotKey.OEMCOMMON.MainOS.spkg with options -pkg -testonly, exit code 1
[... cut ...]
   at Microsoft.WindowsPhone.ImageUpdate.FeatureMerger.FeatureMerger.<>c__DisplayClass45_1.<MergeOneFM>b__13(String feature)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass31_0`2.<ForEachWorker>b__0(Int32 i)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass17_0`1.<ForWorker>b__1()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.InnerInvokeWithArg(Task childTask)
   at System.Threading.Tasks.Task.<>c__DisplayClass176_0.<ExecuteSelfReplicating>b__0(Object )<---

IotCore-Developer, 2018-04-23:
Installing new certificates from

https://support.microsoft.com/en-in/help/4131991/update-for-windows-oem-hal-extension-test-cert-2017-test-only-test-cer

doesn't help, same error occurs.

parameshbabu, 2018-04-23:
This is fixed with an updated sign.cmd file in the tools dir. Pls check the latest master branch updates.

closed 2018-04-23 by parameshbabu

IotCore-Developer, 2018-04-24:
This partial the solution in the windows kits is also a sign.cmd file which
is used by buildfm OEM command. In there are also old certificates used....

> This is fixed with an updated sign.cmd file in the tools dir. Pls check
the latest master branch updates.
https://ci6.googleusercontent.com/proxy/52gsAAAUSH4JF-HsXyTwk4VvOZx3ymqtpE9NaudAEzhmiFXh8AuNWsaHVkKV_YokRmvBPskvdZCegsG7PxRgXeOpEiS-GJPmRndMFexVJGJWeguLMuB-dX0KfUTRGOrtdeFTRp9KQCribloKdp6_KdhYsV_Y8A=s0-d-e1-ft#https://github.com/notifications/beacon/AeYvV-P86LigokstCIlKarKb3Qyj9nGyks5trkBDgaJpZM4Tge4E.gif>