ms-iot/iot-adk-addonkit #228: Request to add

hollypang, 2018-05-01:
Hi,

We had the same build issue as posted here https://github.com/ms-iot/iot-adk-addonkit/issues/163, but the issue was on CHTx64.

Here is the convert.cmd (pls remove the .txt extension) for CHTx64.  I copied the convert.cmd from BSWx64 and modified the contents to work on CHTx64):

[convert.cmd.txt](https://github.com/ms-iot/iot-adk-addonkit/files/1964956/convert.cmd.txt)

Is it possible to put in a request to add tools\bsptools\CHTx64\convert.cmd to the iot-adk-addonkit?

Thanks,
Holly

hollypang, 2018-05-03:
Hi, I validated my convert.cmd works with the IOTG CHTx64 BSP shared [here in Artifactory:](https://ubit-artifactory-ba.intel.com/artifactory/webapp/#/artifacts/browse/tree/General/ped-bsw-local/WinIoTCore/Releases/Beta/ForGoldCandidate/Intel%20Atom%20Processor%20Windows%2010%20IoT%20Core%20BSP%20-%20CHT64Bits.msi)

hollypang, 2018-05-03:
Hi, I have validated my convert.cmd with IOTG's CHTx64 BSP (Intel Atom Processor Windows 10 IoT Core BSP - CHT64Bits.msi) shared in Artifactory [here](https://ubit-artifactory-ba.intel.com/artifactory/webapp/#/artifacts/browse/tree/General/ped-bsw-local/WinIoTCore/Releases/Beta/ForGoldCandidate/Intel%20Atom%20Processor%20Windows%2010%20IoT%20Core%20BSP%20-%20CHT64Bits.msi).

parameshbabu, 2018-05-07:
Fixed with PR #231

closed 2018-05-07 by parameshbabu