ms-iot/iot-adk-addonkit #229: BSWx64 doesn't work with Windows 1803 adk+wdk+packages

michalkovy, 2018-05-05:
I were able to create successfully BSWx64 images with 1709 adk+wdk+packages.

I have updated Windows to 1803, installed 1803 versions of Windows ADK and WDK and installed 1803 version of IoT Core 64bit packages. Then I have taken latest IoT Adk AddOnKit.

I am getting this error when I try to build bsp according to information on https://docs.microsoft.com/en-us/windows/iot-core/build-your-image/createbsps : 
Building BSWx64 bsp packages
 No .pkg.xml files found.
 Processing BSW64.GFX.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\amd64\pkgs\logs\BSW64.GFX.wm.log
 Processing BSW64.GPIO.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\amd64\pkgs\logs\BSW64.GPIO.wm.log
 Processing BSW64.I2C.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\amd64\pkgs\logs\BSW64.I2C.wm.log
 Processing BSW64.UART.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit-master\Build\amd64\pkgs\logs\BSW64.UART.wm.log
Running feature merger
Running FeatureMerger for BSWx64
  Exporting BSWx64 FM files
  Processing BSWx64FMFileList.xml
  Error: Featuremerger failed for BSWx64FMFileList.xml. See C:\iot-adk-addonkit-master\Build\amd64\buildfm_bsp_BSWx64.log

The file C:\iot-adk-addonkit-master\Build\amd64\pkgs\logs\BSW64.GFX.wm.log contains following text:
Creating BSW64.GFX.wm.xml Package with version 10.0.0.0 for MultiLangSample
Microsoft (C) PkgGen 10.0.10011.16384

--Stack Trace--
   at Microsoft.WindowsPhone.ImageUpdate.PkgGenCommon.InfFileConverter.DoConvert(String infFile, String[] references, String[] stagingSubdirs, String hiveRoot, String wimRoot, String targetPartition, CpuId cpu, String diffedHivesPath, String productName, String toolPaths)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.InfConverter.InstallDriver(List`1 driverCapabilityList, Boolean isUniveralBsp)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Driver.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement driverWmElement)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Drivers.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement fromWm)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Identity.ConvertEntries(XElement ToCsi, Dictionary`2 plugins, Config enviorn, XElement FromWm)
   at Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--
Package creation failed with error -1

parameshbabu, 2018-05-06:
Hi,
Can you confirm if you ran this cmd file to fix the BSP with the latest changes? https://github.com/ms-iot/iot-adk-addonkit/blob/master/Tools/bsptools/BSWx64/convert.cmd

michalkovy, 2018-05-06:
Yes, I have the same version and I have execute it.

zhyy2008z, 2018-05-14:
I have the same issue!

parameshbabu, 2018-05-17:
Can you please retry this on the latest tools update
master branch (5.1) and RS3 branch 16299 (4.4).

michalkovy, 2018-05-17:
I have just tested on 5.1 and I still see the same issue.

IotCore-Developer, 2018-05-17:
Still error:

Creating bcm2836pwm.wm.xml Package with version 10.0.0.0 for DigitalSign
Microsoft (C) PkgGen 10.0.10011.16384

--Stack Trace--
   at Microsoft.WindowsPhone.ImageUpdate.PkgGenCommon.InfFileConverter.DoConvert(String infFile, String[] references, String[] stagingSubdirs, String hiveRoot, String wimRoot, String targetPartition, CpuId cpu, String diffedHivesPath, String productName, String toolPaths)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.InfConverter.InstallDriver(List`1 driverCapabilityList, Boolean isUniveralBsp)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Driver.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement driverWmElement)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Drivers.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement fromWm)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Identity.ConvertEntries(XElement ToCsi, Dictionary`2 plugins, Config enviorn, XElement FromWm)
   at Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--
Package creation failed with error -1

anphel31, 2018-05-18:
michalkovy, what language is configured on your host machine?

michalkovy, 2018-05-18:
Windows display language is English (United States), Country or region is United States.  I have installed English (United States) and Czech language packs. Formats also looks like to be en-us defaults.

isdqwq, 2018-06-05:
i have the same issue.
get iot-adk-addonkit  ,and install cert  by `installoemcerts`
download apollo lake bsp https://docs.microsoft.com/zh-cn/windows/iot-core/build-your-image/createbsps
 and follow the web Introduce
[image]

`C:\iot-adk-addonkit\Tools>bsptools\APLx64\convert.cmd
Processing igdlh64.inf
 Authoring APL64.GFX.wm.xml
Processing iaiogpio.inf
 Authoring APL64.GPIO.wm.xml
Processing iaioi2c.inf
 Authoring APL64.I2C.wm.xml
Processing iaiospi.inf
 Authoring APL64.SPI.wm.xml
Processing txei.inf
 Authoring APL64.TXEI.wm.xml
Processing iaiouart.inf
 Authoring APL64.UART.wm.xml
 No .pkg.xml files found.
Fixing the BSPFM.xml file
Fixing the TestOEMInput.xml file
Fixing the RetailOEMInput.xml file
Conversions done

IoTCoreShellv5.1 amd64 10.0.0.0
C:\iot-adk-addonkit\Tools>buildbsp APLx64
Sign APLx64 bsp binaries...
Processing C:\iot-adk-addonkit\Source-x64\BSP\APLx64
 [sys files]
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.SystemInformation
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igdkmd64.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GPIO\iaiogpio.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.I2C\iaioi2c.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.SPI\iaiospi.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.TXEI\x64\TeeDriverW8x64.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.UART\iaiouart.sys
 [dll files]
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\iga32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\iga64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igc32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igc64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd10iumd32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd10iumd64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd11dxva32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd11dxva64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd12umd32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd12umd64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfx11cmrt32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfx11cmrt64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfxcmjit32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfxcmjit64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\libmfxhw32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\libmfxhw64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\mfx_mft_h264ve_32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\mfx_mft_h264ve_64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsClient.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsClientInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsProxy.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsProxyInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\libeay32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\ssleay32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsClient.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsClientInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsProxy.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsProxyInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\libeay32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\ssleay32.dll
Building APLx64 bsp packages
 Processing APL64.GFX.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.GFX.wm.log
 Processing APL64.GPIO.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.GPIO.wm.log
 Processing APL64.I2C.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.I2C.wm.log
 Processing APL64.SPI.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.SPI.wm.log
 Processing APL64.TXEI.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.TXEI.wm.log
 Processing APL64.UART.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.UART.wm.log
Running feature merger
Running FeatureMerger for APLx64
  Exporting APLx64 FM files
  Processing APLx64FMFileList.xml
  Error: Featuremerger failed for APLx64FMFileList.xml. See C:\iot-adk-addonkit\Build\amd64\buildfm_bsp_APLx64.log

IoTCoreShellv5.1 amd64 10.0.0.0
C:\iot-adk-addonkit\Tools>`

parameshbabu, 2018-06-14:
You can also just change the OS Language to English to get this working. Closing this issue.

parameshbabu, 2018-06-22:
"Settings" -> "Time and Language" -> "Region and Language" -> "Administrative Language Settings" -> "Copy settings..." -> Select "Welcome screen and system accounts" -> click "Ok" to reboot system.

Can you try this? See #239

michalkovy, 2018-06-22:
That works as a work-around, thank you. Is there an issue under which this localization bug will be fixed?

parameshbabu, 2018-06-22:
Please use the workaround for the 1803 ADK tools. The issue will be fixed in the next release of the ADK tools.

parameshbabu, 2018-10-04:
Please try with again with 1809 ADK tools.

parameshbabu, 2018-10-08:
Closing this issue as the new version of tools with fix available now.

closed 2018-10-08 by parameshbabu