ms-iot/iot-adk-addonkit #234: Buildimage fails for Dragonboard 410c on 1803

endyb66, 2018-05-10:
I've installed the required Tools

```
Windows ADK 1803
Windows WDK 1803
Windows10_IoTCore_Packages_ARM32_en-us_17134 (Windows10_IoT_Core_ARM_Packages.msi)
Latest iot-adk-addonkit-master
DB410c_BSP
```

When launching the IoTCoreShell.cmd i see the correct versions (I think):

```
IOTADK_ROOT : C:\iot-adk-addonkit-master
ADK_VERSION : 17134
WDK_VERSION : 17134
COREKIT_VER : 17134.1
```

I went ahead with  the following commands:

```
cd .\bsptools\QCDB410C
export C:\iot-adk-addonkit-master\DB410c_BSP
buildpkg all
newproduct MyProduct QCDB410C
buildimage MyProduct Test
```

Everything went fine until the buildimage command, where i get an error and it aborts the process. Here is the log

```
C:\iot-adk-addonkit-master\Build\arm\MyProduct_Test.log

Creating MyProduct Test Image
Build Start Time : 17:40:04.75
Building Packages with product specific contents with version 10.0.0.0
 Running QCDB410C specifics
 Processing Custom.SMBIOS.wm.xml
 No .pkg.xml files found.
 Processing Registry.Version.wm.xml
 No .pkg.xml files found.
 Processing Custom.Cmd.wm.xml
 Processing ProductA
 No .pkg.xml files found.
 Processing Provisioning.Auto.wm.xml
Running FeatureMerger for OEM packages
  Exporting OEMFM files
  Processing OEMFMFileList.xml
Running FeatureMerger for QCDB410C
  Exporting QCDB410C FM files
  Processing QCDB410CFMFileList.xml
Creating Image...
ImageApp: Setting CPU Type to 'arm'.
Microsoft (C) imageapp 10.0.10011.16384

ImageApp!ConfigureServicingStackLogging: Failed initializing WdsCore.dll.
"CreateImage MyProduct Test" failed with error -532462766
```

I've not much experience, so either i made something wrong or there is another issue.

anphel31, 2018-05-10:
Can you look in the "Flash.ImageApp.log" file, and post any errors you see?

endyb66, 2018-05-11:
I have no such log file, where should it be? I've tested with the develop branch of iot-adk-addonkit, there i see that no WDK Content root is defined

```
IOTADK_ROOT : C:\iot-adk-addonkit-develop
ADK_VERSION : 17134
WDK_VERSION : 17134
COREKIT_VER : 17134.1
```

IotCore-Developer, 2018-05-11:
Same here,

running imageapp.exe as administrator gives:

Running FeatureMerger for rpi2
  Exporting rpi2 FM files
  Processing rpi2FMFileList.xml
  [91mError: Featuremerger failed for rpi2FMFileList.xml. See C:\iot-adk-addonkit\Build\arm\buildfm_bsp_rpi2.log[0m
Creating Image...
ImageApp: Setting CPU Type to 'arm'.
Microsoft (C) imageapp 10.0.10011.16384

ImageApp!ConfigureServicingStackLogging: Failed initializing WdsCore.dll.
"CreateImage XXXXXXXXX test" failed with error -532462766

anphel31, 2018-05-11:
The log file should be in the same directory as your flash.ffu (C:\iot-adk-addonkit-develop\Build\arm\MyProduct\Test\Flash.ImageApp.log)
Can you post the exact build version of your host machine? (run "ver" from cmd prompt or "winver")

IotCore-Developer, 2018-05-11:
Microsoft Windows [Version 10.0.17134.48]

IotCore-Developer, 2018-05-11:
I think the problem has something todo with the fact I can't build the rpibsp as well.
Seen some problems with that also on 1803.

IotCore-Developer, 2018-05-11:
Creating bcm2836sdhc.wm.xml Package with version 10.0.0.0 for DigitalSign
Microsoft (C) PkgGen 10.0.10011.16384

--Stack Trace--
   at Microsoft.WindowsPhone.ImageUpdate.PkgGenCommon.InfFileConverter.DoConvert(String infFile, String[] references, String[] stagingSubdirs, String hiveRoot, String wimRoot, String targetPartition, CpuId cpu, String diffedHivesPath, String productName, String toolPaths)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.InfConverter.InstallDriver(List`1 driverCapabilityList, Boolean isUniveralBsp)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Driver.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement driverWmElement)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Drivers.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement fromWm)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Identity.ConvertEntries(XElement ToCsi, Dictionary`2 plugins, Config enviorn, XElement FromWm)
   at Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--
Package creation failed with error -1

endyb66, 2018-05-12:
Build of my host machine

Content of Flash.ImageApp.log

`Error: Please run imageapp.exe in an elevated console.`

endyb66, 2018-05-14:
After the changes a few minutes ago to the master branch, i don't get as much errors as described in the first post. But one error still exists and i don't know how to fix it or what i'm doing wrong:

`Error: Please run imageapp.exe in an elevated console.`

parameshbabu, 2018-05-17:
Can you please retry this on the latest tools update
master branch (5.1) and RS3 branch 16299 (4.4).

IotCore-Developer, 2018-05-17:
building of the bsp itself fails (e.g. rpibsp)

csevajol, 2018-05-17:
Hello. I have the same  error with Windows 1803 new dev PC

endyb66, 2018-05-18:
With everything based on 1709 (build 16299), including the host OS, i can build an Image without any problem. But with everything on 1803, i still get the issue with imageapp.exe. Does the language on the host OS matter on the 1803 toolset?

parameshbabu, 2018-05-18:
@endyb66 We are looking to root cause this issue.. so far we can repro this internally only when you dont have an administrator rights
`Error: Please run imageapp.exe in an elevated console.`
How do you launch the tool? Using IoTCoreShell.cmd or calling Launchtool.cmd directly in a cmd window?

endyb66, 2018-05-18:
Ok thank you for the effort! I'm launching the tool by starting IoTCoreShell.cmd, then Windows promts me  to run it as Administrator, then i hit yes. On this 1803 host machine, there is only one user available (Administrator), which is signed in by a Microsoft account. Current build is 17134.48 (Windows 10 Pro)

endyb66, 2018-05-20:
It seems to be an issue with the toolset. When running the host machine with 1709 (build 16299) and installed the 1803 toolset including the latest addonkit, i get the same error as i wrote above. So for conclusion

- Host machine 1709 (16299) with 16299 toolset -> works
- Host machine 1709 (16299) with 17134 toolset -> issue
- Host machine 1803 (17134) with 17134 toolset -> issue
- Host machine 1803 (17134) with 16299 toolset -> not yet tested

anphel31, 2018-05-23:
Does Flash.ImageApp.log contain anything else besides "Error: Please run imageapp.exe in an elevated console."? Or is that the entire contents of the file?

endyb66, 2018-05-23:
That is the entire content. In front of this error, there is just the timestamp, but that's all.

NadHr, 2018-05-28:
I'm having the same output while generating image for imx6.
the problem is that I didn't find IoTCoreShell.cmd  neither LaunchTool.cmd included in my project. Is there any solution to include them?

NadHr, 2018-05-28:
Is there any available link to download 16299 toolset kits?

endyb66, 2018-05-28:
You can download the addonkit from here. The ADK and WDK links are available on msdn site. But there is no IoTCore image for 16299 available anymore. So i hope the issue with the 1803 toolset is fixed soon.

mawett, 2018-05-28:
WDK: https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads
ADK: https://docs.microsoft.com/de-de/windows-hardware/get-started/adk-install
SDK: https://developer.microsoft.com/en-us/windows/downloads/sdk-archive

parameshbabu, 2018-05-28:
Hi @NadHr , I updated the link to Windows 10 IoT Core Release 1709 (version 10.0.16299.x) in the readme page. It is recommended to use the latest kits 1803 with 1803 tools.

NadHr, 2018-05-29:
@parameshbabu thanks for recommandation. But the 1803 kits and tools that I'm currently using didn't solve the problem of image windows 10 IotCore image generation for imx6.
I'll try to use the 1709 version as a proposition to solve the current problem

hbeberman, 2018-05-30:
Hi @NadHr, the 1709 version of IoT Core is unsupported in the i.MX6 project. The image may build but it will fail to boot because there are dependencies on the 1803 version of IoT Core.

The underlying issue may be the same, but the i.MX6 project does not depend on the IoT ADK AddonKit to build. Please open a new issue for your problem in our Project-Kayla repository and share the following information:

- Are you running Visual Studio as an Administrator?
- Have you completely uninstalled both the WDK and ADK at the same time, then reinstalled them? Sometimes upgrading them one at a time can cause issues with shared files.
- Sometimes a long filepath can cause issues, can you try checking out the repository into a folder at the root of `C:\`?
- Please attach the following file, it should show if all packages built successfully: `C:\Users\ASUS\Desktop\imx6\windows\project-kayla-develop\build\solution\iMXPlatform\GenerateTestFFU\fmlog.txt`

Thanks,
Henry

anphel31, 2018-05-31:
Hello @endyb66 - can you try to completely uninstall the 1803 WDK and ADK, and then reinstall both? Does this resolve your issue?

endyb66, 2018-06-01:
Thank you @anphel31 for the respnse, but unfortunately it does not rresolve my issue, still the same error output. 

Error: Please run imageapp.exe in an elevated console.

This really only happen with the 1803 tools.

NadHr, 2018-06-04:
@hbeberman Thanks for suggesting. But none of these propositions worked. You can find in the link below the following output/ log file content : 
https://stackoverflow.com/questions/50526901/generating-windows-iot-10-image-imageappconfigureservicingstacklogging-failed

endyb66, 2018-06-08:
Are there any news regarding the imageapp error on 1803 kit?

anphel31, 2018-06-13:
We've been able to reproduce this error after configuring a machine with non-EN-us localized version of Windows 10. The issue should be fixed in the next kits release, but in the meantime, here is one way to work around the issue. Open the Computer Management console (compmgmt.msc), select "Local Users and Groups" -> "Groups". Find the Administrators group (which would be localized to another language); rename this group to the English spelling: "Administrators", and then rerun the image creation steps. After image creation you can return the name back to the original spelling.

endyb66, 2018-06-13:
Thank you, will try that later today.

parameshbabu, 2018-06-13:
Closing this issue.

closed 2018-06-13 by parameshbabu

endyb66, 2018-06-24:
Still get the same error. ThreadId8716 ERROR: Please run imageapp.exe in an elevated console.

endyb66, 2018-09-16:
Please reopen this issue. On my machine, i use german language (de-CH), and for it to work i have to
- change the language to "English (United States)"
- Rename Administrator-Group from "Administratoren" to "Administrators"
- Copy language Settings ("Settings" -> "Time and Language" -> "Region and Language" -> "Administrative Language Settings" -> "Copy settings..." -> Select "Welcome screen and system accounts" -> click "Ok" to reboot System)

Otherwise i get the error `ThreadId8716 ERROR: Please run imageapp.exe in an elevated console`

parameshbabu, 2018-09-16:
Hi, 
this is a known issue in 1803 and will be fixed in the upcoming release. Please do use the workaround for 1803. 
thanks.

endyb66, 2018-09-16:
Oh ok, so you mean in the next major release of Windows 10?

parameshbabu, 2018-09-17:
Yes that's right.