ms-iot/iot-adk-addonkit #235: build image stuck in 88% every time

VitorX, 2018-05-14:
I am trying to build image for dragonboard 410c and Try using the latest add-on and 4.3. 
Here is the log:
info: [00:04:00] Staging progress: 82%
info: [00:04:01] Staging progress: 83%
info: [00:04:02] Staging progress: 84%
info: [00:04:03] Staging progress: 85%
info: [00:04:03] Staging progress: 86%
info: [00:04:15] Staging progress: 87%
info: [00:04:21] Staging progress: 88%
fatal error : [00:04:28] Operation failed finalizing changes: HRESULT_FROM_WIN32(ERROR_ADVANCED_INSTALLER_FAILED)
fatal error : [00:04:28] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::ProcessIndividualPackages, line 1250, Error  , Failed to finalize client, error is 0x80073713
info: [00:04:28] Closing session ImgUpd_UpdateOS_Commit
fatal error : [00:04:28] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::ProcessAllPackages, line 1375, Error  , Failed to process individual packages while Commit, error is 0x80073713
fatal error : [00:04:28] onecore\base\cbs\mobile\updatedll\lib\executeupdate.cpp, UpdateMain::ExecuteUpdateOS, line 1066, Error  , Failed to commit partition UpdateOS, error is 0x80073713
info: [00:04:28] Staging progress: 89%
info: [00:04:28] Cleaning staging root.
info: [00:04:30]  - UpdateOS dismounted.
info: [00:04:40]  - StagingRoot cleaned.
info: [00:04:40]  - Default StagingRoot cleaned.
warning : [00:04:40] onecore\base\cbs\mobile\updatedll\lib\prepareupdate.cpp, UpdateMain::PrepareUpdate, line 206, Warning  , The following message is ***NOT an error***, simply an estimate. See earlier in logs to find the real error.
info: [00:04:40] Staging is expected to fit: estimated to take 938983422 bytes, 106133995520 are free.
info: [00:04:40] UpdateDLL: Staging completed unsuccessfully
info: [00:04:40] hr = 0x80073713
info: [00:04:40] Output file written to: \\?\C:\Users\user1\source\repos\IoT\ms-iot\iot-adk-addonkit-4.3R\Build\arm\Temp\IMG28FC.tmp\Data\SharedData\DuShared\UpdateOutput.xml
info: [00:04:40] Staging failed at: 89% progress
info: [00:04:40] UpdateDLL: Performance Measures:
info: [00:04:40]        Total staging time: seconds = 280 (tickcount=280813)
fatal error : Imaging!UpdateImage: Failed call to UpdateDLL::PrepareUpdate with error code: -2147010797 (0x80073713)
info: DismountVirtualHardDiskByFileName: dismounting image C:\Users\user1\source\repos\IoT\ms-iot\iot-adk-addonkit-4.3R\Build\arm\Temp\f896babd3ab74f19aa369af78fa1f5b0.vhd
info: DismountFullFlashImage:[0.297] Cleaning up temporary paths.
info: CleanupTemporaryPaths: Cleaning up temporary path C:\Users\user1\source\repos\IoT\ms-iot\iot-adk-addonkit-4.3R\Build\arm\Temp\IMG28FC.tmp\.
info: Storage Service: Dismounting the image in 0.4 seconds.
info: Imaging: See Flash.cbs.log and Flash.csi.log for details.
info: Imaging: Performance Results:
info:   Total Run Time: 00:05:23.7627473
info:   Image Creation Time:    00:00:00.1024860
info:   Storage Stack Time:     00:00:22.0924686
info:   CompDB Total Time:      00:00:16.9639972
ImageApp: Error log can be found at 'C:\Users\user1\source\repos\IoT\ms-iot\iot-adk-addonkit-4.3R\Build\arm\DemoDevice\Test\Flash.ImageApp.log'.
"CreateImage DemoDevice Test" failed with error 1

endyb66, 2018-05-14:
How did you achieve that you can start building an image? As in my other issue postet, i'm getting errors right after starting the buildimage command. Can you please provide the commands you are executing and which software / packages you exactly installed?

hollypang, 2018-05-15:
Hi,

I am seeing the same issue that VitorX is seeing.  I just downloaded the latest 4.3 kit today.
However, the 14.2 kit that I downloaded on 04/27/2018 works and able to build the FFU image successfully.

I am running CHTx64 16299.15.170928-1543.RS3.  

The commands I issued is as follows:

**“%WPDKCONTENTROOT%\tools\bin\i386\InstallOEMCerts.cmd”** 

**set SIGNFILES=NONE**

**bsptools\CHTx64\convert.cmd** (by the way, this script failed, I had to manually insert the line below in line 39 of the script for it to execute successfully) 
               del /s /q %DST_DIR%\*._pkg.xml >nul 2>nul
**I created an [issue ](https://github.com/ms-iot/iot-adk-addonkit/issues/228#issuecomment-386168002)for this convert.cmd to be added, it was pushed to the repo, but this particular line was deleted somehow??  Can someone comment?**

**buildpkg all**

**buildbsp <BSPName>**

**newproduct <ProductName> <BSPName>** 

**buildimage <ProductName> Test**
After about 15 minutes of running, I get an error in my CHTx64_Product_Test.log
info: [00:12:04] Staging progress: 87%
info: [00:12:09] Staging progress: 88%
fatal error : [00:12:15] Operation failed finalizing changes: HRESULT_FROM_WIN32(ERROR_ADVANCED_INSTALLER_FAILED)
fatal error : [00:12:15] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::ProcessIndividualPackages, line 1250, Error  , Failed to finalize client, error is 0x80073713
info: [00:12:15] Closing session ImgUpd_updateos_Commit
fatal error : [00:12:17] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::ProcessAllPackages, line 1375, Error  , Failed to process individual packages while Commit, error is 0x80073713
fatal error : [00:12:17] onecore\base\cbs\mobile\updatedll\lib\executeupdate.cpp, UpdateMain::ExecuteUpdateOS, line 1066, Error  , Failed to commit partition updateos, error is 0x80073713
info: [00:12:17] Staging progress: 89%
info: [00:12:17] Cleaning staging root.
info: [00:12:20]  - UpdateOS dismounted.
info: [00:12:40]  - StagingRoot cleaned.
info: [00:12:40]  - Default StagingRoot cleaned.
warning : [00:12:41] onecore\base\cbs\mobile\updatedll\lib\prepareupdate.cpp, UpdateMain::PrepareUpdate, line 206, Warning  , The following message is ***NOT an error***, simply an estimate. See earlier in logs to find the real error.
info: [00:12:41] Staging is expected to fit: estimated to take 1208125029 bytes, 104682258432 are free.
info: [00:12:41] UpdateDLL: Staging completed unsuccessfully
info: [00:12:41] hr = 0x80073713
info: [00:12:41] Output file written to: \\?\C:\iot-adk-addonkit-master-4.3\iot-adk-addonkit-master\Build\amd64\Temp\IMGC543.tmp\Data\SharedData\DuShared\UpdateOutput.xml
info: [00:12:41] Staging failed at: 89% progress
info: [00:12:41] UpdateDLL: Performance Measures:
info: [00:12:41] 	Total staging time: seconds = 761 (tickcount=761110)
fatal error : Imaging!UpdateImage: Failed call to UpdateDLL::PrepareUpdate with error code: -2147010797 (0x80073713)
info: DismountVirtualHardDiskByFileName: dismounting image c:\iot-adk-addonkit-master-4.3\iot-adk-addonkit-master\Build\amd64\Temp\5330b9826962455b9dd09f58f423adf5.vhd
info: DismountFullFlashImage:[2.875] Cleaning up temporary paths.
info: CleanupTemporaryPaths: Cleaning up temporary path C:\iot-adk-addonkit-master-4.3\iot-adk-addonkit-master\Build\amd64\Temp\IMGC543.tmp\.
info: Storage Service: Dismounting the image in 8.6 seconds.
info: Imaging: See Flash.cbs.log and Flash.csi.log for details.
info: Imaging: Performance Results:

luyaok, 2018-05-16:
I get the same error with @VitorX when running this command: buildimage ProductA test. I build image for Raspberry Pi. Version: 16299.

**ProductA_test.log**

info: [00:01:22] Staging progress: 88%
fatal error : [00:01:26] Operation failed finalizing changes: HRESULT_FROM_WIN32(ERROR_ADVANCED_INSTALLER_FAILED)
fatal error : [00:01:26] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::ProcessIndividualPackages, line 1250, Error  , Failed to finalize client, error is 0x80073713
info: [00:01:26] Closing session ImgUpd_UpdateOS_Commit
fatal error : [00:01:26] onecore\base\cbs\mobile\iucore\iupartition.cpp, IUPartition::ProcessAllPackages, line 1375, Error  , Failed to process individual packages while Commit, error is 0x80073713
fatal error : [00:01:26] onecore\base\cbs\mobile\updatedll\lib\executeupdate.cpp, UpdateMain::ExecuteUpdateOS, line 1066, Error  , Failed to commit partition UpdateOS, error is 0x80073713
info: [00:01:26] Staging progress: 89%
info: [00:01:26] Cleaning staging root.
info: [00:01:28]  - UpdateOS dismounted.
info: [00:01:33]  - StagingRoot cleaned.
info: [00:01:33]  - Default StagingRoot cleaned.
warning : [00:01:33] onecore\base\cbs\mobile\updatedll\lib\prepareupdate.cpp, UpdateMain::PrepareUpdate, line 206, Warning  , The following message is ***NOT an error***, simply an estimate. See earlier in logs to find the real error.
info: [00:01:33] Staging is expected to fit: estimated to take 923076197 bytes, 105218375680 are free.
info: [00:01:33] UpdateDLL: Staging completed unsuccessfully
info: [00:01:33] hr = 0x80073713
info: [00:01:33] Output file written to: \\?\C:\Users\ritah\Downloads\iot-adk-addonkit-master\iot-adk-addonkit-master\Build\arm\Temp\IMG5AE4.tmp\Data\SharedData\DuShared\UpdateOutput.xml
info: [00:01:33] Staging failed at: 89% progress
info: [00:01:33] UpdateDLL: Performance Measures:
info: [00:01:33] 	Total staging time: seconds = 93 (tickcount=93282)
fatal error : Imaging!UpdateImage: Failed call to UpdateDLL::PrepareUpdate with error code: -2147010797 (0x80073713)
info: DismountVirtualHardDiskByFileName: dismounting image C:\Users\ritah\Downloads\iot-adk-addonkit-master\iot-adk-addonkit-master\Build\arm\Temp\f62088540d614e0da91c24619317a627.vhd
info: DismountFullFlashImage:[0.266] Cleaning up temporary paths.
info: CleanupTemporaryPaths: Cleaning up temporary path C:\Users\ritah\Downloads\iot-adk-addonkit-master\iot-adk-addonkit-master\Build\arm\Temp\IMG5AE4.tmp\.
info: Storage Service: Dismounting the image in 0.3 seconds.
info: Imaging: See Flash.cbs.log and Flash.csi.log for details.
info: Imaging: Performance Results:
info: 	Total Run Time:	00:02:37.7824965
info: 	Image Creation Time:	00:00:00.0382932
info: 	Storage Stack Time:	00:00:53.9858358
info: 	CompDB Total Time:	00:00:09.1402970
fatal error : 2018-05-16 13:42:07, Error      [0x018027] CSI    000001d6 (F) Failed execution of queue item Installer: Winsock App Permitted Lsp Categories ({70b162ba-b169-4546-9c51-7c495c652129}) with HRESULT efffffff [Error,Facility=(0fff),Code=65535 (0xffff)].  Failure will not be ignored: A rollback will be initiated after all the operations in the installer queue are completed; installer is reliable[gle=0x80004005]
fatal error : 2018-05-16 13:42:07, Error                 CSI    000001e8@2018/5/16:05:42:07.803 (F) onecore\base\wcp\plugins\winsockai\dll\namespace.cpp(119): Error HRESULT_FROM_WIN32(10107) originated in function Windows::WCP::WinsockAI::WinsockNamespaceInstaller::Install expression: GetLastError
fatal error : [gle=0x80004005]
fatal error : 2018-05-16 13:42:07, Error      [0x018028] CSI    000001e9 (F) Failed execution of queue item Installer: Winsock Namespace Online Install ({2e6b282b-eda7-4312-ac54-659046dd2496}) with HRESULT HRESULT_FROM_WIN32(10107).  Failure will not be ignored: A rollback will be initiated after all the operations in the installer queue are completed; installer is reliable[gle=0x80004005]
fatal error : 2018-05-16 13:42:07, Error                 CSI    000001fb@2018/5/16:05:42:07.810 (F) Winsock Transports AI: status = 11003; message = Failed to open winsock registry root
fatal error : [gle=0x80004005]
fatal error : 2018-05-16 13:42:07, Error                 CSI    000001fc@2018/5/16:05:42:07.810 (F) Winsock Transports AI: status = 4294967295; message = Failed to install provider; will try sanitizing the configuration for this provider before trying again; info: Tcpip
fatal error : [gle=0x80004005]
fatal error : 2018-05-16 13:42:07, Error                 CSI    000001fd@2018/5/16:05:42:07.810 (F) Winsock Transports AI: status = 8; message = MwcGetProtoArray failed because WSCEnumProtocols failed
fatal error : [gle=0x80004005]
fatal error : 2018-05-16 13:42:07, Error                 CSI    000001fe@2018/5/16:05:42:07.810 (F) Winsock Transports AI: status = 11003; message = Cannot sanitize provider in the native catalog; continuing anyway
fatal error : [gle=0x80004005]
fatal error : 2018-05-16 13:42:07, Error                 CSI    00000202@2018/5/16:05:42:07.810 (F) Winsock Transports AI: status = 11003; message = Failed to open winsock registry root
fatal error : [gle=0x80004005]
fatal error : 2018-05-16 13:42:07, Error                 CSI    00000203@2018/5/16:05:42:07.810 (F) Winsock Transports AI: status = 4294967295; message = 2nd attempt to install provider failed; info: Tcpip
fatal error : [gle=0x80004005]
[... cut ...]
fatal error : 2018-05-16 13:42:07, Error                 SXS    onecore\base\servicing\turbostack\lib\package.cpp(3054)\TurboStack.dll!07B98616: (caller: 0DA0B5B0) ReturnHr(1) tid(19e0) 80073713 An advanced installer failed during setup or servicing.
fatal error :     Msg:[onecore\base\servicing\turbostack\lib\install.cpp(915)\TurboStack.dll!07BA6B1B: (caller: 07B95537) Exception(1) tid(19e0) 80073713 An advanced installer failed during setup or servicing.
fatal error : ] 
ImageApp: Error log can be found at 'C:\Users\ritah\DOWNLO~1\IOT-AD~1\IOT-AD~1\Build\arm\ProductA\test\Flash.ImageApp.log'.
"CreateImage ProductA test" failed with error 1

parameshbabu, 2018-05-17:
Resolved this in master branch (5.1) and RS3 branch 16299 (4.4).

mawett, 2018-05-17:
Is it possible to build an 1803 IoT Core image on a host machine that has 1709 installed? Of course with the latest ADK, WDK and IoT Core Packages installed.
I tried it yesterday and had the same issues as described here. I rolled everything back to the old version now and am wondering if it's worth it to update again and try it again with the latest fix, or if it won't work anyway on 1709. Thanks

parameshbabu, 2018-05-17:
@mawett it is always recommended to keep the Host OS version and the ADK version same. So if you are running a 1709 Host OS, it is better to use the 1709 ADK. Thanks.

closed 2018-05-17 by parameshbabu