ms-iot/iot-adk-addonkit #239: apollo lake bsp build issue

isdqwq, 2018-06-05:
os win10 x64 cn

install all software except wdk(had install, verison 10.0.16299.15) follow https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/set-up-your-pc-to-customize-iot-core

install intel apollo lake bsp follow https://docs.microsoft.com/zh-cn/windows/iot-core/build-your-image/createbsps
[image]

copy C:\Program Files (x86)\Intel IoT\Source-x64 to C:\iot-adk-addonkit
and double click C:\iot-adk-addonkit\IoTCoreShell-x64.cmd

Install certificates
`installoemcerts`

follow bsp command 
[image]

get below error:

`C:\iot-adk-addonkit\Tools>bsptools\APLx64\convert.cmd
Processing igdlh64.inf
 Authoring APL64.GFX.wm.xml
Processing iaiogpio.inf
 Authoring APL64.GPIO.wm.xml
Processing iaioi2c.inf
 Authoring APL64.I2C.wm.xml
Processing iaiospi.inf
 Authoring APL64.SPI.wm.xml
Processing txei.inf
 Authoring APL64.TXEI.wm.xml
Processing iaiouart.inf
 Authoring APL64.UART.wm.xml
 No .pkg.xml files found.
Fixing the BSPFM.xml file
Fixing the TestOEMInput.xml file
Fixing the RetailOEMInput.xml file
Conversions done

IoTCoreShellv5.1 amd64 10.0.0.0
C:\iot-adk-addonkit\Tools>buildbsp APLx64
Sign APLx64 bsp binaries...
Processing C:\iot-adk-addonkit\Source-x64\BSP\APLx64
 [sys files]
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.SystemInformation
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igdkmd64.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GPIO\iaiogpio.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.I2C\iaioi2c.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.SPI\iaiospi.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.TXEI\x64\TeeDriverW8x64.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.UART\iaiouart.sys
 [dll files]
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\iga32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\iga64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igc32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igc64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd10iumd32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd10iumd64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd11dxva32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd11dxva64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd12umd32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd12umd64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfx11cmrt32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfx11cmrt64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfxcmjit32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfxcmjit64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\libmfxhw32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\libmfxhw64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\mfx_mft_h264ve_32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\mfx_mft_h264ve_64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsClient.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsClientInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsProxy.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsProxyInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\libeay32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\ssleay32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsClient.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsClientInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsProxy.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsProxyInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\libeay32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\ssleay32.dll
Building APLx64 bsp packages
 Processing APL64.GFX.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.GFX.wm.log
 Processing APL64.GPIO.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.GPIO.wm.log
 Processing APL64.I2C.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.I2C.wm.log
 Processing APL64.SPI.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.SPI.wm.log
 Processing APL64.TXEI.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.TXEI.wm.log
 Processing APL64.UART.wm.xml
(PkgGenCommon) : error : Failed DoConvert
Error : Failed to create package. See C:\iot-adk-addonkit\Build\amd64\pkgs\logs\APL64.UART.wm.log
Running feature merger
Running FeatureMerger for APLx64
  Exporting APLx64 FM files
  Processing APLx64FMFileList.xml
  Error: Featuremerger failed for APLx64FMFileList.xml. See C:\iot-adk-addonkit\Build\amd64\buildfm_bsp_APLx64.log
`

anphel31, 2018-06-07:
This appears to be the same issue as #229 
Does the error reproduce after changing the system language to en-US temporarily and rebuilding? (I wonder if the error might have something to do with the language)

isdqwq, 2018-06-08:
@anphel31 change os language to EN ,have the same issue.

[image]

isdqwq, 2018-06-11:
@anphel31 
`Creating APL64.GFX.wm.xml Package with version 10.0.0.0 for MultiLangSample
Microsoft (C) PkgGen 10.0.10011.16384

--Stack Trace--
   at Microsoft.WindowsPhone.ImageUpdate.PkgGenCommon.InfFileConverter.DoConvert(String infFile, String[] references, String[] stagingSubdirs, String hiveRoot, String wimRoot, String targetPartition, CpuId cpu, String diffedHivesPath, String productName, String toolPaths)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.InfConverter.InstallDriver(List`1 driverCapabilityList, Boolean isUniveralBsp)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Driver.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement driverWmElement)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Drivers.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement fromWm)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Identity.ConvertEntries(XElement ToCsi, Dictionary`2 plugins, Config enviorn, XElement FromWm)
   at Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--
Package creation failed with error -1`

isdqwq, 2018-06-11:
my pc had install   software list.
[image]
[image]
[image]
[image]
[image]

parameshbabu, 2018-06-11:
Do you have Windows 10 ADK installed in your machine? I didnt see that in the list here?

isdqwq, 2018-06-12:
@parameshbabu  are you said this tools ? 
this work is follow https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/set-up-your-pc-to-customize-iot-core#span-idsoftwarespanspan-idsoftwarespanspan-idsoftwarespansoftware
[image]

anphel31, 2018-06-13:
Can you try the following workaround, and see if this allows you to build the BSP?

"Open the Computer Management console (compmgmt.msc), select "Local Users and Groups" -> "Groups". Find the Administrators group (which would be localized to another language); rename this group to the English spelling: "Administrators", and then rerun the image creation steps. After image creation you can return the name back to the original spelling."

isdqwq, 2018-06-13:
@anphel31  when change os language to english and the administrator groups name default been "Administrators"
[image]

parameshbabu, 2018-06-13:
Closing this issue.

isdqwq, 2018-06-15:
@parameshbabu  why close this issue. i can't build bsp  .

parameshbabu, 2018-06-15:
reopening - looks like this is a different issue than the administrator mismatch.

isdqwq, 2018-06-19:
@parameshbabu  what???   how to fix this issue.
administrator groups default is administrators ,whether it is Chinese or English.
my issue is how to fix this issue.   I can't build.

isdqwq, 2018-06-19:
@parameshbabu  thanks, it's good news for me.

but do you know my said?
win10 x64 CN OS , the administrator groups name is "Administrators", and change language to EN, the groups name is "Administrators", and rebuild bsp failed

bderuijsscher, 2018-06-19:
Hi, I have the exact same problem. I have changed the system language to English (US), the Administrators group is appropriatly named (just as @isdqwq  mentions), however the buildpkg command (called in the build.cmd) fails.

Just for the sake of completeness the following errors are generated for each wm.xml file processed:

```
Creating rpiwav.wm.xml Package with version 10.0.0.0 for DigitalSign
Microsoft (C) PkgGen 10.0.10011.16384

--Stack Trace--
   at Microsoft.WindowsPhone.ImageUpdate.PkgGenCommon.InfFileConverter.DoConvert(String infFile, String[] references, String[] stagingSubdirs, String hiveRoot, String wimRoot, String targetPartition, CpuId cpu, String diffedHivesPath, String productName, String toolPaths)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.InfConverter.InstallDriver(List`1 driverCapabilityList, Boolean isUniveralBsp)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Driver.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement driverWmElement)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Drivers.ConvertEntries(XElement toCsi, Dictionary`2 plugins, Config environ, XElement fromWm)
   at Microsoft.CompPlat.PkgBldr.Base.PkgPlugin.ConvertEntries(XElement parent, Dictionary`2 plugins, Config enviorn, XElement component)
   at Microsoft.CompPlat.PkgBldr.Plugins.WmToCsi.Identity.ConvertEntries(XElement ToCsi, Dictionary`2 plugins, Config enviorn, XElement FromWm)
   at Microsoft.CompPlat.PkgBldr.Program.BuildPackage(Config config, MacroResolver commandLineMacros)
   at Microsoft.CompPlat.PkgBldr.Program.Main(String[] args)
--End Stack Trace--
Package creation failed with error -1
```

bderuijsscher, 2018-06-19:
Extra information: running the following diagnostics on the language set reveals:

```
C:\BSP>wmic os get locale, oslanguage, codeset
CodeSet  Locale  OSLanguage
1252     0409    1043
```
I suspect that the OSLanguage (Dutch in my case) should also be modified somehow?

bderuijsscher, 2018-06-19:
Update: I have got it working by executing the following steps:
* Windows Settings -> Time & Language -> Region & Language
* Only have English (United States) as Windows Display Language and Preferred Language (I have removed my local language)
* In the top right window, also select "Administrative Language Settings", a dialog will appear
* Click Copy Settings button, select both checkboxes at the bottom and press ok (wait with reboot)
* Also click Change System locale and check the "Beta" checkbox
* Then, allow the system to reboot and try again

I hope this helps

anphel31, 2018-06-19:
@bderuijsscher , thank you very much for the tip. I have also confirmed this these steps allow the build to proceed on FR-fr and ZH-cn systems.

The only change I needed to make was: "Settings" -> "Time and Language" -> "Region and Language" -> "Administrative Language Settings" -> "Copy settings..." -> Select "Welcome screen and system accounts" -> click "Ok" to reboot system.

@isdqwq, can you try these steps and let us know if it allows you to build the BSP.

isdqwq, 2018-06-20:
@bderuijsscher  @anphel31   thanks very much ,it's work ok
below is the build log.
IOTADK_ROOT : C:\iot-adk-addonkit
ADK_VERSION : 17134
WDK_VERSION : 16299
COREKIT_VER : 17134.1
OEM_NAME    : Contoso

Setting Environment for Architecture x64
Configuring for x64 architecture
KitsRoot       : [C:\Program Files (x86)\Windows Kits\10\]
WDKContentRoot : []
Retail Signing disabled
BSP_ARCH    : amd64
BSP_VERSION : 10.0.0.0
BSPPKG_DIR  : C:\iot-adk-addonkit\Build\amd64\pkgs
MSPKG_DIR   : C:\Program Files (x86)\Windows Kits\10\MSPackages\Retail\amd64\fre

IoTCoreShellv5.1 amd64 10.0.0.0
C:\iot-adk-addonkit\Tools>bsptools\APLx64\convert.cmd
Processing igdlh64.inf
 Authoring APL64.GFX.wm.xml
Processing iaiogpio.inf
 Authoring APL64.GPIO.wm.xml
Processing iaioi2c.inf
 Authoring APL64.I2C.wm.xml
Processing iaiospi.inf
 Authoring APL64.SPI.wm.xml
Processing txei.inf
 Authoring APL64.TXEI.wm.xml
Processing iaiouart.inf
 Authoring APL64.UART.wm.xml
 .pkg.xml files found. Review generated .wm.xml files for correctness
 Converting APL64.CMD.pkg.xml
 Converting APL64.DeviceLayout.pkg.xml
 Converting APL64.iclsClient.pkg.xml
 Converting APL64.OEMDevicePlatform.pkg.xml
 Converting APL64.SystemInformation.pkg.xml
Fixing the BSPFM.xml file
Fixing the TestOEMInput.xml file
Fixing the RetailOEMInput.xml file
Conversions done

IoTCoreShellv5.1 amd64 10.0.0.0
C:\iot-adk-addonkit\Tools>buildbsp.cmd APLx64
Sign APLx64 bsp binaries...
Processing C:\iot-adk-addonkit\Source-x64\BSP\APLx64
 [sys files]
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.SystemInformation
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igdkmd64.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GPIO\iaiogpio.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.I2C\iaioi2c.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.SPI\iaiospi.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.TXEI\x64\TeeDriverW8x64.sys
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.UART\iaiouart.sys
 [dll files]
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\iga32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\iga64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igc32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igc64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd10iumd32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd10iumd64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd11dxva32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd11dxva64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd12umd32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igd12umd64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfx11cmrt32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfx11cmrt64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfxcmjit32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\igfxcmjit64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\libmfxhw32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\libmfxhw64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\mfx_mft_h264ve_32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.GFX\mfx_mft_h264ve_64.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsClient.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsClientInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsProxy.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\iclsProxyInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\libeay32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x64\ssleay32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsClient.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsClientInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsProxy.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\iclsProxyInternal.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\libeay32.dll
   Signing C:\iot-adk-addonkit\Source-x64\BSP\APLx64\Packages\APL64.iclsClient\x86\ssleay32.dll
Building APLx64 bsp packages
 Processing APL64.CMD.wm.xml
 Processing APL64.DeviceLayout.wm.xml
 Processing APL64.GFX.wm.xml
 Processing APL64.GPIO.wm.xml
 Processing APL64.I2C.wm.xml
 Processing APL64.iclsClient.wm.xml
 Processing APL64.OEMDevicePlatform.wm.xml
 Processing APL64.SPI.wm.xml
 Processing APL64.SystemInformation.wm.xml
 Processing APL64.TXEI.wm.xml
 Processing APL64.UART.wm.xml
Running feature merger
Running FeatureMerger for APLx64
  Exporting APLx64 FM files
  Processing APLx64FMFileList.xml

IoTCoreShellv5.1 amd64 10.0.0.0

parameshbabu, 2018-06-20:
Thanks for the confirmation. closing

closed 2018-06-20 by parameshbabu