ms-iot/iot-adk-addonkit #240: BSWx64 fails for Braswell with Windows 1803

JeffMushmo, 2018-06-05:
First I've called `installoemcerts` and `buildpkg all`.

I've also tried executing the `Tools/bsptools/BSWx64/convert.cmd` before `buildpkg all`.

Here's the output for `buildpkg all`:
```
Building all provisioning packages
Processing all provisioning packages
 Processing Provisioning.Enroll
 Processing Appx.IoTCoreDefaultApp
 Processing Appx.IoTOnboardingTask
 No .pkg.xml files found.
Building all packages under C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Common\Packages
 Processing Custom.BCD.wm.xml
 Processing Custom.OOBEApp.wm.xml
 Processing Custom.Settings.wm.xml
 Processing Device.SystemInformation.wm.xml
 Processing DeviceLayout.GPT4GB.wm.xml
 Processing DeviceLayout.GPT8GB-R.wm.xml
 Processing DeviceLayout.MBR4GB.wm.xml
 Processing DeviceLayout.MBR8GB-R.wm.xml
 Processing Provisioning.Enroll.wm.xml
 Processing Recovery.GPT-BCD.wm.xml
 Processing Recovery.GPT-BcdEdit.wm.xml
 Processing Recovery.MBR-BCD.wm.xml
 Processing Recovery.MBR-BcdEdit.wm.xml
 Processing Registry.Version.wm.xml
 Processing Security.Bitlocker.wm.xml
 Processing Security.DeviceGuard.wm.xml
 Processing Security.SecureBoot.wm.xml
 Processing Settings.HotKey.wm.xml
Building all packages under C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Source-x64\Packages
 Processing Appx.IoTCoreDefaultApp.wm.xml
 Processing Appx.IoTOnboardingTask.wm.xml
 Processing AzureDM.Services.wm.xml
Building all bsp packages
 Processing BSW64.GFX.wm.xml
 Processing BSW64.GPIO.wm.xml
 Processing BSW64.I2C.wm.xml
 Processing BSW64.UART.wm.xml
 Processing CustomMBMx64.GPIO.wm.xml
 Processing MBMx64.Customization.wm.xml
-----build done-----

IoTCoreShellv5.1 amd64 10.0.0.0
```

When trying to buildimage, I get the following error:
```
ThreadId39516 ERROR: Imaging!EnsureAllFilesExist: The OEM input + feature manifest contains the package file(s) that do not exist:
        'C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Build\amd64\pkgs\Intel.BSW64.SystemInformation.cab'
        'C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Build\amd64\pkgs\Intel.BSW64.OEMDevicePlatform.cab'
        'C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Build\amd64\pkgs\Intel.BSW64.DeviceLayout.cab'
```

Here's the log:
```
Creating JetwaySBC1 Test Image
Build Start Time :  9:03:21.14
Building Packages with product specific contents with version 10.0.0.0
 Processing Registry.Version.wm.xml
 Processing JetwaySBC1
 Processing Custom.Cmd.wm.xml
 Processing Provisioning.Auto.wm.xml
 Processing Recovery.WinPE.wm.xml
Running FeatureMerger for OEM packages
  Exporting OEMFM files
  Processing OEMFMFileList.xml
Running FeatureMerger for BSWx64
  Exporting BSWx64 FM files
  Processing BSWx64FMFileList.xml
  [91mError: Featuremerger failed for BSWx64FMFileList.xml. See C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Build\amd64\buildfm_bsp_BSWx64.log[0m
Creating Image...
ImageApp: Setting CPU Type to 'amd64'.
Microsoft (C) imageapp 10.0.10011.16384

ThreadId39516 INFO Imaging: Reading the OEM Input XML file...
ThreadId39516 INFO FeatureAPI: Successfully validated the OEM Input XML: C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Source-x64\Products\JetwaySBC1\TestOEMInput.xml
ThreadId39516 INFO Imaging: Reading the FM XML file 'C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Source-x64\BSP\BSWx64\Packages\BSWx64FM.xml'...
ThreadId39516 INFO FeatureAPI: Successfully validated the Feature Manifest XML: C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Source-x64\BSP\BSWx64\Packages\BSWx64FM.xml
ThreadId39516 INFO Imaging: See Flash.cbs.log for details.
ThreadId39516 INFO Imaging: Performance Results:
ThreadId39516 INFO 	Total Run Time:	00:00:00.4028739
ThreadId39516 INFO 	Storage Stack Time:	00:00:00.0000003
ThreadId39516 INFO 	Total run time:	402.8739ms
ThreadId39516 INFO 	Total staging time:	0ms
ThreadId39516 INFO 	Total commit time:	0ms
ThreadId39516 INFO 	CompDB Total Time (ms):	0ms
ThreadId39516 INFO 	Image format:	FFU
ThreadId39516 INFO 	Stage file time:	0
ThreadId39516 INFO 	UpdateOS time:	0
ImageApp: Error log can be found at 'C:\MARQME~4\OSImages\WIN10-~1\IOT-AD~2\Build\amd64\JETWAY~1\Test\Flash.ImageApp.log'.
"CreateImage JetwaySBC1 Test" failed with error 1
```

This is correct. These files don't exist. But I recall them being generated with previous versions of IoT Core.

Am I missing a step? Is the Intel Braswell BSP compatible with the current release of IoT Core?

JeffMushmo, 2018-06-05:
Managed to get passed this issue. Apparently the call to BSWx64/convert.cmd deletes the .pkg.xml files from the BSP package folder.

But now I'm getting this error:
```
ThreadId61204 ERROR: CbsPackage::CheckCBSFeatureInfo: Feature(featureId:BASE) inside FM(featureManifestId:) doesn't contain a FIP package
ThreadId61204 ERROR: CbsPackage::CheckCBSFeatureInfo: Feature(featureId:OEM_BSW64_DRIVERS) inside FM(featureManifestId:) doesn't contain a FIP package
ThreadId61204 ERROR: CbsPackage::CheckCBSFeatureInfo: Invalid FIP info. Following packages are not in Feature(featureId:OEMDEVICEPLATFORM_BSWX64_DEVICE) inside FM(featureManifestId:). But are present in FIP info inside Intel.BSW64.OEMDevicePlatform
ThreadId61204 ERROR: Intel.OEMDEVICEPLATFORM_BSWX64_DEVICE.BSWx64.MainOS
ThreadId61204 ERROR: CbsPackage::CheckCBSFeatureInfo: Invalid FIP info. Following packages are not in Feature(featureId:DEVICELAYOUT_BSWX64_SOC) inside FM(featureManifestId:). But are present in FIP info inside Intel.BSW64.DeviceLayout
ThreadId61204 ERROR: Intel.DEVICELAYOUT_BSWX64_SOC.BSWx64.MainOS
ThreadId61204 ERROR: Imaging!ProcessCompDBPackages: FIP information in FM 'BSWx64FM.xml' packages for the following Features are incorrect:
        Error : BASE
        Error : OEM_BSW64_DRIVERS
        Error : OEMDEVICEPLATFORM_BSWX64_DEVICE
        Error : DEVICELAYOUT_BSWX64_SOC
```

I recall getting an error like this before as a result of not running the BSWx64\convert.cmd. But in this case, I've run it.

parameshbabu, 2018-06-06:
After running convert.cmd, can you please do a clean build, the FIP info error should be gone on a clean build.

JeffMushmo, 2018-06-06:
This is after a clean build. 

Do I need to re-copy the BSP packages, then run convert.cmd before I run a clean build?

The previous path I've taken is: (copy BSP) > buildpkg > convert.cmd > buildpkg > buildimage

parameshbabu, 2018-06-06:
Yes please run convert.cmd and then do a clean build ( buildpkg and buildimage )

JeffMushmo, 2018-06-06:
I appreciate the quick response. I tried what you said. Here's the result:
'''
ThreadId4568 ERROR: CbsPackage::CheckCBSFeatureInfo: Feature(featureId:BASE) inside FM(featureManifestId:) doesn't contain a FIP package
ThreadId4568 ERROR: CbsPackage::CheckCBSFeatureInfo: Feature(featureId:OEM_BSW64_DRIVERS) inside FM(featureManifestId:) doesn't contain a FIP package
ThreadId4568 ERROR: CbsPackage::CheckCBSFeatureInfo: Feature(featureId:OEMDEVICEPLATFORM_BSWX64_DEVICE) inside FM(featureManifestId:) doesn't contain a FIP package
ThreadId4568 ERROR: CbsPackage::CheckCBSFeatureInfo: Feature(featureId:DEVICELAYOUT_BSWX64_SOC) inside FM(featureManifestId:) doesn't contain a FIP package
ThreadId4568 ERROR: Imaging!ProcessCompDBPackages: FIP information in FM 'BSWx64FM.xml' packages for the following Features are incorrect:
        Error : BASE
        Error : OEM_BSW64_DRIVERS
        Error : OEMDEVICEPLATFORM_BSWX64_DEVICE
        Error : DEVICELAYOUT_BSWX64_SOC
Error : Build failed. See Log for details
'''

JetwaySBC1_Test.log:
```
Creating JetwaySBC1 Test Image
Build Start Time :  1:11:21.60
Building Packages with product specific contents with version 10.0.0.0
 Processing Registry.Version.wm.xml
 Processing JetwaySBC1
 Processing Custom.Cmd.wm.xml
 Processing Provisioning.Auto.wm.xml
 Processing Recovery.WinPE.wm.xml
Running FeatureMerger for OEM packages
  Exporting OEMFM files
  Processing OEMFMFileList.xml
Running FeatureMerger for BSWx64
  Exporting BSWx64 FM files
  Processing BSWx64FMFileList.xml
Creating Image...
ImageApp: Setting CPU Type to 'amd64'.
Microsoft (C) imageapp 10.0.10011.16384

ThreadId4568 INFO Imaging: Reading the OEM Input XML file...
ThreadId4568 INFO FeatureAPI: Successfully validated the OEM Input XML: C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Source-x64\Products\JetwaySBC1\TestOEMInput.xml
ThreadId4568 INFO Imaging: Reading the FM XML file 'C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Source-x64\BSP\BSWx64\Packages\BSWx64FM.xml'...
ThreadId4568 INFO FeatureAPI: Successfully validated the Feature Manifest XML: C:\MarqMetrixGit\OSImages\Win10-IoT-Core\iot-adk-addonkit\Source-x64\BSP\BSWx64\Packages\BSWx64FM.xml
ThreadId4568 INFO Loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\bodiy25n.oj5\update.mum
ThreadId4568 INFO Loading manifest file C:\Users\IoTCoreAdmin\AppData\Local\Temp\bodiy25n.oj5\update.mum
ThreadId4568 INFO 
Done loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\bodiy25n.oj5\update.mum
ThreadId4568 INFO Loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\qbttxrul.bv5\update.mum
ThreadId4568 INFO Loading manifest file C:\Users\IoTCoreAdmin\AppData\Local\Temp\qbttxrul.bv5\update.mum
ThreadId4568 INFO 
Done loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\qbttxrul.bv5\update.mum
ThreadId4568 INFO Loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\bf3wzinv.wzz\update.mum
ThreadId4568 INFO Loading manifest file C:\Users\IoTCoreAdmin\AppData\Local\Temp\bf3wzinv.wzz\update.mum
ThreadId4568 INFO 
Done loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\bf3wzinv.wzz\update.mum
ThreadId4568 INFO Loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\qukgl5qp.uan\update.mum
ThreadId4568 INFO Loading manifest file C:\Users\IoTCoreAdmin\AppData\Local\Temp\qukgl5qp.uan\update.mum
ThreadId4568 INFO 
Done loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\qukgl5qp.uan\update.mum
ThreadId4568 INFO Loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\kwyy5rae.saw\update.mum
ThreadId4568 INFO Loading manifest file C:\Users\IoTCoreAdmin\AppData\Local\Temp\kwyy5rae.saw\update.mum
ThreadId4568 INFO 
Done loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\kwyy5rae.saw\update.mum
ThreadId4568 INFO Loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\y05olora.e25\update.mum
ThreadId4568 INFO Loading manifest file C:\Users\IoTCoreAdmin\AppData\Local\Temp\y05olora.e25\update.mum
ThreadId4568 INFO 
Done loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\y05olora.e25\update.mum
ThreadId4568 INFO Loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\rekf2vtj.v12\update.mum
ThreadId4568 INFO Loading manifest file C:\Users\IoTCoreAdmin\AppData\Local\Temp\rekf2vtj.v12\update.mum
ThreadId4568 INFO 
Done loading Package:C:\Users\IoTCoreAdmin\AppData\Local\Temp\rekf2vtj.v12\update.mum
ThreadId4568 INFO Imaging: See Flash.cbs.log for details.
ThreadId4568 INFO Imaging: Performance Results:
ThreadId4568 INFO 	Total Run Time:	00:00:05.6093745
ThreadId4568 INFO 	Storage Stack Time:	00:00:00.0000001
ThreadId4568 INFO 	Total run time:	5609.3745ms
ThreadId4568 INFO 	Total staging time:	0ms
ThreadId4568 INFO 	Total commit time:	0ms
ThreadId4568 INFO 	CompDB Total Time (ms):	3650.7018ms
ThreadId4568 INFO 	Image format:	FFU
ThreadId4568 INFO 	Stage file time:	0
ThreadId4568 INFO 	UpdateOS time:	0
ThreadId4568 INFO 	CompDB Total Time:	00:00:03.6507018
ImageApp: Error log can be found at 'C:\MARQME~1\OSImages\WIN10-~1\IOT-AD~2\Build\amd64\JETWAY~1\Test\Flash.ImageApp.log'.
"CreateImage JetwaySBC1 Test" failed with error 1
```

parameshbabu, 2018-06-06:
can you share the version of the tools and IoTCore you are using?

JeffMushmo, 2018-06-06:
ADK: 1803
WDK: 1803
IoT Core Packages: April 2018 Update
iot-adk-addonkit: master branch

parameshbabu, 2018-06-06:
Can you also share the host OS version ?

JeffMushmo, 2018-06-06:
I've been able to reproduce this issue on two machines.
My dev box: Windows 10 Pro 1803 17134.48
Azure VM: Windows 10 Pro N 1803 17134.48

JeffMushmo, 2018-06-06:
It's interesting that my buildpkg/buildimage still work for the 1709 bits.

deepgray, 2018-09-23:
the Baytrails BSP has teh same problems.

parameshbabu, 2018-10-02:
I tried this again and it does work. Just to make sure the steps are clear
(copy BSP) > convert.cmd > newproduct TestProd (BSP) > buildpkg > buildimage
the New product that you create should be after the convert.cmd as this updates the OEMInput files. I guess that is what might have caused your problems above. Let me know if this fixes the problem.

parameshbabu, 2018-10-08:
Closing this issue now. If you can still repo, please reopen and let us know.

closed 2018-10-08 by parameshbabu

JeffMushmo, 2018-10-09:
Ok. I got this working on my end. This time, when copying the BSP I made sure to also copy the OEMInputSamples. Also, I deleted the product so that I called newproduct after convert.cmd as you mentioned.

All works fine and have confirmed that the on-screen keyboard works for x64 versions.