ms-iot/iot-adk-addonkit #244: What is the "CUSTOM_CMD" and "PROV_AUTO" means?

moyanming, 2018-06-20:
The sample will be used for new product, but there are OEM features "CUSTOM_CMD" and "PROV_AUTO", does those should be "OEM_CUSTOM_CMD" and "OEM_PROV_AUTO"?
https://github.com/ms-iot/iot-adk-addonkit/blob/a6b43fb53a13d21380c5c4223c428eea711b658a/Source-arm/BSP/QCDB410C/OEMInputSamples/TestOEMInput.xml#L70

parameshbabu, 2018-06-20:
No. Just include it as custom_cmd and prov_auto. Should be fine.

Regards,
Paramesh

https://github.com/ms-iot/iot-adk-addonkit/blob/a6b43fb53a13d21380c5c4223c428eea711b658a/Source-arm/BSP/QCDB410C/OEMInputSamples/TestOEMInput.xml#L70<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fms-iot%2Fiot-adk-addonkit%2Fblob%2Fa6b43fb53a13d21380c5c4223c428eea711b658a%2FSource-arm%2FBSP%2FQCDB410C%2FOEMInputSamples%2FTestOEMInput.xml%23L70&data=02%7C01%7Cpabab%40microsoft.com%7C385caf651df74a54dd6208d5d660bbba%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636650633312124054&sdata=CruVZtSxCxEPs7pbd5jpQD21FCjwL%2Bfj201LmIh4bwk%3D&reserved=0>

parameshbabu, 2018-06-20:
> The document here<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.microsoft.com%2FEN-US%2Fwindows-hardware%2Fmanufacture%2Fiot%2Fdeploy-your-app-with-a-standard-board%23span-idupdatetheprojectsconfigurationfilesspanspan-idupdatetheprojectsconfigurationfilesspanspan-idupdatetheprojectsconfigurationfilesspanupdate-the-projects-configuration-files&data=02%7C01%7Cpabab%40microsoft.com%7C12568da0b56144deb56f08d5d665c8b4%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636650655000827255&sdata=5eM95QBJBh8ffXZEQmGfO%2FDHold3pZb2TROsIl1va7s%3D&reserved=0> ask that those should be the OEM_CUSTOM_CMD, OEM_PROV_AUTO.
> Please see the issue here<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FMicrosoftDocs%2Fcommercialization-public%2Fissues%2F168&data=02%7C01%7Cpabab%40microsoft.com%7C12568da0b56144deb56f08d5d665c8b4%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636650655000857277&sdata=FOjMpzW%2B1q4TKsjmP%2F1%2BEUCJmwr2SkIlX2DmJnXz8CQ%3D&reserved=0>: The docs should update for 1803! #168<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fms-iot%2Fiot-adk-addonkit%2Fissues%2F168&data=02%7C01%7Cpabab%40microsoft.com%7C12568da0b56144deb56f08d5d665c8b4%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636650655000867289&sdata=4%2FsaSsHZK4TOkmuxCS79LjVoZu087tmyjgkgF4DIrt0%3D&reserved=0>

Thanks for reporting. Will have the docs updated.

Regards,
Paramesh

parameshbabu, 2018-06-20:
On the docs on custom_cmd, you can use either one , preferably the one without oem_

It was kept to maintain backward compatibility

See oemcommonfm.xml

     
     <PackageFile Path="%PKGBLD_DIR%" Name="%OEM_NAME%.Custom.Cmd.cab">
       <FeatureIDs>
         <FeatureID>OEM_CustomCmd</FeatureID>
         <FeatureID>CUSTOM_CMD</FeatureID>
       </FeatureIDs>
     </PackageFile>
     <PackageFile Path="%PKGBLD_DIR%" Name="%OEM_NAME%.Provisioning.Auto.cab">
       <FeatureIDs>
         <FeatureID>OEM_ProvAuto</FeatureID>
         <FeatureID>PROV_AUTO</FeatureID>
       </FeatureIDs>
     </PackageFile>

Regards,
Paramesh
________________________________
From: Paramesh Babu
Sent: Tuesday, June 19, 2018 9:51:01 PM
To: ms-iot/iot-adk-addonkit; ms-iot/iot-adk-addonkit
Cc: Comment
Subject: Re: [ms-iot/iot-adk-addonkit] What is the "CUSTOM_CMD" and "PROV_AUTO" means? (#244)

Thanks for reporting. Will have the docs updated.

Regards,
Paramesh

closed 2018-06-21 by moyanming