ms-iot/iot-adk-addonkit #245: Build an image Failed

isdqwq, 2018-06-20:
win10 x64 CN os ,and change language to EN
### double click  IoTCoreShell-x64.cmd  and run below command

- .\bsptools\APLx64\convert.cmd   OK
- buildbsp APLx64  OK

when build test image get below issue, how to fix.
[image]

```
C:\iot-adk-addonkit\Tools>newproduct board APLx64
Creating board Product with BSP APLx64
board product directories ready

IoTCoreShellv5.1 amd64 10.0.0.0
C:\iot-adk-addonkit\Tools>buildimage board test
Creating board test Image, see C:\iot-adk-addonkit\Build\amd64\board_test.log for progress
ThreadId9768 ERROR: Imaging: Failed to initialize ImageStorageManager. Please ensure you are running ImageApp.exe as an Administrator and try again. Message: Unable to load DLL 'ImageStorageService.dll': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
ThreadId9768 ERROR: An unhandled exception was thrown: Imaging: Failed to initialize ImageStorageManager. Please ensure you are running ImageApp.exe as an Administrator and try again. Message: Unable to load DLL 'ImageStorageService.dll': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
Error : Build failed. See Log for details

IoTCoreShellv5.1 amd64 10.0.0.0
C:\iot-adk-addonkit\Tools>
```

isdqwq, 2018-06-25:
@parameshbabu @anphel31  hi, is there any update?

TerrenceHenry, 2018-07-06:
@parameshbabu @anphel31 @isdqwq 
I have the same problem. Any update on how to solve this issue?
TIA

TerrenceHenry, 2018-07-07:
I solved my problem of the missing DLL files by uninstalling both ADK and WDK, then reinstalling both with all options selected for ADK.

mjmeans, 2018-07-20:
I haven't needed to build an image since February, before my dev machine updated to 1806. But I need to now, and I am having this problem as well. I dl'ed the latest iot-adk-addonkit, uninstalled the old 16299 ADK, and installed the latest 17134 ADK. Then to test it, I did lab-1a from https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/create-a-basic-image. But this fails with the same error that OP reported. I did a complete search of my HDD for ImageStorageService.dll and it is missing completely.

The IoT Manufacturing Guide shows WDK as optional (https://docs.microsoft.com/en-us/windows-hardware/manufacture/iot/set-up-your-pc-to-customize-iot-core) so it should not be required for lab-1a. So either update the guide so that it is clear that it is NOT optional, or clear under what specific circumstances it is not be optional, or fix the iot-adk-addonkit to not require this dll.

isdqwq, 2018-07-27:
@mjmeans  so do you build image success?

mjmeans, 2018-07-27:
Yes, the image builds without any error message; however, I have yet to try to the resultant flash.ffu in an actual device.

parameshbabu, 2018-10-08:
Windows 10 1809 ADK version has fixes for language issues. Let us know if you still have open issues here by reopening with exact steps or as a new issue. Closing this one.

closed 2018-10-08 by parameshbabu