ms-iot/iot-adk-addonkit #248: RPI2 - SetDisplayResolution - Your device only supports one display resolution

tordstavik, 2018-06-29:
I am not able to set a custom display resolution with the command SetDisplayResolution <width> <height>.

Message:
Your device only supports one display resolution
I can't change the existing device resolution settings

I have gone through the OEMINPUT.xml file, Flash.DeviceDB.xml and every other place that mentions resolution or supportedresolution, changed the resolution or added additional. Nothing allows me to change it manually.

I do get different resolutions depending on the screen i plug it into.
When I build with Windows 10 Iot Core (17134) I am able to set resolution manually.

anphel31, 2018-06-29:
Can you share which version of IoT core was being used when you saw this error message?

tordstavik, 2018-07-02:
17134

closed 2019-04-17 by saraclay