ms-iot/iot-adk-addonkit #249: How to add a REG_BINARY value through Custom.Settings

mwettste, 2018-07-02:
I try to add a binary value to the registry. However, I can't seem to figure out the correct way to format it.
That's what I tried so far (values have been shortened for this post):
`
<regKey keyName="$(hklm.system)\ControlSet001\Enum\ACPI\QCOM24AD\0">
    <regValue name="Security" type="REG_BINARY" value="0100049000000000000000000000000014000000020060" />
</regKey>`
second try:
`
<regKey keyName="$(hklm.system)\ControlSet001\Enum\ACPI\QCOM24AD\0">
    <regValue name="Security" type="REG_BINARY" value="01,00,04,90,00,00,00,00,00,00,00,00,00,00,00,00,14,00,00,00,02,00,60" />
</regKey>`

With both versions, the reg export looks the same:
`"Security"=hex:01,00,04,90,00,00,00,00,00,00,00,00,00,00,00,00,14,00,00,00,02,\
  00,34`
Everything is correct until the last value (60 originally) which becomes 34 for some reason. Later in the value, more bytes get replaced (e.g. 04 to 02). Additionally, the complete value is shortened for some reason.
What would be the correct way to add this value to the registry? I want to avoid having a .reg file and "reg import" that on first startup if somehow possible.

parameshbabu, 2018-07-02:
The first syntax looks ok.. see https://github.com/ms-iot/iot-adk-addonkit/blob/master/Common/Packages/Security.BitLocker/Security.Bitlocker.wm.xml for a sample.

mwettste, 2018-07-06:
Thanks for the sample, so that should be correct. However, I still can't understand why the values get converted to something else. Any ideas?

parameshbabu, 2018-07-06:
I just tried the first format and I see the generated code which seems ok. Can you check if the generated manifest file inside the cab contains this fragment? 
    <registryKey keyName="HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Enum\ACPI\QCOM24AD\0">
      <registryValue name="Security" valueType="REG_BINARY" value="0100049000000000000000000000000014000000020060" />
    </registryKey>

mwettste, 2018-07-09:
Thanks for the suggestion. I checked the contents of the manifest file and the value of the key is exactly the same as configured in the .wm.xml file. However, after flashing the ffu to the device and checking the registry on the device, the value has changed and therefor the security descriptor does not work.
What's in the wm.xml file:
`<regKey keyName="$(hklm.system)\ControlSet001\Enum\ACPI\QCOM24AD\0">
            <regValue name="Security" type="REG_BINARY" value="010004900000000000000000000000001400000002006000040000000000140000000010010100000000000512000000000018000000001001020000000000052000000020020000000014000000001001010000000000050B0000000000180000000010010200000000000F0200000001000000" />
        </regKey>`

What reg export exports:
`"Security"=hex:01,00,04,90,00,00,00,00,00,00,00,00,00,00,00,00,14,00,00,00,02,\
  00,34,00,02,00,00,00,00,00,18,00,00,00,00,10,01,02,00,00,00,00,00,05,20,00,\
  00,00,20,02,00,00,00,00,14,00,00,00,00,10,01,01,00,00,00,00,00,05,12,00,00,\
  00`

mwettste, 2018-07-19:
How can we proceed with this issue? Would it be better to file a bug through Collaborate, since it could be an issue in IoT Core as well?

anphel31, 2018-07-20:
How did you verify the registry value is correct (reg.exe query from SSH or something else?)? Can you confirm there are no other drivers/devices on your system that are setting this same registry key? And could you please share the version information printed when opening IoTCoreShell.cmd.
Another thing to check: Is the value correct in the image before running on the device? (you can "wpimage mount flash.ffu" and then mount the registry hives with regedit to check)

mwettste, 2018-08-07:
Sorry for the delayed response, I was out of office the last two weeks.
Thanks a lot for your suggestions. I thought I checked that there is no Security Key without my settings package being added to the OS... however, this is not the case. The Security Key also exists without my settings, so it looks like that one overwrites my initial settings.

In that case I see no other solutions than to run a script at startup and hope that the oemcustomization.cmd is run late enough to overwrite the default Security key.

Thanks again and I will close this issue.

closed 2018-08-07 by mwettste